openscad/openscad #4017: On 2021.01 the X axis is unexpectedly offset by 0.1

Kreyren, 2021-12-19:
DISCLAIMER: Unless my math is wrong..

.. The X Axis is unexpectedly offset by +0.1 which significantly complicates a design with 1mm tolerance:

[image]

### Code:

Note: Beginner in OpenSCAD, not confident in simplifying the code

```openscad
//#!/usr/bin/env openscad

//@ Copyright (C) All Rights Reserved - Jacob Hrbek <kreyren@rixotstudio.cz> in 19/12/2021 00:10 CEST

ancor_width = 1;
ancor_height = 10.42;
ancor_lenght = 4.2;

// NOTE-BUG(Krey): On OpenSCAD 2021.01 I had to add `0.1` to difference shapes to avoid rendering issues, remove once this is fixed upstream
    // - Decided to undo the 0.1 as it makes the code ugly af and added version-gate

// NOTE-BUG(Krey): On my version of OpenSCAD (01.2021) I have to use `{#` to maintain syntax highlighting for the contents, remove once this is fixed upstream
    // - Krey: Submitted https://github.com/openscad/openscad/issues/4016

// NOTE-BUG(Krey): On OpenSCAD 2021.01 the X axis is offset by 0.1 which results in hell and pain on designs that work with 1mm accuracy

// NOTE-BUG(Krey): The OpenSCAD 2021.01 does not ignore shebang so it ourputs an annoying error.. and refuses to run -> Undo the comment once this is fixed upstream
    // - Krey: Submitted https://github.com/openscad/openscad/issues/4015

// NOTE-WORKAROUND(Krey): Added workarounds to highlighted issues, remove once these issues are fixed by upstream in a stable release + 2 months and add assert for a version check

// DNR-TODO(Krey):
// - [ ] Design is sufficient to ancor to the Ender 3
// - [ ] Bugs were submitted and referenced
// - [ ] Peer-review
// - [X] Problematic blocks version-gated and scheduled for removal

///! Module designed to be ancored on the alluminium extrusion blocks.
///! Originally designed to ancor on Ender 3's extrusion to support the Z axis at the top
module ancor(x, y, z) {

    difference() {
        cube([x, y, z], center = true);
        
        // Remove top left corner
        translate([-z/10,0,x]) {#
           // Workaround(Krey): Workaround to OpenSCAD 2021.01 issues
           if (version_num() == 82.02101e+7)
             cube([x/2+0.1,y+0.1,x/2+0.1], center = true);
           else
             cube([x/2,y,x/2], center = true);
        }
        
        // Remove bottom left corner
        translate([-z/10,0,-x]) {#
           // Workaround(Krey): Workaround to OpenSCAD 2021.01 issues
           if (version_num() == 82.02101e+7) 
             cube([x/2+0.1,y+0.1,x/2], center = true);
           else
             cube([x/2,y,x/2], center = true);
        }
    }
}

// Create the ancor
ancor(ancor_lenght, ancor_width, ancor_height);
```

Kreyren, 2021-12-19:
Closing as pebcak

closed 2021-12-19 by Kreyren

jordanbrown0, 2021-12-19:
The rendering artifacts that cause the requirement to have the subtrahend of a difference operation extend outside the original object is called "Z-fighting", and is documented in a note on difference() at

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/CSG_Modelling#difference

See also the Wikipedia article at https://en.wikipedia.org/wiki/Z-fighting .

The basic problem is that when presented with a zero-thickness object, the renderer can't quite decide whether it's there or not.

And yes, it's a pain in the neck.