openscad/openscad #4019: Change the tab indentation character from `⟶` on `|` or `»`

Kreyren, 2021-12-19:
I was told that in the tabs vs spaces war the `⟶` character is used by spaces-people who want to deter other users from using tabs as it's probably the most distracting character that can be used for the tab whitespace:

[image]

Proposing to use the more workable:
a) pipe `|`  symbol which is a favorite choice for emacs users where the whitespace can't be easily aligned in the center of the whitespace
a) Right-Pointing Double Angle Quotation Mark `»` which is favorite choise for VSCodium assuming it being aligned in the center of the indentation else it's too distracting.

.. preferably with rainbow indentation to orient in the levels of indentation.

Kreyren, 2021-12-19:
Note: The pipe character is also more portable:

[image]

ChrisCoxArt, 2021-12-19:
Just because your browser or IRC client can't handle extended characters and unicode....

Kreyren, 2021-12-19:
> Just because your browser or IRC client can't handle extended characters and unicode.... -- @ChrisCoxArt (https://github.com/openscad/openscad/issues/4019#issuecomment-997319349)

It can handle it as you can see below :p but my browser randomizes fingerprint so some values are changed on random to change the behavior.

t-paul, 2021-12-19:
As already discussed, that's a feature of QScintilla which is the editor component used by OpenSCAD.
If you would like that feature, the request needs to be made for that project, there's nothing OpenSCAD can do.

Quoting the [documentation](https://docs.huihoo.com/pyqt/QScintilla2/classQsciScintilla.html#ab83469cc9550eadcd5d4e8ca3d20d07b)
> This enum defines the different whitespace visibility modes. When whitespace is visible spaces are displayed as small centred dots and *tabs are displayed as light arrows pointing to the right*.

closed 2021-12-19 by t-paul