openscad/openscad #4021: Unrepairable Open Mesh

HarkonenBade, 2021-12-19:
I have been attempting to edit an existing STL file from a thingiverse project (https://www.thingiverse.com/thing:3262984/files, trying to convert the screw holes into slots to accomodate a closer spacing on a different load cell).
However, openscad claims that the mesh for the stl is not closed `ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. ` and despite trying all the options of repairs with 3d builder, microsofts online service and meshlabs the STL remains broken. Also in many of those services they consider the mesh to be fine and not broken, prusa slicer also considers the mesh to be acceptable.

All of this was tested with openscad version 2019.1 and 2021.1 on window x64.

nophead, 2021-12-19:
NetFabb Studio says the original has self intersections.

[image]

Can you post a version that is supposedly fixed? I can guarantee it isn't because CGAL is never wrong when it says it non-manifold and will not accept anything that isn't. Most STLs on Thingiverse are not manifold in my experience.

HarkonenBade, 2021-12-19:
Here are my 'repaired' stls, they have been processed via prusa slicer and the web repair system. Prusa slicer considers them to be closed with no faults. MS 3D builder also thinks they are fine.

(Files are provided in accordance with the original licence:
Filler - Filament spool holder with integrated filament weight sensor bysb43201is licensed under the
Creative Commons - Attribution - Non-Commercial - Share Alikelicense.
By downloading this thing, you agree to abide by the license: Creative Commons - Attribution - Non-Commercial - No Derivatives)

[HoldGuide_Loadcell.zip](https://github.com/openscad/openscad/files/7741357/HoldGuide_Loadcell.zip)

HarkonenBade, 2021-12-19:
Using those I can reproduce the error with a simple:
```
difference(){
    translate([0, -16, 0]) import("HoldGuide_Loadcell_base.stl");
    cube(25);
}
```

nophead, 2021-12-19:
That part still has self intersections according to NetFabb:

[image]

Unfortunately the free versions can't fix them because it is hard.

HarkonenBade, 2021-12-19:
Ah fair. Well, thank you for helping me determine that the model was the issue in question. A friend who knows how to drive blender has helped me make the required tweaks that I needed for my part.

kintel, 2024-04-01:
Status as of 2024-03-24: Works when using manifold, still broken with CGAL