openscad/openscad #4022: PushPullFeeder assertion errors

brainstorm, 2021-12-20:
Here's a possible test fixture OpenSCAD can use in its testsuite? Seems to fail on the dev release from today, works fine on the 2021-01 release:

https://github.com/markmaker/PushPullFeeder

[image]

```
OpenSCAD 2021.12.19 
https://www.openscad.org/
 
Copyright (C) 2009-2021 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design '/Users/rvalls/dev/personal/PushPullFeeder/PushPullFeeder.scad'.
Compiling design (CSG Tree generation)...
ECHO: "Overall Size:", 15.7
Compiling design (CSG Products generation)...
Geometries in cache: 504
Geometry cache size in bytes: 3439144
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 468544
Compiling design (CSG Products normalization)...
Normalized tree has 446 elements!
Compile and preview finished.
Total rendering time: 0:00:01.784
 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "Overall Size:", 15.7
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1144
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1144
Geometries in cache: 753
Geometry cache size in bytes: 9011456
CGAL Polyhedrons in cache: 84
CGAL cache size in bytes: 104416584
Total rendering time: 0:03:38.024
Top level object is a 3D object:
   Simple:        yes
   Vertices:    16044
   Halfedges:   49556
   Edges:       24778
   Halffacets:  17630
   Facets:       8815
   Volumes:        15
Rendering finished.
```

brainstorm, 2021-12-20:
This is how it looks on build 2021.01 (correct rendering with no assert errors):

[image]

t-paul, 2021-12-20:
As this is a big (also very cool) project, please give some details on the settings. Otherwise I don't see how to help. E.g. what changes against the default values are done - it must be possible to reproduce without customizer setup. It's impossible in the bugtracker to handle without reducing to a smaller test case.

brainstorm, 2021-12-20:
It should fail with the default settings actually, just open it up and render... if you want a smaller test case, perhaps only selecting the "base plate" should trigger the issue, as part of it is not rendered properly in the first comment. Make sure you have the "Individual switches" drop down menu selected (instead of "Everything") in that first folded menu.

Hope that helps? Cheers!

thehans, 2021-12-21:
I narrowed it down to there being an issue with `extrusion_mount_2D();`

Replacing the base_plate code with just this results in an assertion error (albeit somewhat different assertion than the original report):
```
union() {
    echo(lever_thickness+base_thickness-emboss);
    echo(base_thickness-emboss);
    linear_extrude(height=lever_thickness+base_thickness-emboss, convexity=10) extrusion_mount_2D();
    // thinner parts of the base plate (embossed)
    linear_extrude(height=base_thickness-emboss, convexity=10) extrusion_mount_2D();
}
```

If you put a root modifier on `extrusion_mount_2D();` and F6 render, there is some visible odd geometry in one of the corners.  Some pics with successive zoom:
[image]
[image]
[image]

thehans, 2021-12-21:
Here is a version of the scad trimmed down to *only* ~900 lines:
```
function fn_by_err(r_eff) = let(da = (acos((1 - (_fp / max(r_eff, (_fp * 2))))) * 2), _fn = ceil(max(12, min((360 / da), 3600)))) _fn;
function curve(a0, a1, r0, r1, x, y, scale = false) = [let(r_eff = max(abs(r0), abs(r1)), n = ceil(((fn_by_err(r_eff) * abs((a1 - a0))) / 360))) (for(i = [0 : n]) (let(a = (a0 + ((i * (a1 - a0)) / n)), ratio = ((a - a0) / (a1 - a0)), r = (((1 - ratio) * r0) + (ratio * r1)), o = ((i * 90) / n)) (scale ? [((x + ((cos(o) * cos(a0)) * r0)) + ((sin(o) * cos(a1)) * r1)), ((y + ((cos(o) * sin(a0)) * r0)) + ((sin(o) * sin(a1)) * r1))] : [(x + (cos(a) * r)), (y + (sin(a) * r))])))];
function arc(p0, p1, angle) = let(d = (p1 - p0), m = ((p1 + p0) / 2), n = [-d.y, d.x], c = (m + (((cos((angle / 2)) / sin((angle / 2))) * n) / 2)), b = (p0 - c), a0 = atan2(b.y, b.x), a1 = (a0 + angle), r = sqrt((pow(b.x, 2) + pow(b.y, 2)))) curve(a0, a1, r, r, c.x, c.y);
function spring_relaxed_endpoint(p0, p1, angle, angle_relax) = let(d = (p1 - p0), m = ((p1 + p0) / 2), n = [-d.y, d.x], c = (m + (((cos((angle / 2)) / sin((angle / 2))) * n) / 2)), b = (p0 - c), a0 = atan2(b.y, b.x), a1 = (a0 + angle), r = sqrt((pow(b.x, 2) + pow(b.y, 2))), angle_new = (angle + angle_relax), b_new = ((b * angle) / angle_new), c_new = (p0 - b_new), p1_new = (c_new + [((cos(angle_new) * b_new.x) - (sin(angle_new) * b_new.y)), ((sin(angle_new) * b_new.x) + (cos(angle_new) * b_new.y))])) p1_new;function distance(dx = 0, dy = 0, dz = 0) = sqrt(((pow(dx, 2) + pow(dy, 2)) + pow(dz, 2)));
function rot(p, x0, y0, rot) = let(s = sin(rot), c = cos(rot)) [(((c * p.x) - (s * p.y)) + x0), (((c * p.y) + (s * p.x)) + y0)];
function rot_each(v, x, y, rot) = [for(p = v) (rot(p, x, y, rot))];

module extrusion_mount_2D() {
	nut_y = ((extrusion_mount_y + (extrusion_mount_h / 2)) - (extrusion_mount_unit / 2));
	base_begin_r = distance((base_begin - reel_x), (-base_height - reel_y));
	left_edge_y = ((extrusion_mount_y - (extrusion_mount_h / 2)) + wall);
	left_edge_x = (reel_x + sqrt((pow(base_begin_r, 2) - pow((left_edge_y - reel_y), 2))));
	extrusion_mount_contour = [[(tape_inset_begin + e), (-base_height + e)], each (arc([base_begin, (-base_height + e)], [left_edge_x, left_edge_y], (atan2((left_edge_y - reel_y), (left_edge_x - reel_x)) - atan2((-base_height - reel_y), (base_begin - reel_x))))), each (arc([left_edge_x, left_edge_y], [(extrusion_mount_x - (extrusion_mount_w / 2)), (extrusion_mount_y - (extrusion_mount_h / 2))], (30 - atan2((left_edge_y - reel_y), (left_edge_x - reel_x))))), each (arc([(extrusion_mount_x - (extrusion_mount_w / 2)), (hanger_y - hanger_dy)], [((extrusion_mount_x - (extrusion_mount_w / 2)) + (extrusion_mount_tension / 2)), (hanger_y + hanger_dy)], -hanger_angle)), [(extrusion_mount_x - (extrusion_mount_w / 2)), (extrusion_mount_y + (extrusion_mount_h / 2))], each ([for(xx = [0.5 : (extrusion_mount_w / extrusion_mount_unit)]) (each (arc([(((extrusion_mount_x - (extrusion_mount_w / 2)) + (xx * extrusion_mount_unit)) - (extrusion_mount_slot_outer / 2)), (extrusion_mount_y + (extrusion_mount_h / 2))], [(((extrusion_mount_x - (extrusion_mount_w / 2)) + (xx * extrusion_mount_unit)) + (extrusion_mount_slot_outer / 2)), (extrusion_mount_y + (extrusion_mount_h / 2))], (2 * extrusion_mount_slot_angle))))]), [((extrusion_mount_x + (extrusion_mount_w / 2)) - (extrusion_mount_tension / 2)), (extrusion_mount_y + (extrusion_mount_h / 2))], [(extrusion_mount_x + (extrusion_mount_w / 2)), (nut_y + (extrusion_mount_slot_outer / 2))], each (arc([((extrusion_mount_x + (extrusion_mount_w / 2)) - extrusion_mount_slot_thickness), (nut_y + (extrusion_mount_slot_inner / 2))], [((extrusion_mount_x + (extrusion_mount_w / 2)) - extrusion_mount_slot_thickness), (nut_y - (extrusion_mount_slot_inner / 2))], 30)), each (arc([((extrusion_mount_x + (extrusion_mount_w / 2)) - extrusion_mount_slot_thickness), (nut_y - (extrusion_mount_slot_inner / 2))], [((extrusion_mount_x + (extrusion_mount_w / 2)) + extrusion_mount_strength), (nut_y + (extrusion_mount_strength / 2))], 90)), each (arc([((extrusion_mount_x + (extrusion_mount_w / 2)) + extrusion_mount_strength), (nut_y + (extrusion_mount_strength / 2))], [max(((extrusion_mount_x + (extrusion_mount_w / 2)) + extrusion_mount_strength), -extrusion_mount_strength), (extrusion_mount_y + (extrusion_mount_h / 2))], -15)), each (arc([(base_end - inset_edge), (extrusion_mount_y + (extrusion_mount_h / 2))], [base_end, ((extrusion_mount_y + (extrusion_mount_h / 2)) + inset_edge)], 90)), [base_end, (-base_height + e)], [(tape_inset_end - e), (-base_height + e)]];
	difference() {
		polygon(extrusion_mount_contour);
		if(extrusion_mount_hollow_out_right) {
			strength_r = (extrusion_mount_strength * 0.1);
			strength_scale = 0.6;
			offset(r = strength_r) offset(delta = ((-extrusion_mount_strength * strength_scale) - strength_r), chamfer = true) intersection() {
				polygon([[tape_inset_begin, (-base_height + (extrusion_mount_strength * strength_scale))], each (extrusion_mount_contour), [tape_inset_end, (-base_height + (extrusion_mount_strength * strength_scale))]]);
				translate([(handle_lock_axle_x + (handle_lock_diameter / 2)), (extrusion_mount_y + (extrusion_mount_h / 2))]) square([((-sprocket_pitch * 2) - handle_lock_axle_x), (-extrusion_mount_y - (extrusion_mount_h / 2))]);
			}
		}
		if(extrusion_mount_hollow_out_left) {
			strength_r = (extrusion_mount_strength * 0.1);
			strength_scale_left = 0.6;
[... cut ...]
head_half_y = (((nozzle_mount_diameter / 2) + nozzle_plate_portrusion) + nozzle_adapter_strength);
head_face_dist = ((head_half_y + (nozzle_adapter_strength * 2)) + nozzle_mount_groove);
slide_on_z = (nozzle_shaft_length - (nozzle_adapter_strength * 3));
echo("Overall Size:", (reel_width + reel_holder_thickness));

union() {
	linear_extrude(height = ((lever_thickness + base_thickness) - emboss), convexity = 10) extrusion_mount_2D();
	linear_extrude(height = (base_thickness - emboss), convexity = 10) extrusion_mount_2D();
}

```

brainstorm, 2021-12-28:
Thanks @thehans! Is that "odd geometry" something that needs to be solved in the model or in OpenSCAD itself?

t-paul, 2021-12-28:
It seems to be in the model, the outlier is that point
```
[(extrusion_mount_x - (extrusion_mount_w / 2)),    (extrusion_mount_y + (extrusion_mount_h / 2))], 
```
between the 3rd and 4th `each ... arc()`.

markmaker, 2021-12-29:
Hi @t-paul 

Thanks for singling this out. I'll have a look (I'm the author of the model). 

Just to know: I was under the impression that such outliers (or rather "inliers"), i.e., polygon edges that fall together, are allowed, are they not? Two coinciding polygon lines going the opposite ways should cancel each other out, right? 

I also use them to create holes in polygons such as in this extrusion:

[image]

It usually worked so far.

Anyways, I'll try and fix the model there. 

Thanks,
_Mark

markmaker, 2021-12-29:
This should now be fixed, but because I did not test with OpenSCAD 2021.12.19, and I could not provoke the error in OpenSCAD 2021.01, it is technically unconfirmed. 

However, I would like to confirm that the "inlier" was in-deed a coinciding polygon edge, and that is was properly eliminated in prior OpenSCAD version. So this still seems to be a regression in OpenSCAD 2021.12.19.

I wonder if it can be provoked using one or either of these:

```OpenSCAD
linear_extrude(height=1) {
    polygon([
        [-1, -1],
        [-1, 1],
        [-2, 1], // outlier
        [1, 1],
        [1, -1],
    ]);
}
    
translate([5, 0, 0])
linear_extrude(height=1) {
    polygon([
        [-3, 2],
        [-3, -1],
        [-1, -1],
        [-1, 1],
        [-2, 1], // inlier
        [1, 1],
        [1, -1],
        [3, -1],
        [3, 2],
    ]);
}
```

_Mark

markmaker, 2021-12-29:
Confirmed regression and minimal test-case:

```
polygon([
        [-3, 2],
        [-3, -1],
        [-1.5, -1],
        [-1, 1],
        [-2, 1],     // inlier
        [1, 1],
        [1, -1],
        [3, -1],
        [3, 2],
    ]);

```

OpenSCAD 2021.01

[image]

OpenSCAD 2021.12.23.ai9926

[image]

Strangely, the situation reverses, when I create an enclosed hole (which IMHO is a valid application):

```
polygon([
        [-3, 2],
        [-3, -1],
        [-1.5, -1],
        [-1, 1],
        [-2, 1],       // enclosed hole
        [-2, 1.5],     
        [-2.5, 1.5],   
        [-2.5, 1],     
        [1, 1],
        [1, -1],
        [3, -1],
        [3, 2],
    ]);
```

OpenSCAD 2021.01

[image]

OpenSCAD 2021.12.23.ai9926

[image]

_Mark

nophead, 2021-12-29:
GIGO

t-paul, 2021-12-29:
I'm not sure there's was a guarantee for removing those kind of spikes. The only thing that's pretty much always assured is that duplicate points are ok and will collapse to a single point.

That said, maybe there's a way to check the polygon by passing it through the Clipper clean function and warn if it changes or something like that.

markmaker, 2021-12-29:
Hi,

I now found this in the [underlying CGAL doc about polygons](https://doc.cgal.org/latest/Polygon/index.html#Chapter_2D_Polygon): 

> A polygon is a closed chain of edges. Several algorithms are available for polygons. For some of those algorithms, it is necessary that the polygon is simple. A polygon is simple if edges don't intersect, except consecutive edges, which intersect in their common vertex.

I assume that (partially) coinciding counts as "intersection" too (even if that is probably not terminologically accurate).

This limitation should perhaps be documented for OpenSCAD [`polygon()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#polygon) too. 

There are still geometries in the PushPullFeeder model that use these "self-incursions". So far OpenSCAD has supported them, which probably means that none of the said algorithms (that require polygons to be "simple" in that way) was internally touched. I will fix these problems in the model if they appear in future versions of OpenSCAD. 

Or maybe this would be the point where I port this to a [FreeCAD Python script](https://wiki.freecadweb.org/Topological_data_scripting) ;-)

_Mark

thehans, 2021-12-29:
I can't think of any recent changes to openscad code that would have caused such a regression.
We could try to single out a commit with a `git bisect` ...  or it might also be differences in CGAL's internal behavior (in which case `git bisect` on *our* codebase wouldn't find it), since CGAL the version we build against has been updated between the last release and current.
Either way, thanks for providing a much simplified test case.

rcolyer, 2021-12-30:
For your future updates, @markmaker, here is the expected syntax for an enclosed hole, using the multiple paths option:
```
polygon(
  [
    [-3, 2],
    [-3, -1],
    [-1.5, -1],
    [-1, 1],
    [1, 1],
    [1, -1],
    [3, -1],
    [3, 2],
    // enclosed hole
    [-2, 1],
    [-2, 1.5],
    [-2.5, 1.5],
    [-2.5, 1]
  ],
  [
    [0,1,2,3,4,5,6,7],
    [8,9,10,11]
  ]);
```

markmaker, 2021-12-30:
Thanks, I've seen it. 

The following is just my two cents ;-)

In OpenSCAD due to its functional language heritage, it's very hard to number things continuously when you're nested in expressions, loops, sub-functions, and/or concatenating stuff. There are no mutable variables and you can't take the "so far size" of an array you are still "expressing".

Therefore, I believe the numbered vertices approach is bad design choice for a functional language API. Both [`polygon()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#polygon) and [`polyhedron()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron) are almost impossible to use because of that.

For a functional language. the API would be much more concise, if it were to take an array of arrays of vertices for both `polygon()` and `polyhedron()` and then let the underlying code deal with duplicate vertices and number them. I doubt there would be a significant performance or memory requirement difference. I wouldn't be surprised, if in fact this is already done _anyway_, to eliminate _inadvertent_ duplicate vertices. 

_Mark

trivalik, 2022-11-26:
I run into this for just a STL combine, but in my case it fails with version 2021.01.

The snapshot version 2022.11.18 fails with:
```
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1144
```

I just tried to fix a hole in kraken.stl (source https://www.thingiverse.com/thing:5609223/comments) by rotating and combine.

```
import("C:/Users/User/Downloads/kraken.stl");
rotate([0, 0, 120])import("C:/Users/User/Downloads/kraken.stl");
```

I though that the hole is maybe a defect so I cut it out with a cube and saved this as stl. Then I used the above to combine them, but also not working.

Here code to cut the hole:
```
difference() {
  import("C:/Users/User/Downloads/kraken.stl");
  translate([0,0,-100])cube([100,120,200]);
}
```

nophead, 2022-11-27:
You can't mend non-manifold STLs in OpenSCAD with boolean ops because each operand of union, etc, has to be manifold. You need to fix the STL with some other tool, like Meshmixer.

MichaelAtOz, 2022-11-27:
There seems to be something else. BTW @nophead I think by hole @trivalik means the chunk cut out of a leg in the STL.
In Netfabb there are no holes, but one self-intersection (inside the middle, can't see it from outside).
However I can union with a cube.
[image]

Perhaps union()ing two of 75% identical curves, plus some floating point magic?

trivalik, 2022-11-27:
I cut an other angle and it works, maybe be chunk cut was still in, or the self-intersect you mentioned.

working:

```
module halfKraken() {
  difference() {
    import("C:/Users/Triv/Downloads/kraken.stl");
    translate([-101,001,-100])cube([200,200,200]);
  }
}

union() {
  halfKraken();
  rotate([0, 0, 180])halfKraken();
}
```

Sorry to bother.

closed 2026-05-09 by github-actions[bot]