openscad/openscad #4024: CGAL error in large render

delooper, 2021-12-21:
When rendering a fairly large OpenSCAD scene, I get an unfamiliar error message:

**ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 321**

The text appears with a red-pink background colour. 

If I modify my $fn parameter to be slightly lower resolution I can sometimes get the error message to go away. 

Also, when this error message does appears, a feature generated by the minkowski() operation does not render.  So I have an idea for what part of my code is generating the problem. 

Is this a familiar problem?  I have not encountered it before. 

Running 2019.05 in Ubuntu.

ChrisCoxArt, 2021-12-21:
You might want to try 2021.01, or the current dev builds.

thehans, 2021-12-21:
CGAL is the library responsible for render of 3D boolean operations.  When an error like that occurs, its usually due to having input geometry which is not 2-manifold.  There is some description of what it means to be manifold on the wiki pages here:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Export
and here: 
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#Debugging_polyhedra

delooper, 2021-12-22:
Thanks.  I imagine my heavy usage of constructive geometry techniques (hull, minkowski, etc) is leading to too fine a mesh in some places.   These errors must be due to the data being close to the precision limits of floats.

thehans, 2021-12-22:
You mentioned that you already have an idea for which part of the code causes the problem.  I would recommend singling out specific module instantiations (like your minkowski) with the [root modifier](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Root_Modifier) `!` plus possibly [disable modifier](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Disable_Modifier) `*` (or simply commenting out parts) until you have a minimal script which reproduces the error.  

If you can narrow down the problem down to a sensible sized script (a self-contained single file is best) for others to reproduce then please post it.  

Without any way to reproduce, this is not an actionable report.  Closing for now, can re-open if sample is provided.

closed 2021-12-22 by thehans

jordanbrown0, 2021-12-22:
Floating point precision limits are generally a myth, for this sort of problem.  Doubles, which are what is almost always used, are good for 15-16 digits.  That's enough to measure the distance to the Sun in tenths of a millimeter, or a 100cm model in proton radii.

The problem with floating point precision limits is that two values that are mathematically equal can end up not-equal by an infinitesimal amount.  There is for practical purposes no problem with values that are mathematically not-equal ending up equal.

On the other hand, OpenSCAD has a grid-snap mechanism with a resolution of around 10^-6.

```
d=1e-5;
cube(1);
translate([1+d,1,0]) cube(1);
```
When d is 1e-5 the cubes do not touch; when it is 1e-6 they touch (and it's a manifold problem).

I expect that you'll run into grid-snap problems long before you run into floating-point precision problems.

The grid snap is there to work around the problems where mathematically equal values end up infinitesimally not-equal.  It solves that problem (usually?) but introduces others.

jordanbrown0, 2021-12-22:
I'm sorry, I missed a key piece.  Binary STL, which I think is the default export format, is only 32-bit floating point, with about 7 digits of precision.  Precision limits could be more of a factor there.  However, that's only an *export* format; I don't know whether it comes into play in the rendering process per se.

Also, 7 digits is still enough to measure a 100cm model to the nearest 10 nanometers.

delooper, 2021-12-22:
@jordanbrown0 Isn't the grid-snap mechanism essentially a software-engineered "floating point" data type?  Seems to function as one.  Given it's only 10^-6, it would not take many geometric operations (compounded on each other) to get down to that resolution. 

@thehans Thanks for the advice.  I was unaware of the # and ! commands.  Will give them a shot.

jordanbrown0, 2021-12-22:
I wouldn't think of it as a pseudo-floating-point.  (Note that it doesn't affect *calculations*, only geometry.)  If anything, I expect that it behaves like a *fixed* point mechanism; I expect that the grid is 1e-6 no matter how big the model is.

delooper, 2021-12-23:
@thehans How do you use the * and # features for debugging purposes?  

Perhaps I don't understand something basic.  When I preview my model, (with or without * and #) I get the preview with various parts highlighted.  But when I go to render, with or without * and #, one portion of my model does not render.  

Interestingly, if I am not changing my code (other than adding and deleting * and #) the CGAL error does not appear, other than the first time after some more substantial code modification.  Every time I click render after the first CGAL error, it does not produce subsequent CGAL errors.  But the model renders with the one portion missing. 

Are you suggesting I make edits to the part of the model that is removed from my renderings?

jordanbrown0, 2021-12-23:
Your model has problems.

There are, I believe, three kinds of problems:
* Not manifold.  Mostly this means that any edge must be shared by exactly two edges.  The classic failure case is two cubes that meet only at a corner edge.
* Self-intersecting polyhedron.  A polyhedron intersects with itself.
* Polyhedron that is not closed.  It isn't entirely covered by faces.

When one of these happens, preview works but render does not.  Two interesting things happen:
* You get an error message, the *first* time that you render.  After that, the results are cached.  Use Design / Flush Caches to reset.
* Part or all of the model will be missing.

Note that these things only happen if CGAL gets involved, which happens if you use any of the boolean operations.  A polyhedron on its own, even if malformed, will not cause a problem... but when you add in anything else, it will.

MichaelAtOz, 2021-12-23:
> the CGAL error does not appear, other than the first time after some more substantial code modification 

 

The model is cached, so re-rendering does not re-process unchanged parts, including any that lead to an error.

When debugging CGAL errors you should do Design/Flush-caches before a render.

Also note if you only do one part [e.g. an import()] it may not need to call CGAL, so always put a cube(1); somewhere to force a union() so it must use CGAL.

 

Use '*' to temporarily turn off processing of that branch of the code tree, if the error goes away, that branch may be the cause.

Naturally that section will not be rendered, so is missing.

 

Conversely '!' only processes that branch and ignores the rest. If there is no error, it is not in that section.

 

Although sometimes it is the interaction of two sections causing the problem, you then need to go up a level in the code tree.

 

'#' is mainly to visualise a section, I use it to confirm I'm examining the correct section of code, corresponding to the area with a visual issue.

It is a PITA finding you have been debugging the wrong end of a model.

 

See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters 

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

delooper, 2021-12-23:
While debugging I'm sometimes getting OpenSCAD to crash with a segfault. 

Would that help narrow down the issue at all? It's only using about 30% of system memory at max.  

I will continue my debugging using the "--debug all" flag, in case that help.

delooper, 2021-12-23:
I found an essentially equivalent way of performing the minkowski() operation without using the minkowski() command.  My error is gone now.  

In short, I was performing a minkowski operation between a solid that looked much like a thin helix, with the 2nd input object being a sphere with $fn a low number, like 4 or 8.  Not certain why it was failing, but I've worked around my problem, for now.

thehans, 2021-12-23:
Did you ever try a newer version of OpenSCAD, as was suggested in the very first reply here?  
http://openscad.org/downloads.html#snapshots
The CGAL assertion handling has improved since 2019 release and should at least no longer crash.