openscad/openscad #4025: Unable extrude 3D shapes

mdziczkowski, 2021-12-21:
### System

- OS: Debian
- OpenSCAD version: 2021.01

### Issue description

While attempting to ritation extrude on some 3D shapes (for example: cylider), I get a following warning:

> WARNING: module <shape I want to extrude goes here>() does not support 
> child modules in file , line 1

delooper, 2021-12-21:
I think that's by design.  OpenSCAD only lets you extrude from 2d shapes, as far as I know.

If you want to extrude 3d shapes, you can linearly extrude them using two copies (one a translate of the other) and the hull() operation.

nophead, 2021-12-21:
Can you post a simple an example of the code that fails? You can only rotate_extrude() 2D shapes but I would expect a different error message in that case. For example:
```
rotate_extrude()
    cylinder();
```

Gives:
```
WARNING: Ignoring 3D child object for 2D operation in file , line 2
Rendering cancelled on first warning.
WARNING: No top level geometry to render
```

mdziczkowski, 2021-12-22:
I'm not sure if a linear extrude would work in my case because I tryed to design a cylinder with a round hole inside.

To let the things clear, sorry about the leak of my knowledge of the design process in OpenScad, but I had just discovered it today and I'm testing it, because I'm trying to find an best free CAD application for me ;-)

thehans, 2021-12-22:
You can make a cylinder with a round hole using either rotate_extrude or linear_extrude.
The important aspect of both is that they take 2D geometry as input.
```
module cyl_with_hole(d1,d2,h) {
  linear_extrude(h) difference() { 
    circle(d=d2); 
    circle(d=d1);
  }
}

module cyl_with_hole2(d1,d2,h) {
  rotate_extrude() translate([d1/2,0]) square([(d2-d1)/2,h]); 
}

$fs = 0.5;
$fa = 1;

cyl_with_hole(5,20,10);
translate([30,0]) cyl_with_hole2(5,20,10);
```
The two approaches produce identical results, with the exception of a slight difference in number of facets on the inner hole, due to how the shapes are generated.

In the future please post scripting questions to the openscad mailing list or IRC (not as a Github issue).

I would also recommend looking over the documentation such as tutorials, manual, and "cheat sheet" linked from the website: http://openscad.org/documentation.html

closed 2021-12-22 by thehans

mdziczkowski, 2022-01-02:
Thanks. This was helpfull :-)