openscad/openscad #4027: Vector swizzling extensions

t-paul, 2021-12-23:
PR #4010 introduces GLSL style [vector swizzling](https://www.khronos.org/opengl/wiki/Data_Type_(GLSL)#Swizzling) for (xyzw, rgba sets).

Quite a number of use cases in OpenSCAD usage seem to extend the dimensions, e.g. 2D -> 3D which is done in GLSL via contructors like `vec3(v.xy, 0)`. As we don't have that in OpenSCAD it might be easier to extends the swizzling feature to add `0` or `1` values.

Suggested options so far:
* Add `0` and `1`, e.g. `v.xy01`
  This works in most cases, specifically every combination that is not just numbers. But this also depends on the pretty uncommon definition of IDs in OpenSCAD that allows leading digits for variable names (except for floating point values with 'e'/'E'). While this is looking nice it would make it even harder to eventually remove that surprising feature (although it's not clear if that ever can happen)
* Add `_` and `I`, e.g. `v.xy_I`
  Solves the potential problem with leading digits, but uses a slightly more obscure naming convention. The `_` looks quite ok, the `I` would be explained via "Identity"

Other suggestions:
- [ ] Add warning for out-of-bounds access instead of silently adding `undef`
- [ ] Add warning for invalid lookup names instead of silently return `undef`

jordanbrown0, 2025-02-18:
Note that a simple implementation of this doesn't work, even ignoring the "identifiers starting with digits" future problem.

The lexer matches a fractional number with no zero
```
{D}*\.{D}+{E}? |
```
before matching an ID
```
"$"?[a-zA-Z0-9_]+       { parserlval.text = strdup(yytext); return TOK_ID; }
```

jordanbrown0, 2025-02-18:
Also PR#4833 Bitwise Operators makes 0x... not be a valid identifier.

jordanbrown0, 2025-03-09:
My comment of a few weeks ago (discussing lexer issues) is correct but doesn't explain why it's an issue.

The issue is that `v.0yz` would need to be lexically analyzed as an identifier (`v`), a dot (`.`), and another identifier (`0yz`), but is instead analyzed as an identifier (`v`) a number (`.0`) and another identifier (`yx`).

coryrc, 2025-10-24:
> Note that `concat()` will already do some of this; `concat(v.xy, 0)` yields `[ v.x, v.y, 0 ]`.

True

> OpenSCAD almost doesn't have l-values, since variables are immutable. I don't immediately see what it would mean to put a swizzled name on the left side of an `=`.

`newvec%xy0 = othervec%yx` creates `newvec`. Only possible if not using `.` as the signifier.