openscad/openscad #4034: [Feature request] Expansion of the documentation and toturials

mdziczkowski, 2022-01-02:
Hello. I would like to request that the documentation and toturials of OpenSCAD become expanded and wroten in a more beginner friendly way.

Currently, the both mentioned are wroten in a not as easy to understand for the beginers way and mant things haven't been explained at all.

One from the things that are missing is a transformation to recieve elements like (for example) in with one side the material is beeing concaveand at the user's requested shape and ausing in the same time that on the opposite side of this material the same shape becomes convexed.

If this proposal would get implemented, it would help many to create better projects

UBaer21, 2022-01-02:
have you seen http://openscad.org/documentation.html#tutorial

mdziczkowski, 2022-01-02:
@UBaer21 yes, I did, but there are presented only examples on simple shapes, but there is not much about the more advances usages, like one I had presented as an example

UBaer21, 2022-01-02:
You wrote about "beginner friendly"  and now about advanced and complex objects. 
Did you visited https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Tips_and_Tricks  there are more advanced objects.

Also i am not sure to understand the object you are describing.  Do you want something like  an object substracted from itself?
like this 
[image]

Or have a look at the[ libraries](https://openscad.org/libraries.html) which give you more advanced features.

UBaer21, 2022-01-02:
Further you can find https://openscad.org/documentation-books.html  
And especially [mastering Openscad](https://mastering-openscad.eu/buch/introduction/) which can be read online

t-paul, 2022-01-02:
Sounds like an overview page would be useful. Collecting all the different available resources in order of difficulty / level of familiarity with OpenSCAD. Maybe we could focus the discussion a bit more on that.

A very general ticket is pretty much impossible to handle and declare closed. I don't expect there will ever be a situation where there's no improvement possible.

thehans, 2022-01-02:
> Currently, the both mentioned are wroten in a not as easy to understand for the beginers way and mant things haven't been explained at all.

If you want things to be improved, you'll need to be more specific.  Can you list some specific features which you feel "haven't been explained at all"?

mdziczkowski, 2022-01-02:
@UBaer21 I had refered myself to the things that needed to be worked at with the wiki: that it should be wroten in a way that even begingers from every part of the earth could very easyly understand and that many more complex way to use it aren't described (I gave an example of one from it).

> Did you visited https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Tips_and_Tricks there are more advanced objects.

yes, I had seen it tru as well, but there was no example of what I search for

> Also i am not sure to understand the object you are describing. Do you want something like an object substracted from itself?

I try to sktch it in a graphical program and insert it as an attachement when I get it done

@t-paul
> Sounds like an overview page would be useful. Collecting all the different available resources in order of difficulty / level of familiarity with OpenSCAD. Maybe we could focus the discussion a bit more on that.

yes, that would be a good idea. It would improve much the navigation in between the scatered thru the whole wiki pages and allow quickly find needed informations

mdziczkowski, 2022-01-02:
I attach a quick sketch in cross-section

[image]

UBaer21, 2022-01-02:
that could be archived with  a  difference operation or an extruded polygon.
Isn't this what you find in  [chapter 9](https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_9#Challenge) of the tutorial ? 
(only the lower half moved in opposite direction)

Would it help if there were more Examples?  There are also a lot of build in Examples (File⇒Examples)
Maybe we need an overview page for these Examples - would that help?

mdziczkowski, 2022-01-02:
@UBaer21 it is not the same thing. In the sketch I tryed to draw the pressing out a shape in a material.

Let me try to write it on a real life example. Have you ever seen how are metal pot's (before attaching the handles) produced ? it's something simmiliar to what I try to recieve but in a smaller scale.

UBaer21, 2022-01-02:
Well this is not how openSCAD works - there is no morphing or moving of selected vertices or edges. So for this you need to define every point manually (or via a function).  What you want is a feature of a mesh editing software (like blender or MeshLab ).
But not [CSG](https://en.wikipedia.org/wiki/Constructive_solid_geometry) like openSCAD

mdziczkowski, 2022-01-02:
@UBaer21 Do you think that it should be in this case requested as a feature?

I think it's an quite important transformation possibility because many projects may require it and other (even similiar like) ones.

I consider OpenSCAD as an interesting project, but even if there was put so much great work, it still need a loot of improvements to become a super engeneering and design tool

jordanbrown0, 2025-03-09:
It seems like there are potentially at least three issues here:
* Submitter wants a way to deform a disk into a pan.  OpenSCAD doesn't have such a mechanism.  (And would probably never have precisely *that* operation, even if it had general deform mechanisms.  You'd have to define your own deformation that would achieve the result you wanted.)
* Documentation doesn't describe how to create all possible objects.  OpenSCAD is like LEGO.  It has a few simple pieces that can be combined in many ways to create complex objects.  You have to figure out how to combine them to create your particular object.  Galleries of examples can help you to learn techniques, but will only rarely give you exactly what you want.
* The documentation could be improved.  Certainly true, now and probably forever - and not actionable.  Pointers to specific pieces of documentation that are unclear are useful, especially in conjunction with suggested improvements.  A general "needs improvement" comment isn't something that can be addressed.

closed 2025-03-09 by jordanbrown0