openscad/openscad #4035: [Feature request] Change of the way of design

mdziczkowski, 2022-01-02:
I would like to propose of changing the way how the designs are beeing made. Instead of generating the shapes thru the code, it would be easier to draw them in a visual way and the code shold be used only to do additional transformacions in situacion when such one could not be done in the visual way.

This proposal would allow to create the projects faster and more efficient

t-paul, 2022-01-02:
Sorry, you are asking for a completely different program then. Also please consider that people have different approaches for designing things. If you prefer visual design, there's good open source options too, like LibreCAD for 2D designs and FreeCAD for basically everything visual (with some additional scripting support).

closed 2022-01-02 by t-paul

jordanbrown0, 2022-01-03:
And of course you can use a drawing-sculpting sort of program to generate an STL that you can then import into OpenSCAD and further modify.