openscad/openscad #4036: macOS: Provide universal binaries (arm64 + x86_64)

kintel, 2022-01-03:
While OpenSCAD works reasonably well under Rosetta on Apple Silicon, we should really provide native binaries.
The common way of achieving this is through universal binaries:

macOS snapshots and releases should be (automatically) built as universal binaries.

kintel, 2022-01-03:
I'm working on PR for this in #4011, but it's a bit more work than expected. If anyone has recent experience with universal builds on macOS, ideas and advice are welcome!

kintel, 2022-01-03:
Some discussion:

Two main ways of building universal binaries:
1.  **bottom-up universal**: Build each library dependency as a universal binary, then build a universal app as a single build linking against these libs.
2. **top-down universal**: Build dependencies and full OpenSCAD app bundle for each architecture separately, and combine the two into a universal binary.

**bottom-up universal**:
When shipping PPC and 32-/64-bit fat binaries, we used this approach, so we have some experience making that happen. However, the number of dependencies has grown significantly since then, and doing this significantly complicates the dependency build script.

**top-down universal**:
The main benefit is that we can build everything for a single architecture. We still need to perform cross-builds, but this is a built-in feature of clang so it isn't too hard.
The missing piece for this to work is to build a tool for merging two app bundles with all dependencies.

xarbit, 2022-01-14:
I build openscad for Apple Silicon on my machine, just don't want to use rosetta.. 
Definitely looking forward for official universal builds 👍

kintel, 2022-01-20:
Testers wanted: This is a hot off the press, fully CI-built, universal binary of OpenSCAD: https://10369-1049088-gh.circle-artifacts.com/0/tmp/out/OpenSCAD-2022.01.20.dmg

If you give it a spin, please report any issues (or successes) here.

odaki, 2022-01-20:
@kintel Thank you for your continued support of the macOS version!
Your test build binaries work fine on the x86_64 version of the MacBookPro and the Apple M1 version of the MacMini.

Apple M1:
[image][image][image]

x86_64:
[image][image]

The operating system on both machines is Big Sur 11.6.2.
I'm also looking forward to the official universal build.
Thank you!

p.s. Torsten Paul has kindly informed me of the existence of this test binary.

voxpelli, 2022-01-20:
Seems to be working for me on my MacBook Air (M1, 2020) with MacOS 12.1

[image]
[image]

tcurdt, 2022-01-20:
Seems to be working fine here on macOS 12.1 M1. (awesome!)

xarbit, 2022-01-20:
works for me .. thanks! 
(except the warning about unknown developer etc etc ;))

3Dietrich, 2022-01-20:
great! many little things are better (e.g. Customizer!) , good job!

not new (means: it is also in version 2021 and certainly in 2019...):
When the welcome screen (WS) is active and you call the app by a file.scad, the first time it ignores your file (it shows the WS). When you call the file.scad immediatly again, then it opens correctly.
IMHO it should: directly open the file without WS. As if welcome screen is off. The Wolcome Screen is for starting OpenSCAD.app without content.

not new:
In Console you can select something (or all via shortcut cmd+a) but copy (cmd+c) doesn't work.

kintel, 2022-01-20:
@3Dietrich Would you mind opening separate Github tickets for these, if such tickets don't already exist?

kintel, 2022-01-20:
Btw., I forgot to mention initially that the universal binary _should_ be compatible back to macOS 10.13, but I haven't actually tested that. If anyone happens to have a 10.13 machine, a test would be appreciated!

woferry, 2022-01-20:
The dmg above launches for me, but as soon as I try to open any file it instantly crashes.  Tried opening from the welcome screen, open recent, and the open file browser.  Crashed thread attached.  This is on 12.0.1 on an M1 Max.
[OpenSCAD-CI20220120dmg_crash.txt](https://github.com/openscad/openscad/files/7907072/OpenSCAD-CI20220120dmg_crash.txt)
.

DavidPhillipOster, 2022-01-20:
The first time I launched it to open an existing .scad file it gave a system alert asking for permission to open the Documents directory, (which isn't surprising since in there that OpenSCAD puts its library).

The next time, too. 

But, when I clicked the lock at the bottom of System Preferences > Security & Privacy > Privacy, the next time, OpenSCAD added an entry in the Files and Folders section of that preferences pane. All of Removable Volumes, Download  Folder, Documents Folder and Desktop Folder were checked.

But, even so, double-clicking on a file in the Finder, outside the Documents folder will occasionally being up one instance of the system alert asking for permission.

Ditto for opening a file from the Recent Files menu, or the open dialog box.

Overall, much faster and many bug fixes compared to the released version.

kintel, 2022-01-20:
> The dmg above launches for me, but as soon as I try to open any file it instantly crashes.

Could you try the same from the cmd-line? `./OpenSCAD.app/Contents/MacOS/OpenSCAD ../cube.scad -o out.stl`

..and for good measure: `./OpenSCAD.app/Contents/MacOS/OpenSCAD --info`

whosawhatsis, 2022-01-20:
Thanks for this. No problems so far on an M1 Max Macbook Pro. Will let you know if I see any problems.

woferry, 2022-01-20:
Interestingly, launching OpenSCAD from the Terminal it can open the same files that it (continues to) crash when trying to open when OpenSCAD is launched from the Finder.  So there's some difference between running the executable directly from the Terminal and launching it the 'normal ways' (including "open ./OpenSCAD.app" from the same Terminal window), where it can't open any user files without crashing.

One other observation, it seems that in the states where it crashes opening any of my files, it *can* open the files inside the app bundle, i.e. the ones that show in the examples part of the welcome screen or the examples menu.  But those are the only files it can open without crashing it seems, I tried files in various paths including my homedir, /tmp, Desktop and Documents.  And I'm not getting any prompts about accessing any of those directories, just crashing.

./OpenSCAD.app/Contents/MacOS/OpenSCAD --info                                                             
QObject::startTimer: Timers can only be used with threads started with QThread
Fontconfig warning: "/Users/woferry/Desktop/OpenSCAD.app/Contents/Resources/fonts/fonts.conf", line 84: unknown element "blank"
OpenSCAD Version: 2022.01.20 (git 9ec6469ac)
System information: Mac OS X Version 12.0.1 (Build 21A559) MacBookPro18,2 10 CPUs 64.00 GB RAM 
User Agent: OpenSCAD/2022.01.20 (git 9ec6469ac) (Mac OS X Version 12.0.1 (Build 21A559) MacBookPro18,2)
Compiler: Clang "13.0.0 (clang-1300.0.29.30)"
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.2
QScintilla version: 2.13.1
InputDrivers: 
GLib version: 2.71.0
lodepng version: 20210627
libzip version: 1.8.0
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.3.1
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function
Application Path: /Users/woferry/Desktop/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/woferry/Documents
User Documents Path: /Users/woferry/Documents
Resource Path: /Users/woferry/Desktop/OpenSCAD.app/Contents/Resources
User Library Path: /Users/woferry/Documents/OpenSCAD/libraries
User Config Path: /Users/woferry/Library/Application Support/OpenSCAD
Backup Path: /Users/woferry/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/woferry/Documents/OpenSCAD/libraries
  /Users/woferry/Desktop/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /Users/woferry/.local/share/fonts
  /Users/woferry/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/woferry/Library/Fonts
  /usr/X11/lib/X11/fonts/100dpi
  /usr/X11/lib/X11/fonts/75dpi
  /usr/X11/lib/X11/fonts/OTF
  /usr/X11/lib/X11/fonts/Speedo
  /usr/X11/lib/X11/fonts/TTF
  /usr/X11/lib/X11/fonts/Type1
  /usr/X11/lib/X11/fonts/cyrillic
  /usr/X11/lib/X11/fonts/encodings
  /usr/X11/lib/X11/fonts/misc
  /usr/X11/lib/X11/fonts/util
  /System/Library/Fonts/Supplemental
  /usr/X11/lib/X11/fonts/encodings/large

GLEW version: 2.1.0
OpenGL Version: 2.1 Metal - 76.1
GL Renderer: Apple M1 Max
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: Cocoa / CGL
PNG generator: Core Foundation
OS info: Mac OS X Version 12.0.1 (Build 21A559) (arm64 kernel)
Machine: 64-bit

ChrisCoxArt, 2022-01-20:
> The dmg above launches for me, but as soon as I try to open any file it instantly crashes. 
The crash reports shows boost::filesystem throwing an exception during parsing of the file, as it tries to get the current path to the file.
```
bool parse(SourceFile *&file, const std::string& text, const std::string &filename, const std::string &mainFile, int debug)
{
  fs::path filepath = fs::absolute(fs::path(filename));
  mainFilePath = fs::absolute(fs::path(mainFile));
```
My first guess would be something odd in the file path, but I'm not seeing that on your second report.
Are you opening files from another volume, or from within a cloud managed (dropbox, Google Drive, etc.) directory?

kintel, 2022-01-20:
@woferry Thanks. That observation you make about bundled files is interesting. It crashes inside the file parser (I assume when trying to open a file), so it does sound like this is indeed related to the file permission issue that other people have reported.

Since this works for (anecdotally) most users, I guess there is some system configuration on your system that makes this fail. If you have any way of fiddling with settings to see if that changes behavior (or if anyone else listening in can reproduce), that would be really helpful.

I have no idea how the macOS fancy permission system works, nor can I reproduce this on my machine, so this can take a bit of time to sort out. Ideas are welcome :)

ChrisCoxArt, 2022-01-20:
Looks like the try/catch around parsing documents is insufficient to catch everything (only catches HardWarningException, not std::exception or ...)

tcurdt, 2022-01-20:
For the folks where it's crashing - could you try to see if a possible quarantine makes a difference? 

`xattr -drs "com.apple.quarantine" "/path/to/OpenSCAD.app"`

That's what I instinctively did after downloading the build.

ChrisCoxArt, 2022-01-20:
I reproduced the crash if you denied access to the directory that the file is in.
```
  fs::path filepath = fs::absolute(fs::path(filename));
  ```
Throws a std::exception that the code path does not catch, resulting in a call to terminate().

kintel, 2022-01-20:
@ChrisCoxArt Nice! Do you have a (preferrably cmd-line) way of denying/allowing access to a directory that can be used for testing?

ChrisCoxArt, 2022-01-20:
Kinel - no, I don't have a command line method.  I'm denying access when prompted in the GUI.
You can try referencing a document in a directory without read or write permission.

kintel, 2022-01-20:
For some reason, I don't get asked this question, even after removing all permissions in Preferences->Security & Privacy. I guess my laptop has decided I should get access to everything by default : /

As an aside: Has anyone successfully run a macOS M1 guest VM on an M1 host? That may be the best way of establishing a consistently reproducible setup.

tcurdt, 2022-01-21:
> For some reason, I don't get asked this question, even after removing all permissions in Preferences->Security & Privacy. I guess my laptop has decided I should get access to everything by default : /

Maybe a debug vs release build difference? I recall reading somewhere that Xcode signs a debug build in a specific way - so it might have more permissions just on your machine?

tcurdt, 2022-01-21:
Also try `sudo tccutil reset All org.openscad.OpenSCAD`

ChrisCoxArt, 2022-01-21:
I created a new issue to deal with the crash. #4061

ChrisCoxArt, 2022-01-21:
Pull request #4062 submitted to fix issue #4061, but it'll need some review.

kintel, 2022-01-24:
New universal binary with @ChrisCoxArt's fix: https://10400-1049088-gh.circle-artifacts.com/0/tmp/out/OpenSCAD-2022.01.23.dmg

kintel, 2022-02-21:
The nightly development snapshot should be universal starting 2022-01-22. Will close this ticket once they have been verified to work from the official download at https://openscad.org/downloads.html#snapshots

tcurdt, 2022-02-21:
Awesome work. Thanks!

whosawhatsis, 2022-02-22:
@kintel Did you mean 2022-02-22?

kintel, 2022-02-22:
@whosawhatsis Yes, edited my message

whosawhatsis, 2022-02-24:
It ended up taking a few more days, but there's now a universal 2 mac build there. I noticed that the button on the download page still says that it's an intel 64bit build, but MacOS verifies that it is universal.

kintel, 2022-02-24:
Thanks for testing. The full build took longer as we had to rebuild all dependencies and retrigger the build every time CircleCI timed out. Subsequent builds should be fast though.

Will make sure to update the website text

t-paul, 2022-02-24:
Website is updated.

closed 2022-02-25 by kintel

darconeous, 2023-04-21:
The latest official download still appears to be intel only...

[image]

What am I missing?

kintel, 2023-04-22:
The nightly snapshots are universal, and the next official release will be as well. Significant changes were needed to build universally, and it's not feasible to backport that to the previous official release.