openscad/openscad #4037: Out-of-bounds memory access in DXF loader (path identification)

eldstal, 2022-01-05:
### Summary
A DXF-format drawing with particular (not necessarily malformed!) properties may cause an out-of-bounds memory access when imported using `import()`.

### Vulnerable versions
* OpenSCAD Linux (commit eedf37059ca9ce81c09b9270c99257c3d7f74c43)
* OpenSCAD Windows x64 (2021.01)

### Steps to reproduce

Two Proof-of-concept files are provided. `oob_dxfdata_505` is larger, but more reliably reproduces the fault. `oob_dxfdata_k5` is my attempt at a minimal working example, but only reliably crashes the linux build from the latest commit on the `main` branch (i.e. it does not crash in the windows release). Both illustrate the same crash, so I expect the `_min` example to be more helpful in testing.

[oob_dxfdata_505.zip, the larger messier p-o-c](https://github.com/openscad/openscad/files/7817409/oob_dxfdata_505.zip)
[oob_dxfdata_k5.zip, the smaller p-o-c](https://github.com/openscad/openscad/files/7817410/oob_dxfdata_k5.zip)

1. Unzip one of the provided proof of concept files
2. Open the `.scad` file in OpenSCAD
3. Render with F6
4. Observe the application crashing

Alternatively, for headless operation:
1. Unzip one of the provided proof of concept files
2. `openscad --export-format stl -o /dev/null oob_dxfdata_k5.scad`
3. Observe the segmentation fault

### Screenshot
[image]

### Cause
I haven't quite been able to wrap my head around the DXF parser yet, but the OOB access occurs in `src/dxfdata.cc` on one of the lines [470](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L470), [475](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L475), [505](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L505), or [510](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L510) depending on the input file.

It appears to be related to the fact that if multiple line segments share points in common, they are merged into contiguous paths. As a part of this, The [ADD_LINE](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L99) macro manipulates the `grid.data<>` and `lines<>` structures. By creating lines either in the `ENTITIES` section of the DXF file or outside of it, an attacker is able to ensure that the `grid.data` entry created on line [108](https://github.com/openscad/openscad/blob/eedf37059ca9ce81c09b9270c99257c3d7f74c43/src/dxfdata.cc#L108) points to an index in `lines` which never becomes valid.

On line 470 (and the others listed above), this value (`k`) is used as an index into `lines`, out of bounds.

### Impact
It appears that the out-of-bounds data can only be read, and not written. An out-of-bounds read does not expose a security vulnerability on its own, but can be used to bypass automatic security features such as stack canaries and pointer encryption.

### Proposed mitigation
The algorithm employed in `DxfData::DxfData` needs to be revised to prevent this type of aliasing. Without more insight into the DXF format and how it is used by OpenSCAD, I cannot say for sure where the actual flaw lies.

It seems likely that the `ADD_LINE` macro on lines 108-109 is the culprit, since it inserts values that will be used for indexing even though the values are not yet valid indices.

ChrisCoxArt, 2022-01-16:
I see lots of warnings, but no crash on MacOS with the current development code.
I even tried with guard malloc, guard edges, and zombie (use after free) detection enabled.
Ah, but got it to fail with malloc_scribble enabled!

ChrisCoxArt, 2022-01-16:
And there are at least 3 places in the code with the same fragility.
OK, probably best to skip any bad indices, and log a warning about the bad index in the DXF file.  I hope this doesn't break any import tests...

ChrisCoxArt, 2022-01-16:
I thought maybe we could catch the error earlier in the process (when adding lines), but so far I haven't figured out a way to do it.
This DXF code is more than a little fragile.

kintel, 2022-01-16:
OpenSCAD's DXF support is really a rather large piece of technical debt. At some point it's worth digging further into replacing it with an existing library, see https://github.com/openscad/openscad/wiki/Project%3A-Improve-DXF-import-and-export

ChrisCoxArt, 2022-01-16:
Ok, I have a fix that is safe, if not optimal.  Trying to catch the error earlier runs into a lot of code that I cannot decipher easily (it's a huge state machine, with many inter-dependent arrays of data) -- so I cannot say that any changes there would be safe.

eldstal, 2022-02-04:
Thanks for taking a look at this issue! Has the fix been merged?

This vulnerability has been assigned CVE-2022-0496.

ChrisCoxArt, 2022-02-04:
No, the fix has not been merged. It's waiting on other pull requests to be merged (which have been sitting for 3 weeks).

ChrisCoxArt, 2022-02-04:
Now it has been merged! yay!

t-paul, 2022-02-04:
Hold the horses, it will be in a bit :)

commit 52ce0f7a (2022-02-05): Merge pull request #4090 from openscad/dxf-updates

Add safety to line lookups in DXF import (fixes #4037).
commit 770e3234 (2022-02-05): add safety to line lookups in DXF import, fixes #4037

Add safety (test for, and continue past, bad indices).
Report warnings about bad indices
Add variables just to make the array indices easier to read and debug.
closed 2022-02-05 by t-paul