openscad/openscad #4039: ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron associated with linear_extrude

djwarn, 2022-01-07:
An openscad script I have developed gives the error "ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron" when rendered, but not in preview. I have minimised the code to the following, which still exhibits the problem:

a=110;  // twist, limited to 110 to avoid th strands touching
a2=150; // to set the start point such that the strand would 
        // end on the Z axis if a = 180
h=30;   // height
R=20;   // radius
r=2;    // strand radius

// strand 1
linear_extrude(height=h,convexity=10,twist=a,scale=[0,1])
  translate([R*sin(a2),R*cos(a2),0])circle(r);

// strand 2  
rotate(90,[0,0,1])
  linear_extrude(height=h,convexity=10,twist=a,scale=[0,1])
    translate([R*sin(a2),R*cos(a2),0])circle(r);

The above code will exhibit the issue, commenting either strand 1 or strand 2 will render as expected. The value of a is set to avoid the strands crossing in case this was the cause of the issue, as in issue #3805.

ChrisCoxArt, 2022-01-07:
The given script does not work, because RSin and RCos are not defined functions.
If you convert those to R*sin and R*cox, then the script works just fine, without error in OpenSCAD 2021.01

Please fix your script so that it works.
And tell us what OS and what version of OpenSCAD you are using.

ChrisCoxArt, 2022-01-07:
Looks like your error is in the scale parameter to linear_extrude, resulting in infinitely thin lines at the top end of the extrusion, which results in a lot of degenerate triangles, which can't close the mesh.
Change the [0,1] to [0.01,1] so they have some width to create geometry with.

djwarn, 2022-01-07:
Thanks Chris. My code did have the R*sin etc. the '*' somehow got lost in the copy and paste. I confirm that the change to [0.01,1] fixes the issue. Thanks again for your help.

thehans, 2022-01-07:
> Edit: Ah, I see the asterisk disappears from the post. I am using Intel 64 bit 2021.01

Asterisk and other special characters are interpreted as markdown (*italics* for asterisk) in github comments.  To post multiple lines of code, you can put three backquotes on a line by themselves, immediately before and after your code.  That way nothing between those lines will be interpreted as markdown.  (That is why ChrisCoxArt's code sample displayed inside a gray block with monospaced font.)

trivalik, 2022-01-13:
I have another with same error. I add "+2" on cube function to cover whole body in polygone.

```
ueberlappung=0.02;   
    
breiteKleinesQuadrat=1.55;
hoeheKleinesQuadrat=2.4;
breiteGrossesQuadrat=2.67;
hoeheGrossesQuadrat=2.8;
laengeGrossesQuadrat=9.9;
laenge=30.4;

laengeNase=3.7;
hoeheNase=6.35;
hoeheNasenEnde=1.44;
schraegeNase=0.9;
NasenspitzenBreiten=1.45;
NasenErsterAnstieg=1.15 + 1; //etwas mehr damit kreis wegschneiden kann
TolleranzeFuerKreisCut=0.5;
NaseErsteEinbuchtung=2;
    
rad=4;

//if you comment these both out, polyhedron works
translate([0,breiteKleinesQuadrat/-2,0])
cube([laenge-laengeNase+2,breiteKleinesQuadrat,hoeheKleinesQuadrat]);

CubePoints = [
  [  laenge-laengeNase, -breiteKleinesQuadrat/2, 0 ],  //0 vorn links unten
  [ laenge,  -breiteKleinesQuadrat/2,  0 ],  //1 vorn rechts unten
  [ laenge,  breiteKleinesQuadrat/2,  0 ],  //2   hinten rechts unten
  [  laenge-laengeNase,  breiteKleinesQuadrat/2,  0 ],  //3  hinten links unten
  [  laenge-laengeNase + schraegeNase,  -breiteKleinesQuadrat/2,  hoeheNase ],  //4 vorne links oben
  [ laenge,  -breiteKleinesQuadrat/2,  hoeheNasenEnde ],  //5 vorne rechts oben Spitze
  [ laenge,  breiteKleinesQuadrat/2,  hoeheNasenEnde ],  //6 hinten rechts oben Spitze
  [  laenge-laengeNase+schraegeNase,  breiteKleinesQuadrat/2,  hoeheNase ], //7 hinten links oben
  [  laenge-laengeNase+schraegeNase + NasenspitzenBreiten,  -breiteKleinesQuadrat/2,  hoeheNase], //8 vorne rechte Dach
  [  laenge-laengeNase+schraegeNase + NasenspitzenBreiten,  breiteKleinesQuadrat/2,  hoeheNase], //9 hinten rechte Dach
  [  laenge- NaseErsteEinbuchtung+TolleranzeFuerKreisCut,  -breiteKleinesQuadrat/2, hoeheNasenEnde+ NasenErsterAnstieg], //10 vorne rechte verbindung
  [  laenge- NaseErsteEinbuchtung + TolleranzeFuerKreisCut,  breiteKleinesQuadrat/2, hoeheNasenEnde+ NasenErsterAnstieg], //11 hinten rechte verbindung
  ];

// I verified with paper work:-)
CubeFaces = [
  [0,1,2,3],  // bottom
  [0,1,5,10,8,4],  // front
  [4, 7,9, 8],  // top
  [5,6,2,1],  // right
  [7,3,2,6,11,9],  // back
  [7,4,0,3], // left
  [11,10,8,9], // innen zum Dach
  [11,10,5,6], // innen nach außen
  ];

//with polyhedron does F6 return the error, but just with polyhedron and removed cube it works
polyhedron( CubePoints, CubeFaces );

```

jordanbrown0, 2022-01-13:
@trivalik two of the faces on your polyhedron have the points listed in the wrong order - the front and the "innen nach außen".  Reverse the order of the points in those two faces.

To detect this, set View/Thrown Together, preview, and look for purple faces.

This is not considered a problem until you have a second object and so OpenSCAD must do boolean operations.

jordanbrown0, 2022-01-13:
[image]

jordanbrown0, 2022-01-13:
The points need to be listed clockwise when viewed from outside the polyhedron.

trivalik, 2022-01-14:
thanks, that helped me alot. Seems I just not read the right lines in the wikibook to find it :-).

jordanbrown0, 2022-01-14:
Read the section starting at https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron and particularly the "Debugging polyhedra" section.

SimpleSafeName, 2022-01-15:
Not an OpenSCAD problem per se, but HD12_CFx_MGN9_YCarriage_LH_UpperBody_v1.1 from https://www.thingiverse.com/thing:4886459/files

was giving me the same error, but only when it intersected another part .

Here's the OpenSCAD code:

`$fn=64;

difference() {
  translate([0,0,0])
  import("HD12_CFx_MGN9_YCarriage_LH_UpperBody_v1.1.STL");

  translate([0,0,0])
  cube ([1,100,1]);
};`

Importing it into Autocad's MeshMixer, selecting "Analysis", then "Inspector", and then selecting "Auto Fix All" then export it to a STL and that gets rid of the error (for me).

Other curious things in OpenSCAD is when you import a "fixed" STL and then import one that isn't fixed the second one won't display anything. Individually the objects will display, but when they intersect the result will be nothing (and no errors).

Clearing the cache will stop this, and go back to showing the original error.

So my problem is "fixed", hopefully this will help someone else.

rwm, 2022-01-28:
I encountered the same error when the extruded 2d shape has an intersection a single point. I found this when working with SVG files, but this can also be reproduced with basic shapes. To me, this looks very much like a bug, because everything is working fine if the extruded 2d shape has an intersection in a line.

To reproduce:
```
linear_extrude(10){
        // two squares intersecting in a point
        square(10, 10);
        translate([10, 10, 0])
            square(10, 10);
} // ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 

translate([30, 0, 0])cube(10);
```

Whereas the following variations work fine:
```
linear_extrude(10){
        // two squares intersecting along line
        square(10, 10);
        translate([9, 10, 0])
            square(10, 10);
} // works fine

translate([30, 0, 0])cube(10);
```

```
linear_extrude(10){
        // two squares barely without intersection
        square(10, 10);
        translate([10.0001, 10, 0])
            square(10, 10);
} // works fine

translate([30, 0, 0])cube(10);
```

ntfshard, 2022-04-27:
Can be reproduced with `union` of two `import` calls for `stl` files, created by OpenSCAD.

jordanbrown0, 2022-04-27:
On 4/27/2022 1:24 PM, Maksim Derbasov wrote:
>
> Can be reproduced with |union| of two |import| calls for |stl| files,
> created by OpenSCAD.
>

There are many, many ways to get this class of error.

Roughly, they can be broken down into three sub-classes:

  * Polyhedra that are malformed - that have missing faces, faces
    oriented the wrong way, or self-intersections.
  * Polyhedra that have points that are too close together, that
    OpenSCAD snaps together when it forces everything onto its grid.
  * Models that are not 2-manifold, that have volumes that touch only
    along a single line.

Although the error messages involved are crummy, none of these are
considered bugs.  They are all considered to be limitations of the
geometry engines being used.

thehans, 2022-04-27:
@rwm the model is not 2-manifold.  One simple rule of that is: any edge of a 3d object must be shared by exactly two faces. In other words for a 2d object being extruded, all vertices must share exactly two edges.  That is why two squares which meet at a point are not valid, while those meeting along a line are fine.

ntfshard, 2022-04-28:
> On 4/27/2022 1:24 PM, Maksim Derbasov wrote: Can be reproduced with |union| of two |import| calls for |stl| files, created by OpenSCAD.
> There are many, many ways to get this class of error. Roughly, [...]

I see (more or less). Just reported one more observed case, in some way to show a severity of the problem (how often users can met it). My workaround was to use blender to join extruded parts.

jordanbrown0, 2022-04-28:
On 4/27/2022 5:14 PM, Maksim Derbasov wrote:
>
> I see (more or less). Just reported one more observed case, in some
> way to show a severity of the problem (how often users can met it).
>

Unfortunately, the answer to that question is well-known:  a lot.  And,
also unfortunately, mostly the only thing to do about it is for users to
learn how to avoid the problems.  I don't think there are any active
developers who are familiar with the deep magic that goes into the
rendering engine (CGAL) - it's imported from another open-source project.

(There *is* one check that I think might be possible for a mere mortal
to implement, that every edge in a polyhedron is visited exactly twice,
in opposite directions.  Now it's just a matter of somebody finding time
and interest...)

trivalik, 2022-08-26:
I run again in this issue, but just with difference from an STL file:
[image]

STL importt alone works fine with rendering. But if I try to remove a cube of it fails with error of this ticket.

jordanbrown0, 2022-08-26:
Yes.  OpenSCAD will let you import anything, or construct any polyhedron, and it will pass both F5 and F6, because a single object doesn't trigger the "smarter" processing that can detect errors.

As soon as you bring in another object, whether for union, difference, or whatever, the problem gets detected.

(I think it would be better if it got detected even when it's the only object in the model.)

trivalik, 2022-08-26:
Does that mean my STL file is broken? Or does that mean the import of STL will never work with difference and union?

jordanbrown0, 2022-08-26:
Yes, it means that your STL is broken.

The easiest thing to do inside OpenSCAD is to set it into View/Thrown Together mode, and look for purple.  Everything should be yellow - any purple, no matter how small, is a problem.

The problem could be a gap in the model, so that it does not form a complete closed surface.  It could also be a face that has its vertexes reversed.  Both cases are visible (though perhaps from difficult-to-reach points of view) as purple.

There are online tools that can help to repair broken STLs, but I am not familiar with them.

If you supply the STL, others may be able to help to diagnose it.

trivalik, 2022-08-26:
I used a online repair tool and it found something, but still same result.

Code:
```
difference() {
import("C:/.../Proximity_Sensor_D18_Holder_fixed.stl", convexity=3);
translate([0,-100,-31/*nach unten*/])cube([100,120,100]);    
}
```

STL: Proximity_Sensor_D18_Holder.stl from https://www.thingiverse.com/thing:2001681/files

jordanbrown0, 2022-08-26:
I duplicate the error.

I don't see anything wrong with the model, but I am far from an expert at finding those problems.

nophead, 2022-08-26:
Netfabb says it has self intersections here:

[image]

The free version won't fix them.

trivalik, 2022-08-27:
I got a fixed model from online page https://formware.co/OnlineStlRepair but still same result.

MichaelAtOz, 2022-08-27:
@trivalik upload the "fixed" STL, I'll have a look.
But if you get that error, 99.905% bad STL.

trivalik, 2022-08-27:
here the fixed STL file:
[Proximity_Sensor_D18_Holder_fixed.zip](https://github.com/openscad/openscad/files/9437765/Proximity_Sensor_D18_Holder_fixed.zip)

nophead, 2022-08-27:
Still has self intersections according to NetFabb.

[image]

I don't think free repair tools fix those because it is hard, so they are all behind paywalls.

trivalik, 2022-08-27:
Thanks, meshmixer solved it somehow in "Analysis". The weird thing is that the ascii STL export shows some at F12 the pink color. 
But render with F6 works now, thanks for the big support.

[Proximity_Sensor_D18_Holder_MeshMixed_fixed_Ascii.zip](https://github.com/openscad/openscad/files/9437856/Proximity_Sensor_D18_Holder_MeshMixed_fixed_Ascii.zip)

[image]

MichaelAtOz, 2022-08-27:
As Nophead said, it was not totally fixed.

It did fix a couple of holes and a degenerate face.

But two areas of self-intersections persist.

 

> Thats why I ask whether difference form STL is possible or not. 

 

So, no, not with an invalid STL.

 

You can of course print it, and difference(mode=hacksaw);

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2022-08-27:
Pink = bad.

Pink is inverted triangles, ie the triangles inside face is inverted and pointing outside the solid.

 

If yuo just want to print it via a slicer to a 3D printer, it MAY work.

If you want to use that in another OpenSCAD operation it will likely fail.

 

  _____  

 

 <https://user-images.githubusercontent.com/3148279/187028132-e27cc055-a4b9-4cba-a874-774acabff62c.png> image

 

https://www.avg.com

MichaelAtOz, 2022-08-27:
> But render with F6 works now

 

If you add a cube(1); it will likely fail.

If you just F5/F6 the import() without another operation, like difference(), CGAL does not get called, so you don't get the CGAL error.

 

  _____  

 

 

https://www.avg.com

trivalik, 2022-08-27:
It does not fail. That's why I reported it. Only the ascii version of STL has it, the binary version has no pink!

MichaelAtOz, 2022-08-27:
Bed time now. 
I need to copy stuff to another box to get the thrown-together view (due to VMs). Tomorrow.
But.
```
// When working on CGAL Errors, 
// you must do Design/Flush-cache to bypass cache & force a re-evaluation.
import("Proximity_Sensor_D18_Holder_fixed.stl", convexity=8); // 3 was no enough
cube(1);  // this forces a union(), thus CGAL MUST process the mesh.
```
If you just have one object (ie `import()`) you don't get the error.
The STL is bad.

[I think we need to have CGAL errors clear the cache, for that branch]

trivalik, 2022-08-27:
I restarted OpenScad to be sure, so cache should not be the cause. I tried your code and it works also well. The original file from thingiverse fails.

ninp0, 2023-03-08:
I found this useful - hopefully others will as well:

https://www.formware.co/onlinestlrepair

tpchuckles, 2023-08-11:
> Can be reproduced with `union` of two `import` calls for `stl` files, created by OpenSCAD. (for 2021.01 version)

I have this same issue, which I would definitely consider a "bug" (despite arguments elsewhere in this thread). 

The first geometry is exported _with no errors, from OpenSCAD itself_:

`openscad -o petal.stl waterlily.scad 
Geometries in cache: 66
Geometry cache size in bytes: 24596784
CGAL Polyhedrons in cache: 3
CGAL cache size in bytes: 102254960
Total rendering time: 1:19:20.616
   Top level object is a 3D object:
   Simple:        yes
   Vertices:    31030
   Halfedges:  115648
   Edges:       57824
   Halffacets:  54296
   Facets:      27148
   Volumes:         2
`

but importing that stl and booleaning it gives the error. 

**either OpenSCAD export is broken, or OpenSCAD import is broken**. either way, something is broken. OpenSCAD should be exporting "correct" STLs (or telling you if it fails to do so), and importing "correct" STLs without error. 

here is the scad file (file suffix changed to .txt so i can upload to github). uncomment line 5 and comment out 8-14 to export petal.stl. then recomment 5 and uncomment 8-14 to see this error. a quick comment on "why": for a highly complicated geometry with a boat-load of objects, it's faster to compile once and then read the STLs, at least for being able to have a usable preview. 
[waterlily.txt](https://github.com/openscad/openscad/files/12323261/waterlily.txt)

nophead, 2023-08-11:
GCAL can calculate geometry with infinite precision using rational numbers. When it is exported to STL it gets snapped to a grid and converted to floats and that will cause self intersections if the vertices are closer together than the grid, so it will be invalid when imported.

Not much you can do other than reducing the resolution or repairing the STL with a third party tool that will fix intersections.

tpchuckles, 2023-08-11:
> GCAL can calculate geometry with infinite precision using rational numbers. When it is exported to STL it gets snapped to a grid and converted to floats and that will cause self intersections if the [...]

this still means openscad is generating flawed STLs despite claiming to have been successful. does the export need some sort of (additional) checker to see if it created a valid file? ([documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Import_and_Export#Export) covers the "simple = no" output error, but it is apparently still possible for "simple = yes" to still be "broken").

nophead, 2023-08-11:
Yes the STL export is broken and a lot more of OpenSCAD as well because it caches meshes that have been snapped to a grid or just converted from rational to double or double to float. You simply can't ever reduce the precision of the vertices without having a full mesh repair tool to follow that because it can easily create degenerate triangles (which are easy to repair) and self intersection, which are harder to repair.

I can't even generate a single petal on my PC with your code as I haven't got enough memory. I can preview it and it looks incredibly high res and perhaps has knife edges which tend to be where self intersections occur. In general whenever there is an intersection or union of curved surfaces you can't guarantee the vertices are not arbitrarily close. Even a simple pipe joint might create an STL that is broken if you are unlucky.

I don't often encounter it because I design functional objects for FDM printing, so my resolution is much less and so there is a much lower chance of two vertices being too close but the grid does bite me occasionally.

pca006132, 2023-11-24:
The problem with STL is not because of lose of precision. It is just because there is no way of storing the required topology information on STL. It is recommended to use something like 3MF, although we currently don't get the topology information from it either (needs indexed polygon support in PolySet). I just opened a PR that may accept some of these STLs with manifold enabled.

pca006132, 2023-11-24:
@tpchuckles tested with manifold and #4855, it works now and renders in 5s.
[image]

pca006132, 2023-11-29:
@kintel should we close this or leave this open? This is still an issue for the CGAL backend.

kintel, 2023-11-29:
Perhaps close as duplicate of #791 ?
..unless this is actually separate.

pca006132, 2023-11-29:
I think this is different, the base polygon does not seem to be self-intersecting.

kintel, 2023-11-29:
I'm a bit confused now: The original example in this issue is still failing because of the scale to zero.

pca006132, 2023-11-29:
I forgot if I tried that, need to check later today.

kintel, 2023-11-29:
@pca006132 That's odd. This is what I get for latest master:
```
$ ./openscad ../issue4039.scad -o out.stl --enable=manifold
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold
Current top level object is empty.

```

gsohler, 2023-11-29:
isn't that a test included in the ctests ?
could that be the result of reindexing too often ?

kintel, 2023-11-29:
This test isn't (yet) in our test framework. It's just the reported issue saved to a local file.

pca006132, 2023-11-29:
I got the same issue when running with the command line. Probably something wrong due to different code path being taken. Will try to fix that.

pca006132, 2023-11-29:
I looked at the stl and it seems that the extrude function produces boundary edges that are not repairable.
I tried to do this directly with manifold, but the linear extrude behaves differently in manifold so this is not directly implementable there. Probably need some tweaking.

pca006132, 2023-11-30:
OK, it seems that the reason is the scale to 0 makes some degenerate polygon, which confuses CGAL and generated a mesh that is not closed and not repairable with the polygon soup repair function.
I tried generating it using modified manifold (because manifold linear extrude is slightly different from that in openscad), the result is indeed manifold and the exported stl can be imported back to openscad as closed mesh (though there is no guarantee). This should be easy to fix as we are now using indexed PolySet.

kintel, 2023-11-30:
This test looks similar to https://github.com/openscad/openscad/blob/master/tests/data/scad/3D/features/linear_extrude-scale-zero-tests.scad

We should add to that, or another, test case for future regression testing.

kintel, 2024-04-01:
Closing this as a duplicate of #5046 - added this as an example to make sure we test it.

FYI: All the other reported issues here now works when using the Manifold backend.

closed 2024-04-01 by kintel

tn-muc, 2025-02-20:
I get same errors as in the thread opening with various designs and it's always a long way to find a work around.
Here's a most simplistic example and I'd be deeply grateful, if anyone could explain the reason and how to repair!
`polyhedron([[0,0,0],[0,1,0],[1,1,0],[1,0,0],[0,0,1],[0,1,1]],
			[[0,1,2,3],[0,1,5,4],[3,4,5,2],[0,4,3],[1,2,5]], 9);`

tn-muc, 2025-02-20:
oops! I was sure that I tried this - and am more than surprised that it indeed shows the error!
Anyway, a 1000 thanks for you pointing at it again.
btw: any chance to set the "thrown-together view" as default? Don't find it in preferences.

tn-muc, 2025-02-20:
While that example's been embarrassingly simple, here's a more restive one:
`
PHI=(1+sqrt(5))/2;
P=[ [0,1,PHI],[0,1,-PHI],[0,-1,-PHI],[0,-1,PHI],
	[1,PHI,0],[1,-PHI,0],[-1,-PHI,0],[-1,PHI,0],
	[PHI,0,1],[-PHI,0,1],[-PHI,0,-1],[PHI,0,-1] ];
F=[ [9,0,3],[6,9,3],[5,6,3],[8,5,3],[0,8,3],[1,10,2],[1,2,11],[1,11,4],[1,4,7],[1,7,10],
	[0,7,4],[0,4,8],[8,4,11],[8,11,5],[5,11,2],[5,2,6],[6,2,10],[6,10,9],[9,10,7],[9,7,0] ];
polyhedron(P,F);

`
Same question(s).

kintel, 2025-02-20:
The last `,9` convexity setting should be pretty useless for a convex object like this.

tn-muc, 2025-02-20:
@ kintel: absolutely, it _should_.

[8 more replies]