openscad/openscad #4045: Feature request: Start external editor from TabManager.cc

mchendriks, 2022-01-12:
I would like to have a conbfigurable option to start an external editor like gvim.

I added the following to  TabManager::createTab() in tabmanager.cc and this works, 

/* call external editor gvim*/
char cmd[4000];
QByteArray array = filename.toLocal8Bit();
char* buffer=array.data();
sprintf(cmd,"gvim --remote-silent-tab %s",buffer);
system(cmd);

It would be nice if cmd would be a preference setting and this would be part of the official source code.

t-paul, 2022-01-12:
Why mix using both the OpenSCAD internal editor and the external one? What's the workflow for this?

mchendriks, 2022-01-13:
> Android [4].

I can think of the following reasons for using an external editor:

1. An editor like for instance (g)vim has lot's of more functionality 
like: working with multiple buffers, macro's, syntax highlighting, 
autocomplete with your own dictionary, so you can add your own 
functions. And so on. As a programmer you can be more productive.

2. Personal preference, as a programmer in multiple languages you get 
used to certain editors. Switching between different editors is hard.

3. openscad is already supporting the use of an external editor, if you 
change the file outside openscad, the preview is automatically 
generated. But you have to open the file twice, in openscad and in your 
preferred editor

4. Supporting an external editor is code-wise very easy to do. Just make 
a system call to start up the editor with the corresponding filename, 
within Tabmanager.cc. And one preference setting as a string and you are 
done.

/* working code example */

/* call external editor (gvim) in TabManager::createTab() */
char cmd[4000];                             /* string to hold the system 
command *.
QByteArray array = filename.toLocal8Bit();  /* get the filename */
char* buffer=array.data();                  /* convert filename to 
string */
sprintf(cmd,"gvim --remote-silent-tab %s",buffer);  /* create system 
command, this should be a preference/option string if empty don't 
start*/
system(cmd);                                         /* start the editor 
*/

Greetings Matthieu

-------------------------

Links:
------
[1] 
https://github.com/openscad/openscad/issues/4045#issuecomment-1011400315
[2] 
https://github.com/
[3] 
[4]

t-paul, 2022-01-13:
I'm not asking why you would want to use an external editor. As you already mentioned, that's an officially supported and documented feature.

I don't see how force-opening tabs always in an external editor works. That means you can't even open tabs in the application anymore without going back to preferences.

mchendriks, 2022-01-13:
I'm not sure (due to my bad English) if I completely understand what you 
are saying :(

I downloaded the complete sourcecode of openscad on Ubuntu. added the 
sourcecode lines to Tabmanager.cc compiled it and it opens the external 
editor just fine, also opens the internal editor with the different 
tabs. Otherwise you can't swith between different tabs voor preview.

But I think this is not the answer on your statement, so help me out 
here please :)

Links:
------
[1] 
https://github.com/openscad/openscad/issues/4045#issuecomment-1012053253
[2] 
https://github.com/
[3] 
[4]

t-paul, 2022-03-27:
It's still completely unclear to me how that is supposed to work. If you want to use an external editor, that's fine and supported as explained in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_an_external_Editor_with_OpenSCAD

closed 2022-03-27 by t-paul