openscad/openscad #4046: Feature Request: Possibility to specify additional search paths within the scad file.

lucky62, 2022-01-13:
As I know there is only one way to specify libraries locations - OPENSCADPATH env variable.

This solution is not enough flexible - users (including me) are looking for other ways...
See: #2384, #2377, #604, etc. 

From my point of view - it will be enough to implement simple directive to specify additional locations - e.g.:

`include_path /path/to/libs:/another/path`

The specified paths can be prepended (appended?) to the paths already specified by OPENSCADPATH - if any.
The paths can be absolute or relative to the project file.
Absolute paths can be added without changes.
Relative paths will be internally converted to the absolute paths by prepending the path by the current file location.
For example - file location is:
`/OpenSCAD_Projects/Project1/project1.scad`
and this file s containing the directive:
`include_path project_libs`
then the following path will be added:
`/OpenSCAD_Projects/Project1/project_libs`

I think this can be quite simple to implement and enough flexible to use...
What do you think?

t-paul, 2022-01-13:
Allowing scripts to manipulate the include path can cause confusing and evaluation order dependend problems. I don't see how to implement that in a stable & safe way.

What is the specific problem that this should solve?

lucky62, 2022-01-13:
There is a lot of cases when this feature will be useful:
- when you need to use project specific library (which you don't want to put into common library location)
- when the OPENSCADPATH is defined in /etc/profile and you have no privileges to change it nor to put the library to common location
- when you would like to test the new version of library (overriding older version)
- etc.

BTW what can be different when include path will be specified by include_path directive instead of OPENSCADPATH env variable? 
In both cases the path can be modified by user, but directive can simplify things. No need to modify other files (user or system profiles), no reboot...

Current behavior is:
> OpenSCAD looks for the file in the following directories in the following order:
>-    the directory of the calling .scad file
>-    the User-Defined library paths (OPENSCADPATH)
>-    the Built-In library (i.e. the O/S dependent locations above)
>-    the Installation library

New bahavior can be:
> OpenSCAD looks for the file in the following directories in the following order:
>-    the directory of the calling .scad file
>-    the User-Defined library paths (**include_path directives**)
>-    the User-Defined library paths (OPENSCADPATH)
>-    the Built-In library (i.e. the O/S dependent locations above)
>-    the Installation library

I think there is no difference when path is specified by OPENSCADPATH or by directive.
I cannot imagine any new problem caused by this.

jordanbrown0, 2022-01-13:
Do these produce the same result?
```
include_path /foo
include <bar.scad>
```

```
include <bar.scad>
include_path /foo
```

--

Does `include_path` affect the behavior of files included, or only the behavior of the file it is found in?  What about files used?

--

Note that in most system configurations /etc/profile is executed before personal shell startup scripts - $HOME/.profile, $HOME/.bashrc, et cetera - and so those personal startup scripts can be used to override the system setting of $OPENSCADPATH.  And of course you can always set it interactively before starting OpenSCAD.

For a project-specific library, normally a path relative to the .scad file is exactly what you want.

lucky62, 2022-01-13:
> Do these produce the same result?

Depends on how the scad file is processed. I don't know how the parser is working.
But it is not really important if the behavior will be documented.

> Does `include_path` affect the behavior of files included, or only the behavior of the file it is found in? What about files used?

The same as for previous question - depend on file processing/parsing.

> Note that in most system configurations /etc/profile is executed before personal shell startup scripts - $HOME/.profile, $HOME/.bashrc, et cetera - and so those personal startup scripts can be used [...]

It is much easier to put the directive into file which you are editing than to edit the hidden files...

closed 2026-05-10 by github-actions[bot]