openscad/openscad #4047: Highlighting  with  some characters

UBaer21, 2022-01-14:
when a comment or string contain certain character the highlight is moved 

[image]

```
a="äööaa";
// comment ö
// comment
```

jordanbrown0, 2022-01-14:
Something is counting bytes instead of characters.  ä and ö are two bytes each.

Similar to #3939.

ChrisCoxArt, 2022-01-14:
yep, working on this now. Seems that a conversion to QString was unnecessary and counted glyphs instead of bytes, but qscintilla wants bytes.

ChrisCoxArt, 2022-01-15:
testing the fix now.

ChrisCoxArt, 2022-01-15:
The fix seems solid across many languages. I tried UDHR translations for samples, plus emoji, plus random glyphs.
pull request submitted, but it'll take time to pass all the checks.

t-paul, 2022-01-15:
Fixed by #4048.

closed 2022-01-15 by t-paul