openscad/openscad #4053: Viewport controls

psteiner, 2022-01-16:
Provide UI controls to set or constrain viewport parameters, for example to rotate the z-axis to a specific angle, without altering the angles of other axes. This is difficult to achieve when click-dragging in the viewport with a mouse.

Constraints could be implemented by modifier keys, e.g. holding control, alt, shift, etc. while dragging to constrain movement to a single axis, or by providing fields to input values, e.g. clicking on a viewport value transforms it into an field for keyboard input.

UBaer21, 2022-01-16:
So besides [$vpr](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#$vpr,_$vpt,_$vpf_and_$vpd)
and customizer use?
- maybe if we move away from the fixed buttons to an interactive gizmo for viewport control - (like the view cube used by autodesk)

psteiner, 2022-01-16:
Yes, $vpr does what I want! I didn't know about that one, thanks!

closed 2022-01-16 by psteiner