openscad/openscad #4054: Panic on latest beta (OSX) after a few minutes of usage

salamanders, 2022-01-16:
I've been getting a full OS crash about 5 minutes after starting OpenSCAD.  AFAIK it only happens when OpenSCAD is open, so I don't think it is my computer finally giving up.  (hopefully).  I did update to the latest OS not long ago.

No other apps open.

- OpenScad 2022.01.16
- OSX 10.15.7
- Macbook Pro (2012)
- [panic.txt](https://github.com/openscad/openscad/files/7877706/panic.txt)

Anyone else with similar crashes?

kintel, 2022-01-16:
Ouch, full OS crashes are never fun to deal with. This looks like a GPU driver crash.

If you are able to reproduce this relatively consistently, it would be awesome if you could give some of the older snapshots a spin to see if this behavior was introduced in OpenSCAD itself, or emerged from the OS. Past snapshots are available at https://files.openscad.org/snapshots/

ChrisCoxArt, 2022-01-16:
Yeah, it's the GPU that panic'd, and it does look like the driver generated bad addresses or code for the GPU.
You can file this with Apple, but don't expect them to fix it.  Apple has been dropping NVidia support, and NVidia no longer has a team working on drivers for MacOS (after Apple cut them off).

kintel, 2022-01-16:
Yeah, pretty low chance to get a driver for a 2012 GPU fixed : (
..but in the off chance that we introduced a bug where we end up doing something funky and unnecessary, it's always a possibility that we can find and fix it :)

PS! I'm not sure which GPU this is, but I had an old Mac from around that time where the GPU had some soldering issues which caused random crashes/freezes. I had to pull out the motherboard and bake it in the oven for 15 minutes or so, which prolonged the lifetime of that laptop with a couple of years.

salamanders, 2022-01-17:
I did more reading - YUP, totally a 2012 Mac GPU issue.  And I'm guessing one of the few times I use the "better" GPU is OpenSCAD.  So - hardware fault on my end, please close!

closed 2022-01-17 by salamanders