openscad/openscad #4058: Openscad.exe infected

mcv93us, 2022-01-20:
I wanted to test the 64-bit development version for Windows (version 2022.01.19), I downloaded the zipped package and after unzipping it my Avast antivirus quarantined the openscad.exe file notifying me that it was infected with Win32: Miner- HC [PUP].
The version of Avast Free Antivirus is 21.11.2500 (build 21.11.6809.528).
The version of the virus definition is 220120-4.
My system is Windows 10 64bit version 21H2 build 19044.1469
I currently use Openscad in version 2021.01 without any problems.

t-paul, 2022-01-20:
It's not infected:
https://www.virustotal.com/gui/url/6bf0b6e51791ee4d4ec76ef24e7a4e627783be10cbda7bdd5d6c0fcd5fe69a32?nocache=1

closed 2022-01-20 by t-paul

t-paul, 2022-01-20:
Uploaded the exe separately and that looks more like an issue with the scanner:
https://www.virustotal.com/gui/file/2e187aedaccc198ef379495ff8a1b26580dc5eabce91d171babb1404d7e83dcd?nocache=1

MichaelAtOz, 2022-01-20:
Most times when I download a new dev version AVG flags it as it has not come across that exe before, it sends it off for further analysis, disables it until it gets an answer. Once in a while it says it has one of the 'generic' viruses, but not for long. I've never has an actual infected file.