openscad/openscad #4059: open .scad first time --> need to confirm "access to Documents" for every Preset

3Dietrich, 2022-01-20:
OpenSCAD Version 2022.01.20:
macOS 12.0.1, MBAir 2020 M1

Every time I open a file.scad with presets the first time, I need to confirm for every preset(!) to get access to documents

[image]

[see in this screenshot video](https://user-images.githubusercontent.com/4608796/150386570-00562e90-5114-41ce-8bba-2cc7df87f663.mov)

ChrisCoxArt, 2022-01-23:
I haven't been able to reproduce this on MacOS 12.1, Intel processor.
I've tried removing permissions for OpenSCAD, installing new dev snapshots, and making new development builds.  But I only get prompted for the document directory of files I open, never presets.

ChrisCoxArt, 2022-01-23:
OK, I now have 12.1 installed on my M1 machine, and still cannot reproduce it requesting permission for every preset.
Where did you install/copy the OpenSCAD.app bundle to run it?

3Dietrich, 2022-01-23:
Yea, I can't reproduce it any more, it seems it was **not** because of the presets. 

But maybe for several .scad file in the Download folder? I come to this idea because in this screenshot video the files are at the start greyed out (before I press ok the first time) and become 'accessible' (full colored) one by one each time I pressed "ok":

https://user-images.githubusercontent.com/4608796/150676763-7a47573b-4973-4a50-b44b-df3f84e612d6.mov

in any case, _any_ question about "download [bla] ?" is totally superfluous because I allowed it long ago:
[image]

eaton, 2022-03-03:
Just dropped in the latest dev build and I'm seeing the same issue: Even though I've given OpenSCAD full disk access, I get "OpenSCAD wants to access…" prompts dozens of times. In particular, it's asking for access to the _Documents_ folder where libraries are stored, so I suspect it might have something to do with that. I'll see if removing a bunch of the custom libraries I've installed helps.

ChrisCoxArt, 2022-03-03:
And I still cannot reproduce that. I suspect something is very wrong with the permissions or security settings on your system.

DavidPhillipOster, 2022-03-15:
cannot reproduce

OpenSCAD version 2022.03.14 (git 9f68ea0a6).  I removed any previous OpenSCAD, and in before first run removed OpenSCAD  from **SystemPreferences >Securoty&Privacy>Files&Folders** After a restart, I opened, rendered, and edited a file that used a [file](https://media.prusaprinters.org/media/prints/86604/stls/916969_13504cb4-efce-4d93-a60d-ed89d29be548/hex-grid.scad#_ga=2.230717724.2038053186.1639416566-1034880233.1639163113) that #included a BOSL2 file. Although on first launch, I did get the **"OpenSCAD” cannot be opened because the developer cannot be verified.** alert, I never got any complaints about access to any folders. macOS Monterery version 12.3, M1 Mac Mini.

tcurdt, 2022-03-23:
I am seeing the same thing.

It seems a little suspicious that there are no entitlements at all

`codesign --display --entitlements - /Applications/OpenSCAD.app`

Some that could be relevant:

```
com.apple.security.app-sandbox
com.apple.security.files.user-selected.read-write
com.apple.security.app-protection
com.apple.security.documents.user-selected.read-write
com.apple.security.temporary-exception.files.home-relative-path.read-write
```

The strange thing is - even adding OpenSCAD to full disk access it keeps asking.

I still have a pending macOS upgrade and see if that makes a difference.

eaton, 2022-03-23:
@tcurdt, are you by any chance storing your Desktop and Documents folders in iCloud? It just occurred to me that I'm doing that on both machines I tested with; by default, that also places OpenSCAD's Libraries folder in an iCloud controlled directory as well.

tcurdt, 2022-03-23:
@eaton No, I am not using iCloud (more than I have to).
And my OpenSCAD library even resides outside of the Documents folder. (via `OPENSCADPATH`)
So that can't be it.

tcurdt, 2022-03-24:
If OpenSCAD does not plan on releasing through the MAS, maybe the entitlement

```
<key>com.apple.security.app-sandbox</key>
<false/>
```

would save us from all the headaches?

Within the sandbox it probably also needs full disk permissions to make use of `OPENSCADPATH`.

tcurdt, 2022-03-24:
Interesting - and weird! It happens only when the file is on the desktop (or a sub dir thereof). When I move it somewhere else everything just works. When I move it back to the desktop I again get the permission popups.

thehans, 2022-03-26:
Just a shot in the dark, are any of the paths with problems containing symbolic links?

tcurdt, 2022-03-26:
@thehans Na, the path to the desktop doesn't contain any links.

eaton, 2022-04-17:
Just had a chance to set up a fresh new Mac _and_ re-image the MacBook where I initially spotted the issue; it's definitely still present.

On both freshly-imaged machines, before and after enabling iCloud Desktop/Document sharing, I downloaded OpenSCAD-2022.04.15.dmg from the site and installed it.

* On launch, it requests access to the Documents folder.
* On saving a file to the Desktop, it requests access to the Desktop (predictably). Saving the file again doesn't re-trigger the prompt, but other operations — rendering, for example — do trigger requests to access both the Desktop and the Documents folder.
* Opening existing .scad files shows the same behavior, and the permission appear to pop up multiple times if the files use `include`, even when the included files are in folders that were previously 'prompted' for.
* Files that use the Customizer seem to do it more (I lost count after clicking to grant permissions to my Documents folder upwards of 30-40 times).
* This behavior is present even when OpenSCAD is granted full disk access in System Preferences.

Other than this particular issue, the new dev builds are BLISTERINGLY fast on the M1. When I have a chance next week I'm going to dig in and do some more specific controlled tests around individual operations to see if there are any other patterns.

tkircher, 2022-05-08:
I'm having this same issue. OpenSCAD version 2022.04.09 (git b13ec70a4), MacOS Monterey 12.3.1. It doesn't matter if I have a .scad file on the desktop, in a folder on the desktop, or in the documents folder. It asks me for permission every single time. When opening the documents folder, it sometimes asks thirty or forty times in a row. Clean system install. A mitigation strategy is to put files in your home directory only, which at least works for me. There does not appear to be a way to turn off privacy protections on MacOS anymore.

tcurdt, 2022-05-08:
@tkircher Does it also ask for permission when in a subfolder of home?
As long as I stay away from the desktop it works OK. I haven't tried the documents folder though.

tkircher, 2022-05-08:
@tcurdt A subdirectory of the home directory also works fine. So Desktop and Documents are basically unusable directories in MacOS.

tcurdt, 2022-05-08:
@tkircher I just tried the documents folder. I can confirm the same.
So yes, Desktop and Documents are causing problems.

fergu, 2022-05-12:
I can also confirm this bug. I can also confirm that moving the relevant `*.scad` files out of either Documents or Desktop does clear up the warnings when building the model, though you will still get asked for permissions every time you go through file->Open. Interestingly, I also note that I get asked for Desktop and Documents permissions twice each time (So 4x total), though sometimes I get asked for one of the two twice, and the other only once.

This bug does not appear when using the current release download (2021.01).

Maybe this might be related to code signing on Apple Silicon? https://developer.apple.com/forums/thread/125794

I don't know enough about OpenSCAD's internals, but maybe something isn't quite right with code signing and that's causing the security measures to freak out by thinking the binary is constantly changing? I wonder if this issue with the nightly build also exists on Intel Macs, and whether it can be cleared up by stripping the code signature if it does. Unfortunately unsigned ARM binaries will refuse to run, so I can't test that idea on an M1 directly.

ChrisCoxArt, 2022-05-12:
The problem is: this only happens to small subset of users.
There is something different about your systems, or where/how you copied the application.

tcurdt, 2022-05-12:
> The problem is: this only happens to small subset of users. There is something different about your systems, or where/how you copied the application.

Probably - but for me it happened on a fresh system.
I got the new M1 laptop and during the setup also installed OpenSCAD.
So I am really wondering how the system could be different.

fergu, 2022-05-12:
Mine isn't fresh, but I installed it to the normal `/Applications/` directory. I've also tried deleting the permissions for OpenSCAD from `Settings -> Privacy` to force it to ask again with no luck.

I am using iCloud while @tcurdt (I assume based on earlier posts) is not, so it seems like it's likely not an iCloud issue.

The common thread between the two of us seems to be that we're both on M1 Macs, so I'm tempted to think this is something specific to that. The only thing I've found that's different is that MacOS running on M1's is more strict about code signing than on Intel. 

Is there maybe a way to get an intel only build so we can see if running under Rosetta clears this up? That would at least narrow it down to an M1 issue.

tkircher, 2022-05-12:
Clean system here, M1 Max, Monterey 12.3.1, official ARM build of OpenSCAD downloaded from the website, development snapshot 2022.04.09 (git b13ec70a4)

tcurdt, 2022-05-12:
> I am using iCloud while @tcurdt (I assume based on earlier posts) is not, so it seems like it's likely not an iCloud issue.

Not using iCloud. So that doesn't seem to be related.

> The only thing I've found that's different is that MacOS running on M1's is more strict about code signing than on Intel (Edit: Though I'm admittedly unsure why that'd affect permissions requests)

Do you have more info about that? I am not aware of the "more strict" part.

(Was a) clean system, M1 Pro, Monterey 12.3.1, still on ARM build 2022.03.16 (git a6392b3ec)

tkircher, 2022-05-12:
@tcurdt It is correct that M1 macs have a more tightly locked down system than Intel macs.

fergu, 2022-05-12:
I've been testing on ARM (universal) 2022.05.09 (The "nightly" build available from the website). M1 Air, Monterey 12.3.1.

> Do you have more info about that? I am not aware of the "more strict" part.

[This Apple Developer release note](https://developer.apple.com/documentation/macos-release-notes/macos-big-sur-11_0_1-universal-apps-release-notes) mentions that Apple Silicon systems will refuse to run unsigned code, but that Xcode will sign the code as part of the linking process. That's enough to allow the code to run, but Gatekeeper will still flag it. (You'd need to go through the whole Apple Developer registration process thing to get around that which isn't necessary)

They also mention some caveats where the signing process may not work correctly. That said, I'd expect that kind of issue to prevent OpenSCAD from running, not to cause this weird permissions issue. So Code Signing itself might not be the problem, but maybe something adjacent to how M1 systems are more strict on security policies?

ChrisCoxArt, 2022-05-12:
I also have an M1 iMac (in addition to a few Intel systems), and no such problems.
There is still something odd happening with a subset of systems and users.

fergu, 2022-05-12:
I decided to pull on this thread a bit harder. It seems like something may be funky with how the build is being signed, and maybe that's related to these issues.

Trying to verify the code signature of the OpenSCAD bundle (I have the nightly build on my Desktop since I re-installed the stable build to `/Applications` for now)

```
fergu@cumulus % codesign --verify --deep --verbose /Users/fergu/Desktop/OpenSCAD.app
/Users/fergu/Desktop/OpenSCAD.app: code object is not signed at all
```

Picking a random other (non-App store) app that is also universal:

```
fergu@cumulus % codesign --verify --deep --verbose /Applications/HandBrake.app
/Applications/HandBrake.app: valid on disk
/Applications/HandBrake.app: satisfies its Designated Requirement
```

The codesign utility can print out SCAD's signature which shows it is signed, but looks like maybe not quite right in order to make MacOS security policies happy:

```
fergu@cumulus % codesign --verify --deep --display --verbose /Users/fergu/Desktop/OpenSCAD.app 
Executable=/Users/fergu/Desktop/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier=OpenSCAD
Format=app bundle with Mach-O universal (x86_64 arm64)
CodeDirectory v=20400 size=139457 flags=0x20002(adhoc,linker-signed) hashes=4355+0 location=embedded
Signature=adhoc
Info.plist=not bound
TeamIdentifier=not set
Sealed Resources=none
Internal requirements=none
```

Doing the same thing with HandBrake from before:

```
fergu@cumulus % codesign --verify --deep --display --verbose /Applications/HandBrake.app      
Executable=/Applications/HandBrake.app/Contents/MacOS/HandBrake
Identifier=fr.handbrake.HandBrake
Format=app bundle with Mach-O universal (x86_64 arm64)
CodeDirectory v=20500 size=8194 flags=0x10000(runtime) hashes=245+7 location=embedded
Signature size=9011
Timestamp=Oct 3, 2021 at 07:10:41
Info.plist entries=33
TeamIdentifier=5X9DE89KYV
Runtime Version=11.3.0
Sealed Resources version=2 rules=13 files=233
Nested=XPCServices/HandBrakeXPCService2.xpc
Nested=XPCServices/org.sparkle-project.InstallerLauncher.xpc
Nested=Frameworks/HandBrakeKit.framework
Nested=XPCServices/HandBrakeXPCService3.xpc
Nested=XPCServices/HandBrakeXPCService.xpc
Nested=XPCServices/HandBrakeXPCService4.xpc
Nested=XPCServices/org.sparkle-project.InstallerConnection.xpc
Nested=Frameworks/Sparkle.framework
Nested=XPCServices/org.sparkle-project.Downloader.xpc
Nested=XPCServices/org.sparkle-project.InstallerStatus.xpc
Internal requirements count=1 size=184
```

fergu, 2022-05-12:
Okay, looks like this is definitely a code signing issue. I noticed in the Apple developer docs that `If you use a custom workflow involving tools that modify a binary after linking (e.g. strip or install_name_tool) you might need to manually call codesign(1) as an additional build phase to properly ad-hoc sign your binary.` - so I started to wonder if that might be the issue.

I created a self-signed code signing identity called `SCAD Test` as outlined by Apple [here](https://developer.apple.com/library/archive/documentation/Security/Conceptual/CodeSigningGuide/Procedures/Procedures.html) to let me sign binaries myself. (This is an archive page that I found in googling - I'm sure there's something more recent now)

I then re-signed the OpenSCAD bundle with my identity, with the `--deep` flag to indicate that all the resources should be recursively signed as well.

```
fergu@cumulus % codesign -s "SCAD Test" --deep --verbose OpenSCAD.app
OpenSCAD.app: signed app bundle with Mach-O universal (x86_64 arm64) [org.openscad.OpenSCAD]
```

Then doing the same verification as above:

```
fergu@cumulus % codesign --verify --deep --display --verbose /Users/fergu/Desktop/OpenSCAD.app        
Executable=/Users/fergu/Desktop/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier=org.openscad.OpenSCAD
Format=app bundle with Mach-O universal (x86_64 arm64)
CodeDirectory v=20400 size=139566 flags=0x0(none) hashes=4355+3 location=embedded
Signature size=1749
Signed Time=May 12, 2022 at 14:51:43
Info.plist entries=15
TeamIdentifier=not set
Sealed Resources version=2 rules=13 files=255
Nested=PlugIns/platforms/libqcocoa.dylib
Nested=PlugIns/imageformats/libqgif.dylib
Nested=Frameworks/libzip.dylib
Nested=PlugIns/imageformats/libqjpeg.dylib
Nested=Frameworks/libgmp.dylib
Nested=Frameworks/QtDBus.framework
Nested=Frameworks/lib3MF.1.dylib
Nested=Frameworks/libfontconfig.dylib
Nested=Frameworks/libgmpxx.dylib
Nested=Frameworks/QtMacExtras.framework
Nested=Frameworks/libuuid.dylib
Nested=Frameworks/QtMultimedia.framework
Nested=PlugIns/styles/libqmacstyle.dylib
Nested=Frameworks/libharfbuzz.dylib
Nested=PlugIns/bearer/libqgenericbearer.dylib
Nested=Frameworks/libboost_system.dylib
Nested=Frameworks/QtGui.framework
Nested=Frameworks/QtGamepad.framework
Nested=Frameworks/libcairo.dylib
Nested=PlugIns/mediaservice/libqtmedia_audioengine.dylib
Nested=Frameworks/QtConcurrent.framework
Nested=Frameworks/QtCore.framework
Nested=Frameworks/libmpfr.dylib
Nested=PlugIns/imageformats/libqico.dylib
Nested=PlugIns/audio/libqtaudio_coreaudio.dylib
Nested=PlugIns/iconengines/libqsvgicon.dylib
Nested=PlugIns/printsupport/libcocoaprintersupport.dylib
Nested=Frameworks/QtOpenGL.framework
Nested=Frameworks/QtPrintSupport.framework
Nested=Frameworks/QtWidgets.framework
Nested=Frameworks/QtSvg.framework
Nested=Frameworks/libboost_filesystem.dylib
Nested=Frameworks/libboost_regex.dylib
Nested=PlugIns/imageformats/libqsvg.dylib
Nested=Frameworks/libGLEW.dylib
Nested=Frameworks/libboost_program_options.dylib
Nested=Frameworks/QtNetwork.framework
Nested=Frameworks/libfreetype.dylib
Nested=Frameworks/libglib-2.0.dylib
Nested=Frameworks/libpixman-1.dylib
Nested=Frameworks/libopencsg.dylib
Nested=Frameworks/libqscintilla2_qt5.dylib
Internal requirements count=1 size=100
```

Now all the various nested resources have been signed as well. I can verify that the signature is valid:

```
fergu@cumulus % codesign --verify --deep --verbose /Users/fergu/Desktop/OpenSCAD.app           
/Users/fergu/Desktop/OpenSCAD.app: valid on disk
/Users/fergu/Desktop/OpenSCAD.app: satisfies its Designated Requirement
```

Launching and using the newly signed app works as expected with the permissions dialog appearing a single time and not repeatedly. This includes the case where the open files are in `Documents` or `Desktop`. I also tested this by deleting the existing permissions in System Settings and having OpenSCAD request the permissions again.

So it looks like maybe the build step for OpenSCAD on Mac may just need an additional `codesign` invocation to get everything properly signed?

tkircher, 2022-05-12:
@fergu Thanks for figuring this out.

falquaddoomi, 2022-05-12:
Just to add another piece of evidence to the pile, I'm an M1 Macbook user who recently switched from the stable release to the M1-compatible preview (specifically `version 2022.05.09 (git 841808390)`). I love everything about the preview except for this issue with getting prompted multiple times to approve access to the "Documents" folder, despite going through the steps others have in this thread with explicitly giving access to "Documents" and even to the full filesystem.

fergu, 2022-05-13:
I think [this is the line](https://github.com/openscad/openscad/blob/b929e705e76e8ff47d170c5e9849f86002f3e09f/scripts/release-common.sh#L357) that needs to be updated to get the CI builds signed. The [macdeployqt documentation](https://doc.qt.io/qt-5/macos-deployment.html) mentions a `-codesign=<identity>` flag that I imagine does the same thing as I did in my previous post. I know next to nothing about CI, so I'm not sure if there's a way to create/maintain a code signing identity for automatic builds. Maybe the CI configuration has a way to use a consistent signing identity? I'd submit a PR for this, but I'm not sure if my fix is a practical one at all.

As an aside, I went to try this myself on my local machine to see if it actually works, but the building of dependencies failed in `gettext` with a complaint about missing headers for reasons I'm not totally sure of.
```
In file included from ../../../gettext-tools/src/write-catalog.c:23:
../../../gettext-tools/src/write-catalog.h:22:10: fatal error: 'textstyle.h' file not found
#include <textstyle.h>
```

sroeper, 2022-06-24:
I have the same problem, just updated to 2022.06.24 without any change. Is there any progress on this issue?

sroeper, 2022-07-05:
Just tried Version 2022.7 (2022.07.04), still no change in the behaviour. If I open a new script and write "use " it starts to request "access to Documents" and repeats this for every character of the filename, very annoying.

t-paul, 2022-07-05:
I'm not aware of anyone actively working on the issue. I hope it's getting picked up again at some point.

t-paul, 2022-07-05:
@fergu I can merge, the problem is I have no way of testing. The latest build updates to the macOS version mean it's not compatible with my old MacBookPro anymore, so I'm locked out of the game.
I suppose code signing will need a proper certificate, but I can see if I can sponsor that from my company for the time being. So the PR would need to allow for a secure variable to be passed giving the signing info. It would be great to get that into place.

tcurdt, 2022-07-05:
> I suppose code signing will need a proper certificate, but I can see if I can sponsor that from my company for the time being.

So far I've only ever had a full and paid developer program certificate, so I am a not entirely sure about the details - but I am pretty sure unless you want to make openscad available through the AppStore I think (on macOS) there should also be an option to use a free cert (so no need to sponsor).

sroeper, 2022-07-05:
I would be able to be the volunteer for testing :-)

t-paul, 2022-07-05:
@tcurdt That sponsoring is not a money issue :grinning:, it's only that I need to figure out how to clearly separate things. We pay for the dev account anyway for our iOS development stuff.

fergu, 2022-07-05:
How do other open source apps do the signing? AFAIK the full (paid) signing certificate is only needed if the app will be distributed on the App Store and/or to bypass Gatekeeper. It seems like a self-signed certificate generated in a similar manner to the one I made should be sufficient - it will just require the user to do the usual right click->open process on first launch that's typical of other open source apps. I just don't know how one of those would be made and stored for a CI environment.

I'd also imagine that `codesign --verify` could be used to test that the package was properly signed at build time by looking for the `satisfies its requirements` string in the response? It wouldn't verify that the repeated permissions issue that started this thread doesn't exist, but it seems like code signing is what fixes the issue so maybe verifying a proper signature is enough?

sroeper, 2022-07-05:
As far as I understand this site (https://developer.apple.com/support/certificates/) you need a Developer ID Application Certificate (Mac applications), which you can request from within Xcode. Seems all you need is an Apple ID to request a developer certificate.

But some further readings make me think that the problem is different. Even when I open OpenScad the application is not added to the list of applications which requested Folder access (You can see that under System Preferences/Data protection/Files and Folders). So, for me, it looks like this is an issue with MAC (Mandator Access Control). You can read more here: https://developer.apple.com/forums/thread/678819

At least there are some more problems regarding the Info.plist. If you look into that file you will find weired properties like 
  <key>CFBundleIconFile</key>
  <string>/Users/distiller/project/resources/icons/icon-nightly.icns</string>

and I am not sure if it is ok to have a CFBundleSignature of four question marks:
        <key>CFBundleSignature</key>
        <string>????</string>

I think this is the signature to identify files (i.e. bla.scad) and may be also the application itself.

sroeper, 2022-07-05:
Very cool!

I just changed the Info.plist and put the correct name into CFBundleIconFile (icon-nightly.icns) and OpenScad does not ask anymore!
Can't believe it...

sroeper, 2022-07-05:
The hooray was to early. Now I have another problem. I can open all files (all located under Documents/OpenSCAD) and never get the permission question. BUT when I press F5 (even in a new file with just a cube) I get 
`ERROR: Parser error: file access denied
ERROR: Compilation failed! (No top level object found)`

Any ideas?

sroeper, 2022-07-05:
Ok, fixed. A simple reboot and everything seems to work now.

sroeper, 2022-07-05:
A last amendment, I also renamed my OpenSCAD folder in Documents and let OpenSCAD create a new one. After that I moved my libraries Folder an my other files to this new folder and deleted the old one (including the backups folder). Before that I still got file access errors when using my libraries.

Now everything is working like a charm.

sroeper, 2022-08-09:
The error with the wrong info.plist (icon path) is still not fixed in version 2022.08.08, is there anybody out there who can do that please?

tcurdt, 2022-08-09:
I guess it would be good to create PR for this - but I am a little lost on what still needs to be done.

MaximilianGaedig, 2022-09-03:
I encountered this issue aswell,
a workaround for me was disableing the gatekeeper, but this shouldn't be normally required for a cad software,
`brew reinstall --cask openscad --no-quarantine`

jonathanalevi, 2022-09-30:
Still having this issue. Renaming the icon-nightly.icns didn't fix it after a restart... 
How can we fix this once and for all? It's quite frustrating if you're using BOSL2, because it asks 40-50x each time you change a parameter...

tkircher, 2022-10-07:
> Still having this issue. Renaming the icon-nightly.icns didn't fix it after a restart... How can we fix this once and for all? It's quite frustrating if you're using BOSL2, because it asks 40-50x each time you change a parameter...

It does not appear that any devs care about fixing this bug. So the only real workaround other than paying for a signing certificate is to create a subdirectory in your home directory for any OpenSCAD files, and never try to open anything in Documents or the Desktop or any subdirectories under them. This should at least minimize the number of warnings.

tkircher, 2022-10-07:
When most of the comments deny that it's even happening, it is about caring. If it were simply that nobody knows the best way to solve the problem, that's quite different.

MichaelAtOz, 2022-10-07:
This is an Open Source, community developed & supported project. 
People volunteer their efforts.
Bitching doesn't help.

Where is the Mac community? There has to be a programmer in there somewhere.

Someone want to sponsor some supported hardware?

t-paul, 2022-10-07:
I have no idea why it does not happen for some users, accusing those people who took the time to help investigate and comment their findings here of not caring seems very strange. That's also not a new thing, I've observed that personally in other macOS issues.

Anyway, that's not helping to solve the issue. At this point the only option I see is to come to some concensus of what will fix the issue, there's a couple of potential problems @sroeper has outlined. As I can't validate results by running the build, it would be great to have a full summary of changes (or even better a PR). I'll try make time to getting the changes in, but this will need someone to validate the results.

t-paul, 2022-10-07:
As for hardware, I might get access to a MacBook which at least should be new enough to build and run OpenSCAD later this year due to hardware upgrades at my company, we'll see.

tcurdt, 2022-10-07:
Bitching does indeed not help.

Everyone is free to hate Apple but this isn't really about them.
This does not need a paid cert.
It should be possible (albeit painful) to iterate via CI/CD.
But I'd also be willing to build on my hardware for testing.

But frankly, [I am at a loss](https://github.com/openscad/openscad/issues/4059#issuecomment-1209178888) what needs to be fixed at this stage.
If a fix was found this needs to be turned into a PR - and then we can try.

DavidPhillipOster, 2022-10-08:
Info.plist is missing required key-value pair: NSDocumentsFolderUsageDescription
Apple documents it here: https://developer.apple.com/documentation/bundleresources/information_property_list/nsdocumentsfolderusagedescription

So the fix might be as simple as adding that key-value pair to the Info.plist.

I did inspected version 2022.3 and did a fresh pull of the source code from github and did not find that key value pair.

tcurdt, 2022-10-08:
I've added a PR for the privacy keys.
We probably still need a fix for the icon.

t-paul, 2022-10-08:
Build with changes by @tcurdt - https://app.circleci.com/pipelines/github/openscad/openscad/2944/workflows/506261a4-d1ec-4b9e-a757-6d5b65e70867/jobs/12282/artifacts

tcurdt, 2022-10-08:
Alright. Some progress.
Thanks for the branch build, @t-paul 

This what is in the repo:

[image]

This is what is how it comes out in the build:

[image]

And this is how it works on my machine after adjusting the plist:

[image]

Seems like I missed there are 3 files related to the Info.plist:

```
$ find . -name "*Info.plist*"
./xcode/Info.plist
./Info.plist.in
./Info.plist
```

It seems like `./Info.plist.in` creates `./Info.plist`.

```
MACOSX_BUNDLE_INFO_PLIST ${CMAKE_SOURCE_DIR}/Info.plist.in
```

But then why is `Info.plist` in the repo?
And what is the `./xcode/Info.plist` for?

Anyway, it seems `./Info.plist.in` is the one to deal with.

I've now updated the PR. After further testing it seems the privacy keys are actually not required at all.
It really is only the `MACOSX_BUNDLE_ICON_FILE` causing the problem.

To fix the icon issue we need to set the `MACOSX_BUNDLE_ICON_FILE` to a relative path. This is wrong:

```
set(MACOSX_BUNDLE_ICON_FILE ${RESOURCE_DIR}/icons/icon-nightly.icns)
```

I have adjusted the makefile accordingly.

tcurdt, 2022-10-08:
Alright. I am somewhat confident things are fixed now :)

The new build has the icon set correctly. And I also fixed the `CFBundleVersion` which needs to be updated between builds. It seems like it is being used to cache some of the behaviour on macOS which might have been one of the reasons for a few head-scratchers on this issue.

Maybe give the build a try and give some feedback. It's available at

https://output.circle-artifacts.com/output/job/8ac867f5-7882-4774-a8f3-90ba4496c55a/artifacts/0/tmp/out/OpenSCAD-2022.10.08.dmg

[48 more replies]

closed 2024-10-31 by kintel