openscad/openscad #4060: Console cmd+a works, but cmd+c doesn't

3Dietrich, 2022-01-20:
OpenSCAD Version 2021.01
OpenSCAD Version 2022.01.20

@ macOS 12.0.1, MBAir 2020 M1

sometimes I want copy something out of the console. for copying I am very used to shortcut 'cmd+c'. 
But this doesn't work in console. cmd+a (select all) works, cmd+c doesn't.

thehans, 2022-01-24:
The Console pane is unrelated to QScintilla, it is a Qt [`QPlainTextEdit`](https://doc.qt.io/qt-5/qplaintextedit.html) widget.
Ctrl-C in the Console works for me on Linux, FWIW.

Sounds like something strange is going on with focus on Mac, where its being forced back to the editor.

MichaelAtOz, 2022-01-24:
On Windows 7 & 2021.01, when the console has focus from a click, ctrl-a doesn't work, with selected text ctrl-c does work.

Both are on the context menu, 'Copy' has the underlined C shortcut, 'Select All' doesn't - it is usually an underlined A, sometimes 'l' (that's lower L).

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

RAMilewski, 2022-06-30:
On MacOS Mojave and Monterey, neither command-C nor copy from the Edit menu work for me. Copy from the context menu does.

On Ubuntu 20.04 Ctrl-C does, copy from the context menu does, but copy from the Edit menu does not.

thehans, 2022-08-11:
@RAMilewski wrote:
> ... copy from the Edit menu ... 

I'm pretty sure this is intended to copy from the Editor pane only, not the Console.  Hence why it doesn't work on Mac nor Linux for you.

jordanbrown0, 2022-08-11:
Ctrl+C works from the Console in Windows.  (But only when the Console has focus, as hinted by the fact that the highlight goes away when focus moves.)

But Edit/Copy does not.

It is probably relevant that the context menu in the Console has Copy/Ctrl+C and Select All/Ctrl+A on it.

jordanbrown0, 2022-08-11:
In Windows, Ctrl+A does work in the Console.  (Windows 10, but I doubt that's relevant.)

thehans, 2022-08-13:
I think I have a solution to this, in PR #4326.  These changes should also allow the Edit->Copy menu to be used on either Editor or Console.

I've only tested on Linux Mint 22.  I don't have access to a mac to verify that this fixes the original issue of cmd-c shortcut in Console.
But if @ChrisCoxArt or @3Dietrich can test the build [here](https://app.circleci.com/pipelines/github/openscad/openscad/2867/workflows/0113bee3-6fec-4f8d-a098-2cc2ae77e2bd/jobs/12068/artifacts), then we can get it merged to master and wrap up this issue.

thehans, 2022-08-18:
Or @RAMilewski ... or anyone else with a mac that could test?  Bueller?

3Dietrich, 2022-08-18:
> Or @RAMilewski ... or anyone else with a mac that could test? Bueller?

YES!
I could copy selected content from the Console Window into an editor.

closed 2025-08-16 by jordanbrown0