openscad/openscad #4065: Preview transparency issue: depends on order

avivhu, 2022-01-22:
I'm making a transparent cylinder with an opaque cube within. The preview won't show it as transparent in some cases. 

[image]

Any way to make this work regardless of objects order?

```
module pipe(){
    color("green", alpha=0.2)
    linear_extrude(height=10, convexity=10, center=true) {
       difference() {
           circle(5, center=true); // Outer circle 
           circle(4, center=true); // Inner circle 
       }
     }
}

// OK: Pipe is transparent
module do_it1()
{
    cube(5, center=true);
    pipe();
}

// Wrong: why is pipe not transparent?
module do_it2()
{
    pipe();
    cube(5, center=true);
}

translate([-8, 0, 0]) do_it1();
translate([8, 0, 0]) do_it2();

```

nophead, 2022-01-22:
It is a long standing problem, see #1390

t-paul, 2022-01-22:
Closing as duplicate.

closed 2022-01-22 by t-paul

thehans, 2022-01-22:
[Order-independent transparency](https://en.wikipedia.org/wiki/Order-independent_transparency) is not a simple problem to solve, and particularly impossible for preview mode ([OpenCSG](http://opencsg.org/)) which already must do convoluted tricks to simulate CSG operations without computing the mesh.

nophead, 2022-01-22:
I doubt it is impossible. I draw all my transparent objects after the things behind them and that mostly works unless there is a transparent object behind. Then it depends on the view direction but I think I could actually make that depend on $vpr if I could be bothered.