openscad/openscad #4071: Can't save customizer json file when path contains unicode characters on windows.

rkail, 2022-01-26:
With OpenSCAD version 2021.01 on windows 10 the customizer json file will not be saved when the path to the file contains unicode characters. Examples:

With 
C:\Users\Richard\Desktop\xx\Path\path2\test.scad it works as exprected

With
C:\Users\Richard\Desktop\xx\Path\path2\öäüÖÄÜß\test.scad
openscad tries to save the file as 
C:\Users\Richard\Desktop\xx\Path\path2\Ã¶Ã¤Ã¼Ã–Ã„ÃœÃŸ\test.json
which doesn't work, obviously.

When the filename is ÄÄtest.scad, openscad save the file as Ã„Ã„test.json.

ChrisCoxArt, 2022-01-26:
This appears to work correctly on MacOS.
Looks like the error would have to be in ParameterWidget::getJsonFile, or under boost::property_tree::write_json .
We don't seem to have any platform specific code in there, but something could be wrong in the way STL, QString, or boost handle unicode paths.

MichaelAtOz, 2022-01-26:
> When the filename is ÄÄtest.scad, openscad save the file as Ã„Ã„test.json.

and in the console, after Save-As, it is:

 

 

But later:

Parsing design (AST generation)...

Saved backup file: C:/Users/MeB/Documents/OpenSCAD/backups/ÄÄtest-backup-bLFRiddg.scad

 

&

 

Parsing design (AST generation)...

Saved backup file: C:/Users/MeB/Documents/OpenSCAD/backups/ÄÄtest-backup-bLFRiddg.scad

ERROR: Parser error: syntax error in file ��test.scad, line 2 

Execution aborted

 

??

 

That error is due to:

 

ÄÄtest=2; // [1:10]

 

I thought Unicode in variables was valid.

The was work on fixing Unicode some time ago, something regressed.

Let me know if you want this in a separate issue.

 

-----------------------------------------------------------------------------------------------------------------------

 

I'm including this image, in case any of the above Unicode doesn't survive the emailing

 

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2022-01-26:
Well, that didn't email well...this is the image I mentioned.

[image]

ChrisCoxArt, 2022-01-27:
The original report may be a duplicate of  https://github.com/openscad/openscad/issues/113
And https://github.com/openscad/openscad/issues/2658

No, unicode in names is not valid in OpenSCAD.  We already have a feature request to allow unicode in names.

rkail, 2022-01-27:
from the users view, there are two problems with this:

1) that the pathname is mangled

2) that the failure of writing the json file is completely silent. It would be nice to mention the fact at least in the log. IMHO this is even more severe, because the customization data is silently lost.

PRouzeau, 2022-02-28:
Might be a duplicate of #2453 (closed)

closed 2025-11-20 by kintel