openscad/openscad #4072: Issues with Fedora 35, flatpak, rpm and themes...

Dorro101, 2022-01-27:
I run OpenSCAD on Fedora 35 under Wayland using the Adwaita Dark theme. The theme is applied, however there are several issues with the UI that make it unusable. The main window does not meet the edges when maximised, the menus appear below the menu bar instead of matched to the bar. The UI flashes and stutters when moving the mouse from window to window, sometimes showing plain solid black or randomly the desktop elements underneath, sometimes the same when the render window refreshes. When I delete all the local .config files, the UI elements return to their correct positions, but soon returns to its usual sad state of affairs, immediately when creating a new tab. The RPM provided by Fedora has exactly the same issues. I tried using the AppImage, which crashes outright on opening a file from the file dialogue. So I am currently using a snap install, which uses the theme provided in the snap I think, and does not recognise the default desktop theme. If you would like screen shots, I can provide...

Ciao
Dorian

woolmonkey, 2022-02-21:
Same issues but with Pop-OS

robdaemon, 2022-04-22:
[image]

Attached is a screenshot of Fedora 35, Wayland and GNOME desktop, on a 4k display, showing the menu dropping well below the bar.

thehans, 2022-04-22:
We use Qt for GUI, and no specific code for positioning menu dropdowns AFAIK.  That is normally handled automatically, somewhere between Qt and the window manager I would guess.
So these sound like issues between Wayland and Qt5, which we don't have much control over.

As for the AppImage crashing when opening a file dialog, mentioned in the original post, that is a separate issue #3523, related to how the appimage is created (using linuxdeploy) which should grab all the proper dependencies, but [it seems it fails to do so](https://github.com/linuxdeploy/linuxdeploy-plugin-qt/issues/91).