openscad/openscad #4073: Increasing max_buttons

chschmitt, 2022-01-29:
I tested gamepad navigation with a PS3 controller, which has 17 buttons. The right button on the left arrow pad would be `Button16`, but max_buttons is only 16, so it is not possible to assign it.

My OpenSCAD version is `2022.01.25.ai10450 (git 6f3767b99)`

t-paul, 2022-01-29:
Wow, that's a lot of buttons :grinning:. Yes, increasing the number should not be a problem.

closed 2022-01-30 by t-paul

MichaelPFrey, 2022-01-30:
is that as easy as you think?

Under Linux with the joystick driver it should be.

When you are under Windows and using QGamePad, we are limited by the idiosyncrasies of QGamePad: https://doc.qt.io/qt-5/qgamepad.html (including that the xbox 360 d-pad is mapped to buttons, not axis). To access more then 16 buttons under windows, we would need a native windows driver.

t-paul, 2022-01-30:
I tested on Linux using https://github.com/lemmingDev/ESP32-BLE-Gamepad flashed to an ESP32 and defining a bluetooth gamepad with 24 buttons. That worked just fine via the Joystick driver. The jstest-gtk application did show the 24 buttons working too. Other drivers might have potential issues, we'll see I suppose.

chschmitt, 2022-01-30:
@t-paul Thanks for the quick fix. I already tested the new nightly and it works like a charm.

t-paul, 2022-01-30:
@chschmitt good to hear, thanks. The changes grew a bit when looking at things, but that's how it goes sometimes :smiley:.

commit 993f6351 (2022-01-30): Increase input device button count to 24 (fixes #4073).
commit e4b2888c (2022-01-30): Merge pull request #4074 from openscad/increase-nr-of-buttons

Increase input device button count to 24 (fixes #4073).