openscad/openscad #4075: Implement sketcher to make prototyping more efficient

Kreyren, 2022-01-31:
For context my current workflow is to use FreeCAD to do the prototyping and initial design and then switch to OpenSCAD to define the shape and make it functional as i find the FreeCAD's sketcher workbench significantly more efficient for that purpose.

So in practice i have:

[image]

That then enables me to easily define:
[image]

Where from my experience with both solutions FreeCAD's sketcher is sigfinifantly more efficient as it enables to define constraints and constructions line to figure out the design while making the design interactive for quick changes to produce prototype.

### Proposal

Implement similar solution in OpenSCAD's editor to enable parametrization of polygon shapes to make the prototyping in OpenSCAD more efficient

Kreyren, 2022-01-31:
Also it's very difficult to make the shape functional (purple) in OpenSCAD as i can't really find the exact dimensions for this point (green)

[image]

Where in FreeCAD it just takes like 5 sec to define a construction line and get the dimension.

[image]

[image]

t-paul, 2022-02-04:
It's not clear what you are asking. Is this about a GUI based sketcher?

There's some disucssion on adding some means to declare piecewise 2D geometry, maybe with support for a set of constrained based features to simplify calculations.

Kreyren, 2022-02-04:
> Is this about a GUI based sketcher? -- @t-paul (https://github.com/openscad/openscad/issues/4075#issuecomment-1030113011)

seems like a correct description of the proposal, basically FreeCAD's sketcher but in OpenSCAD that interprets in polygons and polyhedrons.

> 

There's some disucssion on adding some means to declare piecewise 2D geometry, maybe with support for a set of constrained based features to simplify calculations. -- @t-paul (https://github.com/openscad/openscad/issues/4075#issuecomment-1030113011)

Reference it plz

t-paul, 2022-02-04:
https://github.com/openscad/openscad/issues/3632#issuecomment-767648396 and following. Other discussions were in chat only.

Anyway, I think a solution which would integrate UIs like FreeCAD sketcher and/or LibreCAD would be great, but that's out of scope of a normal feature request. If someone would take on that project, I'm happy to support that, but there's not going to be a separate implementation of such a full CAD GUI in OpenSCAD directly any time soon.

Kreyren, 2022-02-04:
why is it out of scope? Prototyping is important..

t-paul, 2022-02-04:
Because a tiny team can only maintain something of a limited scope. There's already > 600 open issues, quite a number of them feature requests very near the current application scope. Adding essentially a second GUI application is simply not feasible at this point.

As I said, if someone wants to take on that topic, especially if it's integration with other projects, that would be great. But we don't need an open ticket for that which makes it look like someone might start working on that any day.

nophead, 2022-02-04:
Isn't the whole point of OpenScad that you program solids with
scripts instead of drawing them with a  GUI? There are plenty of GUI based
CAD programs. OpenSCAD is for people who think in code and not mouse
movements.

Kreyren, 2022-02-05:
@nophead I think in code and i have no idea how to achieve this prototyping workflow in OpenSCAD or OpenSCAD-way, any ideas?

nophead, 2022-02-05:
No you need a different workflow in OpenSCAD. You describe positions and shapes mathematically. All the dimension lines become constants in the code and the positions of the vertices of the polygons are calculated by trigonometry from the specified dimensions.

Kreyren, 2022-02-05:
i am not aware of different workflow for prototyping like prototyping even a small thing in OpenSCAD is like ~100 lines that you need to rewrite multiple times that takes like ~3min per change in comparison to FreeCAD's solution that takes <5 sec..

I don't dispute that OpenSCAD is significantly better for the final product though

t-paul, 2022-02-05:
It's an interesting discussion, but a ticket here is not the best place to have it.

Let me summarize:
- Add Sketching GUI -> Would be interesting as integration with other tools, but not implemented in OpenSCAD
- Current 2D language is not very flexible, additional feature are requested for that already, libraries might provide some quite useful ways for specifying 2D geometry in a different way, e.g. [Turtle Style](https://openhome.cc/eGossip/OpenSCAD/lib3x-footprints2.html)
- Code + Constrained based definition would be interesting and has been asked before and should fit much better with OpenSCAD scope

closed 2022-02-05 by t-paul