openscad/openscad #4076: Feature request: allow module default parameter values to refer to previous parameters

dc42, 2022-02-01:
It would be useful if default values for module parameters could refer to the values of other parameters declared earlier in the parameter list. Here's an example of how I would like to use this feature:

```
// Cube centred at a location, by default centred at X0 Y0 and touching Z0
module ccube(x=1, y=1, z=1, cx=0, cy=0, cz=z/2) {
	translate([cx-x/2,cy-y/2,0]) cube([x,y,z]);
}
```

nophead, 2022-02-01:
It might cause problems if you want to access a global variable with the same name. In your example if there was a file scope variable named z, currently I think z/2 would reference that z.

t-paul, 2022-02-01:
Yes, this has been discussed in the past and so far there's no good solution for implementing this sequential assignment behavior (like `let()`) without breaking cases that rely on the parallel assignment.

I'll leave the issue open, but it's not clear if there's a solution anytime soon.

One way to implement this now is to check if the parameter is defined
```
module a(s, y = 2, z = y) {
    echo(s, y = y, z = z);
}

module b(s, y = 2, z) {
    effective_z = is_undef(z) ? y : z;
    echo(s, y = y, z = effective_z);
}

y = 3;

a("a: no params");
a("a: with param z", z = 4);

b("b: no params");
b("b: with param z", z = 5);
```
It's not 100% identical, but should cover most cases.

dc42, 2022-02-01:
Thanks t_paul, that gives me a workaround.

closed 2025-03-09 by jordanbrown0