openscad/openscad #4080: Difference in the side of extruded shape.

bvwj, 2022-02-02:
When I try to cut a hole in the side of an extruded part, the F5 render fails.  F6 works.

Here s the linear_extrude example from the documentation, with a cylinder bore through the side.  You can see what I am trying to do with the % operator.  When you remove the %, the preview is a little nuts.

Is there a better way to accomplish this?

```
difference() {
    linear_extrude(height = 60, twist = 90, slices = 60) {
       difference() {
         offset(r = 10) {
          square(20, center = true);
         }
         offset(r = 8) {
           square(20, center = true);
         }
       }
     }
     translate([0,0,30])
     rotate([90,0,90])
   %  cylinder(h=60,r=5,center=true);
 }
```

nophead, 2022-02-02:
You need to set the convexity parameter. It looks OK with convexity = 10.

bvwj, 2022-02-02:
It does, thank you.  I though convexity=10 was the default.

jordanbrown0, 2022-02-02:
Nope.  The default is 1.  I've proposed raising it to 10, but so far that's stalled on concerns that it will adversely impact performance.  (There **are** performance impacts, but they seem to be very small unless you have a **very** complex model.)

When I get a chance, I'm going to come back with a proposal to have the default be a preference item that can be overridden with a $ variable.

t-paul, 2022-02-02:
Ignoring all the concerns raised will not make things easier. Why not start slowly and simple with a change near the state of the discussion?

nophead, 2022-02-02:
Note in this case basic shape only has a convexity of two or three but the ripples caused by the triangulation of the slices add to it, so with the worst case viewing angle it is probably slices / 2.

t-paul, 2022-02-02:
Specific issue seems solved, for more general discussion, see #4081.

closed 2022-02-02 by t-paul