openscad/openscad #4093: KVM Switch causes crash during startup device enumeration

whited, 2022-02-05:
I discovered a crash in OpenSCAD.
Some troubleshooting has narrowed it to an HID device enumeration that goes badly when my KVM switch is connected to my USB port.

Some details:
* OpenSCAD v 2021.01
* Windows 10
* KVM is a Chinese brand TESmart dual-monitor KVM ([Amazon Link](https://www.amazon.ca/gp/product/B086DMQNV2))
* Also have a 3DConnexion SpaceMouse Wireless connected.

With the KVM connected, OpenSCAD crashes immediately after leaving the "Welcome" dialog window.  Disconnecting the KVM solves the issue.

With some guidance, I enabled tracing of HIDAPI events, and captured the logs from that, both with and without the crash issue:

Event log, with KVM connected and OpenSCAD crash: [full log w/ error](https://gist.github.com/whited/4360de8aa57c65aa481658018a8ca5e1)

The interesting bit is right at the beginning:

```
0: HidApiInputDriver::open()
1: Enumerating HID devices...
104: D: 046d:c52b | path = \\?\hid#vid_046d&pid_c52b&mi_01&col04#8&4f58a8e&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}, serial = <null>, manufacturer = అ㠊脂쀆׀ऌꄁ蔁锃甂ᔐ☁˿ęＪ脂쀀ą耉ơ҅ƕɵĕ̥舉脉茉ٵ΁ۀﾼ蠉ơࢅƕࡵĕＦᤀ⤁臿쀀৿ꄁ蔁锐甆ᔈ☀ÿĉĉۀ＀ȉơᆅ᎕ࡵＦऀ脂ऀ鄂쀀৿ꄄ蔁锠甎ᔈ☀ÿ䄉䄉ↅᾕ䈉䈉ীꄁ萁ؠ需錀甀脀ꂊʐꆅ׋萀倀蔄똀ྀ臠蔂舠쌀qኅ耊႐ꨛµↁႈ É, product = ࠂ舆΁ˤꀈʁꊁʌ言ʁ膑ʒ芁₁釗ҍ逈肁
```

For reference/comparison, I also captured the HIDAPI event log, where the only difference is the KVM was disconnected.  (OpenSCAD worked normally for this) [full log, no error](https://gist.github.com/whited/7bde5589f04c10dc74deacd898b47ced)

Scopeuk, 2022-02-07:
@whited is the spacemouse connected via the KVM or directly to the machine in question?

t-paul, 2022-03-27:
@whited Could you please try with the latest dev snapshot? This has the HID driver library updated. I suppose it's more likely the issue is on OpenSCAD side, but there's a small chance the new library could fix the issue. It seems to come with quite some changes compared to the one used till now.

whited, 2025-10-29:
Sorry -- I no longer own the KVM switch this issue refers to, I won't be able to confirm.

closed 2025-10-29 by whited