openscad/openscad #4104: CGAL version gate needs to match the new value in the README

dannypike, 2022-02-09:
I noticed that the README.d has had the CGAL version updated to 5.3 in version 42f222ab51d1f894c73445b39886c070883fcfec:

```
* [CGAL (5.3 ->)](https://www.cgal.org/)
```

However, the file `src/CGAL.h` is still gating for version 4.6:

```
#if CGAL_VERSION_NR <= CGAL_VERSION_NUMBER(4, 6, 0)
#pragma error("CGAL 4.6.0 or above is required.")
#endif
```

t-paul, 2022-02-09:
I'm not sure we want that.

The compile check blocks even trying to compile with a version that does not support the new features at all and so would fail compilation later anyway with some more obscure error messages.

However the readme indicates the minimum version we support as it has a number of bugfixes and improvements we don't want to work without.

@ochafik what do you think?

thehans, 2022-02-09:
I think if we want a compile time version check for any dependencies, it should be done within `CMakeLists.txt` by specifying a minimum version in the `find_package` command.  
Currently no version is given for CGAL:
https://github.com/openscad/openscad/blob/86cff4eae7e4944884880447175a63b06fc78113/CMakeLists.txt#L298

t-paul, 2022-02-09:
Good point. It should move there. The question about which number to use remains though.

dannypike, 2022-02-10:
I also agree that putting version numbers into ```CMakeLists.txt``` is the best thing to do; I'm planning to add a `vcpkg` manifest file so that we can ensure that they get used in the Visual Studio (VS) build.

Currently, the VS build is using the latest builds of all of the dependencies, but that stopped working with the most recent merge after 02FEB2022. There are problems with CGAL renaming its templates in 5.4.

dannypike, 2022-02-11:
Sorry for splitting this comment over two days; I was called away just after I posted the above, so this is continuation of that ...

This code in the file ```cgalutils-tess.cc``` is causing the build to fail under MSVC:
```
#if CGAL_VERSION_NR < 1050401000
#include <CGAL/Triangulation_2_projection_traits_3.h>
typedef CGAL::Triangulation_2_filtered_projection_traits_3<K> Projection;
#else
#include <CGAL/Projection_traits_3.h>
typedef CGAL::Filtered_projection_traits_3<K> Projection;
#endif
#include <CGAL/Triangulation_face_base_with_info_2.h>
```
The current ```vcpkg``` port is CGAL version 5.4, which has a ```CGAL_VERSION_NR``` of 1050400900, which means that Visual Studio tries to compile the first block in the ```#if``` statement. However, there is no definition of ```CGAL::Triangulation_2_filtered_projection_traits_3<K>``` in ```vcpkg's``` version of CGAL 5.4, so the build fails.

However, the other ```typedef``` does appear to compile OK under ```vcpkg's``` port of CGAL 5.4.

Any idea what's going on? And, what is the CGAL version that corresponds to ```CGAL_VERSION_NR == 1050401000```?

Here is the header file that I get when I pull CGAL from ```vcpkg```:

```
// Copyright (c) 2009  
// Utrecht University (The Netherlands),
// ETH Zurich (Switzerland),
// INRIA Sophia-Antipolis (France),
// Max-Planck-Institute Saarbruecken (Germany),
// and Tel-Aviv University (Israel).  All rights reserved.
//
// This file is part of CGAL (www.cgal.org)
//
// SPDX-License-Identifier: LGPL-3.0-or-later OR LicenseRef-Commercial
//
// Author(s)     : -

// This file is automatically created by cmake
// Do not edit manually.

#ifndef CGAL_VERSION_H
#define CGAL_VERSION_H

#define CGAL_VERSION 5.4-I-900
#define CGAL_VERSION_NR 1050400900
#define CGAL_SVN_REVISION 99999

#define CGAL_RELEASE_DATE 20220210

#include <CGAL/version_macros.h>

#endif
```

ochafik, 2022-02-12:
The situation of the version compatibility w/ CGAL is a bit awkward at the moment because of fast-csg + https://github.com/CGAL/cgal/issues/6308. Until 5.3.2 (and presumably 5.4.1) are out with the fix, we have both a minimum CGAL version *and* a maximum version (5.2.x branch), at least on Win64.

(I could cut fast-csg out when those versions are detected but it seems less effort to just wait for their patch releases).

Otherwise I'd be voting for 5.4.1 as minimum version as it comes with new corefinement visitor methods, but I don't think that's realistic for now. I'm unaware of any meaningful bugfixes that affect us enough to change the min version tbh.

ochafik, 2022-02-12:
Mmh actually it looks like some parts of CGAL are still moving targets (e.g. the corefinement callbacks are in flux, just hit a snag rn https://github.com/openscad/openscad/pull/4117/commits/787dc34ad0dda53e8e2766c51eac1722fa6972c1), so the later version the better :-D

t-paul, 2022-02-13:
I would ignore the compilation issue for that as it's a platform dependent bug.

Otherwise, I'm fine with setting min-version to something higher, but probably not 5.4.1 as I think we should at least be able to compile against a very recent setup (specifically Linux distros) without the need to get a patched CGAL. Using some newer features with a version switch is ok.

Unfortunately Ubuntu 20.04 has CGAL 5.0, most other distros seem to have at least 5.2.

So would it be ok to go with "compiles with" >= 5.0 (goes into CMakeLists.txt) and "works best with" >= 5.4.1 (goes into readme, one that's released)?

t-paul, 2022-02-13:
@dannypike That version issue means vcpkg has the beta version of CGAL 5.4. I don't know why that became so widespread in packaging. Fedora seemed to have done that too, but now switched to the release version.

Could you check if using `< CGAL_VERSION_NUMBER(5, 4, 0)` instead works? I'm not sure how that covers beta versions.

t-paul, 2022-02-13:
Sounds good, for now I'll put 5.0 as requirement for cmake, and 5.4 as recommended into the readme (via #4118).

commit 8d1fa8a8 (2022-02-13): Merge pull request #4118 from openscad/cgal-version

Update CGAL version numbers (fixes #4104).
commit c32efe04 (2022-02-13): Update CGAL version numbers (fixes #4104).
closed 2022-02-13 by t-paul