openscad/openscad #4105: Dark theme in nightly AppImage

EmVee381, 2022-02-10:
Hi,

i would like to use dark theme in OpenScad nightly AppImage... but i found it is not possible use system qt5 theme, but i found that theme must be part of AppImage ??? if it is true can you add any dark theme, or can you help me how to run AppImage version with dark theme?

t-paul, 2022-02-13:
I tried a couple of suggestions, e.g. packaging `qt5ct`and so far nothing works. I'll leave that open, but this will have to wait for someone helping out with new ideas on how to get that working.

stevealexr, 2022-10-30:
Failed Attempt 
1) Using [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) with the argument `-extra-plugins=styles/libqt5ct-style.so,platformthemes/libqt5ct.so` to repackage the appimage in ubuntu bionic beaver (18.04.6). (linuxdeployqt can only be used with the oldest supported linux version). The theme isn't changing and it wont launch in other ubuntu versions.

Somewhat Successful Attempt
1) Install qt5ct using `sudo apt install qt5ct`
2) Grab a copy of [official openscad appimage](https://openscad.org/downloads.html) and run it with `--appimage-extract` to get the App Dir which is a folder named squashfs-root.
3) Edit the file `squashfs-root/apprun-hooks/linuxdeploy-plugin-qt-hook.sh` and change the line `export QT_QPA_PLATFORMTHEME=gtk2` to `export QT_QPA_PLATFORMTHEME=qt5ct`
4) Copy the `libqt5ct.so` file and `libqt5ct-style.so` file from `/usr/lib/x86_64-linux-gnu/qt5/plugins/platformthemes` and `/usr/lib/x86_64-linux-gnu/qt5/plugins/styles` to `squashfs-root/usr/plugins/platformthemes` and `squashfs-root/usr/plugins/styles` respectively.
5) Use [appimagetool](https://appimage.github.io/appimagetool/) to package the App Dir back to an appimage and you should be able to run it and change the theme via qt5ct.

Here is a copy of openscad-nightly-2022.10.26 appimage for x86_64 that is packaged in ubuntu focal fossa (20.04.5). It works in ubuntu focal fossa and ubuntu jammy jellyfish (22.04.1) but not in ubuntu bionic beaver (18.04.6).
[Google drive link](https://drive.google.com/file/d/17IWZHOwcdwzprcUcYoByFhWvGVqFj5LV/view?usp=share_link)
Magnet Link(not working right now) -> magnet:?xt=urn:btih:3768EB42DD07D24E6CFC796DE3407F2DD6EC98A9&dn=3768EB42DD07D24E6CFC796DE3407F2DD6EC98A9.meta&tr=https%3A%2F%2Ftracker.expli.top%3A443%2Fannounce

Nikorasu, 2023-06-27:
Does anyone know a way to fix this? I tried using the version @stevealexr uploaded (system has qt5ct installed), but no luck, should I try doing those 5 steps myself?

I'm on Linux Mint 20, using the system Mint-Y-Dark theme, and most windows obey that fine.. However, the AppImage for OpenSCAD still uses a white theme for the window colors outside the editor/3d preview. (If I install OpenSCAD from the terminal, that obeys the theme, but that installs a version from 2019..)

[image]

Is there any way I can get a more recent version to work with the system dark theme?

tpimh, 2024-02-05:
I haven't looked deeply into that, but it seems that the devlopers of vnote got it fixed for them: https://github.com/vnotex/vnote/commit/e534eec083c32ec2e04146a7ce8871a162da11aa

They are using [probonopd/linuxdeployqt](https://github.com/probonopd/linuxdeployqt).