openscad/openscad #4109: fast-csg eats fabric

rcolyer, 2022-02-11:
I tried many old designs, and fast-csg mostly performed admirably with good performance boosts.  One that it completely failed on (with just the fast-csg feature enabled and no related optionals enabled) is fabric_v2.scad, the second file on this download page:  https://www.thingiverse.com/thing:1693052/files

With the master branch and fast-csg off, it renders correctly for me in 4min 41s.  With it on, it completes in 6s without error, but the output is empty.

ochafik, 2022-02-11:
Thanks for reporting this (and cool models!!).

I can't seem to reproduce this on macosx (tried a fresh build and yesterday's nightly snapshot), it builds fine in 13s here.
Which platform did you get this on? (and did you build OpenSCAD yourself / with which CGAL version?)

ochafik, 2022-02-12:
Ok, unclear if it's the same bug but it looks like `fast-csg` eats [lizards](https://www.thingiverse.com/thing:1034488/files) (not all of them if not using `fast-csg-trust-corefinement`, to make things more interesting)

rcolyer, 2022-02-13:
I did a fresh pull and clean build, so I'm up-to-date, and it still displays nothing on render.  But I remembered you saying you were doing most of your tests from the command line, so I did a command line stl output of the same, and it succeeded!  So I went back and did the gui render, nothing displays, but the stl output from the gui still succeeds even with nothing shown on render!  So fast-csg has technically done a successful render of this, but for some reason enabling fast-csg results in the display of the render result going empty.

t-paul, 2022-02-19:
```
module base() {
    linear_extrude(5) offset(3) square([20, 8]);
}

module lid() {
    difference() {
		base();
		cube(); // <-- 
    }
}

base();
translate([0, 0, 18]) mirror([0, 0, 1]) lid();
```
Another one, disabling the `cube()` changes the behavior.

UBaer21, 2022-02-19:
```
// with lazy union activated the objects (min 2) in union below top level will vanish in render view (but are still there)
cube();  // remove this
translate([10,0])cube(2); // and this and the union will apear in render again

translate([10,10])union(){  // these vanish
  sphere(); // remove one from the union and the other re apear
  translate([10,0])cylinder();
}
```

UBaer21, 2022-02-19:
```

//translate([0,0,10])cube();  // even without this if more "groups" are in scene

translate([20,20]) {
		cube(5);
		cube(); // <-- 
}

translate([10,0])intersection() {
		cube(5);
		cube(); // <-- 
}

translate([10,10])difference() {
		cube(5);
		cube(); // <-- 
}

translate([10,30])union() {
		cube(5);
		cube(); // <-- 
}
```

closed 2022-02-22 by t-paul