openscad/openscad #4114: fast-csg performance regressions when combined with trust-corefinement and/or lazy-union

wolfwood, 2022-02-12:
#3641 is exciting work that is giving me real improvements, however I am seeing `fast-csg-trust-corefinement` take significantly longer in several cases (though the problem models, _thumb_, _fingers_, and _assembly_, are related). I'm also seeing that enabling `lazy-union` with `fast-csg` can give slower results than `fast-csg` alone (for _plate_ and _assembly_).

I gathered commandline render times on my branch https://github.com/wolfwood/tryadactyl/tree/fast-csg-test using the script there called `time.sh`

|Model | stock render | `lazy-union` | `fast-csg` | `fast-csg` `lazy-union` |   `fast-csg` `trust-corefinement` |   `fast-csg` `trust-corefinement` `lazy-union` |
|:---:|---:|---:|---:|---:|---:|---:|
| _rest_          | 4.238 | 4.229 | **3.806** | 3.869 | 3.849 | 3.855 |
| _thumb_      | 63.618 | 62.283 | 21.795 | **19.468** | 115.209 | 35.996 |
| _plate_        | 396.910 | 349.068 | 72.449 | 108.148 | **24.474** | 63.494 |
| _fingers_     | 168.728 | 166.485 | 84.211 | **76.821** | 864.384 | 840.219 |
| _assembly_ | 705.781 | 605.382 | **190.565** | 216.741 | 1005.702 | 935.563 |

ochafik, 2022-02-13:
Amazing, thanks for the test cases and for taking the time to crunch the numbers!

I couldn't run *fingers*, *thumb* (missing from the repo) or *assembly* (depends on missing `../KeyV2/includes.scad`), but I reckon these very slow runs would probably be improved (or fixed) by the remeshing (can try it out https://gist.github.com/ochafik/28bde2491d1f8cc0992a51c8e6f25716 if you're willing to build OpenSCAD :-D)

wolfwood, 2022-02-13:
oh, whoops. I fixed the missing files, and dropped the unnecessary dependency.

successfully followed that gist, I'll give that a shot.

wolfwood, 2022-02-13:
the numbers for `thumb` are not significantly different with the new build.

ochafik, 2022-02-13:
Oh sorry I should have mentioned the new build has a hidden ‘fast-csg-remesh’ feature 😅

wolfwood, 2022-02-13:
`--enable=fast-csg --enable=fast-csg-remesh` segfaults on all my models.

wolfwood, 2023-02-12:
I've updated my test script to make it easier to use with multiple test case sources, and given it its own repo https://github.com/wolfwood/tryadactyl/tree/fast-csg-test

kintel, 2024-11-28:
Closing as fast-csg has been deprecated

closed 2024-11-28 by kintel