openscad/openscad #4115: fast-csg does not render correct previews for one of my designs

wojciechka, 2022-02-12:
I have tried the nightly build from 2022-02-10, enabled `fast-csg` via preferences and rendering is faster, but previewing one of my designs is now showing glitches. Rendering works properly, so it is not a big issue, but thought I would report it here.

With `fast-csg` the preview looks like this:

[image]

Without `fast-csg` the preview renders properly:

[image]

I am not sure if it is a problem with `fast-csg` or my model lacking some 

Here is the design (it's a stand for my phone):

```
$fa = 1;
$fs = 0.1;

module DeviceModel(width=187, depth=6, height=146, round=10, addition=0.0) {
  translate([round - width / 2 - addition / 2, round - height / 2 - addition / 2, 0])
  linear_extrude(height=depth) {
    minkowski() {
      circle(r=round);
      square(size=[width - round * 2 + addition, height - round * 2 + addition]);
    };
  };
}

module Frame(width = 193, height1 = 18, height2 = 40, depth = 12, angle = 80, back_height = 1.12, back_depth = 50, round = 1.0, hole_width = 145, hole_depth = 25, cable_height = 10, cable_depth = 9) {
  shift1 = tan(90-angle) * height1;
  shift2 = tan(90-angle) * height2;
  translate([0, -back_depth / 2, 0])
    difference() {
      rotate([90, 0, 90])
        minkowski(convexity=10) {
          sphere(r=round);
          union() {
            linear_extrude(height=width, center=true, convexity=10) {
              polygon(points=[
                [0, 0],
                [shift1, height1],
                [shift2 + depth, height2],
                [depth + (shift2 * back_height / height2), back_height],
                [back_depth, back_height],
                [back_depth, 0],
              ], convexity=10);
            };
            translate([10, 0, -width / 2])
              cube([cable_depth, cable_height, width]);
          }
        };
      translate([-hole_width / 2, back_depth - hole_depth - round + 0.001, -round - 0.001])
        cube([hole_width, hole_depth + round * 2, back_height + round * 2 + 0.1]);
      translate([-width, -back_depth * 1.5, -20])
        cube([width * 2, back_depth * 4, 20.1]);
      translate([-width, -back_depth * 1.5, -20])
        cube([width * 2, back_depth * 4, 20.1]);
    };
}

module FrameHole(width=90, height=80, depth=21, size=40, offset=-17) {
  rotate([90, 0, 270])
  translate([offset, 0, -0.01])
  linear_extrude(height=width, center=true, convexity=10) {
    polygon(points=[
      [0, 0],
      [size, 0],
      [size - depth, height],
      [0, height],
    ]);
  };
}

module CableHole(width=100, offset=-11.75, depth=12, height = 8.5) {
  translate([-width / 2, offset, -0.01])
  rotate([90, 0, 270])
  linear_extrude(height=width, center=true, convexity=10) {
    polygon(points=[
      [-depth / 2, 0],
      [0, height],
      [depth / 2, 0],
    ]);
  }
}

difference() {
  Frame();
  translate([0, 0.1, 72 + 16])
    rotate([80, 0, 0])
    DeviceModel(addition = 0.5);
  FrameHole();
  CableHole();
};
```

wojciechka, 2022-02-12:
I have created a minimal way to reproduce it:

```
difference() {
  rotate([90, 0, 90])
    minkowski(convexity=100) {
      sphere(r=1);
      linear_extrude(height=100, center=true, convexity=10) {
        polygon(points=[
          [0, 0],
          [10, 40],
          [20, 1.2],
          [50, 1.2],
          [50, 0],
        ], convexity=10);
      };
    };
  translate([-40, 20, -1.1])
    cube([80, 35, 3.3]);
};
```

Removing `minkowski()` that I use to smooth out the frame with `sphere()` resolves it.

Here's how it looks like without fast-csg:
[image]

With fast-csg:
[image]

wojciechka, 2022-02-13:
> This has nothing to do with fast-csg which is used for F6 mesh calculation. This is a convexity issue with preview and happens exactly the same with the 2021.01 release.

Thank you, I am not an expert on internals of OpenSCAD.

When I tested it on the night build it was only happening with fast-csg enabled, which made me think it's the root cause. But I'm only using the tool, so I have no idea of internals and am confident in your judgment :-)

t-paul, 2022-02-13:
No problem, there seems to be some interaction with fast-csg. Maybe it just highlighted the issue. Thanks for the trimmed down test case, this helps a lot.

It also showed we were missing the `convexity` parameter in the cheat sheet, that's fixed now. The bug itself is a bit more obscure as the convexity given in the script seems to be passed down. It will need some more investigation where it's lost.

thehans, 2022-12-27:
Fixed by 6fbc677

closed 2022-12-27 by thehans