openscad/openscad #4119: search function is broken when searching lists of lists

acwest, 2022-02-13:
In Openscad version 2021.01, which appears to be the latest available in Arch Linux, if I try to use the `search` function to find a list in a list of lists, I get a warning and it returns the wrong value
```
_BOLT_SIZES = [
  ["M1.8", 1.8, 0.35],
  ["M2", 2.0, 0.4],
  ["M2.2", 2.2, 0.45],
  ["M2.5", 2.5, 0.45],
  ["M3", 3.0, 0.5],
  ["M3.5", 3.5, 0.6],
  ["M4", 4.0, 0.7],
  ["M4.5", 4.5, 0.75],
  ["M5", 5.0, 0.8],
  ["M6", 6.0, 1, 0.75],
  ["M7", 7.0, 1, 0.75],
  ["M8", 8.0, 1.25, 1],
  ["M9", 9.0, 1.25, 1],
  ["M10", 10.0, 1.5, 1.25, 1]
];

echo(search("M3", _BOLT_SIZES));
```
Results in:
```
WARNING: search term not found: "3" in file bug.scad, line 18
ECHO: [0]
```
I can see why it isn't finding "3", as it isn't there, but I don't quite see why it is searching for it.
In any case, if it CAN'T find it, why is it returning an index of 0? In my case, I would prefer it to return an empty list with no warning (as I generally work with `stop on first warning` set

acwest, 2022-02-13:
I will note that I get the same behaviour in 2022.02.13

UBaer21, 2022-02-13:
search term should be a list 
`echo(search(["M3"], _BOLT_SIZES));`

t-paul, 2022-02-13:
That has not changed since at least 2015.03.

closed 2022-02-13 by t-paul

acwest, 2022-02-13:
That is not how the function is documented
```
data=[ ["a",1],["b",2],["c",3],["d",4],["a",5],["b",6],["c",7],["d",8],["e",9] ];

search("a", data, num_returns_per_match=0);
```
What possible meaning would there be to have multiple search terms?

I will also note that if I change the code to:
```
_BOLT_SIZES = [
  [1.8, 1.8, 0.35],
  [2, 2.0, 0.4],
  [2.2, 2.2, 0.45],
  [2.5, 2.5, 0.45],
  [3, 3.0, 0.5],
  [3.5, 3.5, 0.6],
  [4, 4.0, 0.7],
  [4.5, 4.5, 0.75],
  [5, 5.0, 0.8],
  [6, 6.0, 1, 0.75],
  [7, 7.0, 1, 0.75],
  [8, 8.0, 1.25, 1],
  [9, 9.0, 1.25, 1],
  [10, 10.0, 1.5, 1.25, 1]
];

echo(search(3, _BOLT_SIZES));
```
it returns the correct value (including in the not found case

UBaer21, 2022-02-13:
if you search only 1 character (or number) this is different to searching a number 34 or string "34"  as a string is a list so it will search for each member of that list.  ⇒ hence searching for M and 3

acwest, 2022-02-13:
Well, re-reading the docs, this IS the correct behaviour. It's a little weird, but I can work with it :-)

t-paul, 2022-02-13:
Agreed, it should be replaced with likely multiple clearly defined functions.