openscad/openscad #4120: roof and offset issue for some combinations

UBaer21, 2022-02-14:
openSCAD version 2022.02.04

skeleton error when preview:
"ERROR: Skeleton computation error. /root/workspace/src/roof_vd.cc line 374: Voronoi error in file , line 9"

`for(i=[2:4],fn=[75:77])translate([i*10,(fn-75)*10])roof()offset(i/10,$fn=fn)difference(){square(5);circle(4,$fn=6);}`
When using  offset (.3,$fn=76)  - other shapes will show problems for other combinations of offset and $fn
[image]

CGAL Errors when render
"ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418"

```
for(i=[0:10],fn=[93:120])union(){
translate([i*10,(fn-100)*10])roof()offset(i/10,$fn=fn)difference(){square(5);square(4);}
cube(.001);
}
```

[image]

khumarahn, 2022-04-28:
https://github.com/openscad/openscad/pull/4240

khumarahn, 2022-05-01:
To the best of my understanding, this issue has two parts. I fixed the one related to
```ERROR: Skeleton computation error. /root/workspace/src/roof_vd.cc line 374: Voronoi error in file , line 9```

But
```Expr: e_below != SHalfedge_handle()```
is the F5-before-F6 problem: it can be worked around by cleaning the caches and pressing F6 right away before F5. In the particular example above, the offset shape has with many segments which are close to a circle
[image]
This leads to many close-by vertices in the Voronoi diagram. The renderer (F5) snaps these to a grid, and this destroys the geometry. I'm not sure how to solve this.

t-paul, 2022-05-01:
Yes, the F5 before F6 issue is separately tracked via #2991 and may need some cache rework.

MichaelAtOz, 2026-10-12:
Un-stalled. `roof()` is flaky.
OP 2nd condition persists with CGAL, worse. *Note: final CGAL union runs for a long time*.
Manifold works.

CGAL, console output seems to have missing bits
> Parsing design (AST generation)...
Saved backup file: C:/Users/me/Documents/OpenSCAD/backups/crash-backup-mhyNETNV.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.

It sat there with 999/1000 progress for a long time, then progress bar clears (to version), without extra console output. (but see below)

& more missing bits than shown in OP:
[image]

A subsequent F6 (after cache flush) had expected console output. I wonder if console is still flaky...
>Parsing design (AST generation)...
Saved backup file: C:/Users/me/Documents/OpenSCAD/backups/crash-backup-mhyNETNV.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.
Geometries in cache: 928
Geometry cache size in bytes: 15773672
CGAL Polyhedrons in cache: 309
CGAL cache size in bytes: 93378448
Total rendering time: 0:21:30.935
Top level object is a 3D object (Nef polyhedron):
   Simple:     yes
   Vertices:    23258
   Halfedges:   95824
   Edges:       47912
   Halffacets:  49992
   Facets:      24996
   Volumes:       172
Rendering finished.

MichaelAtOz, 2026-10-12:
I found a trigger for the missing console messages, do a file/save while it is still rendering. I think there may be an open issue re logging - else I'll raise another.

jordanbrown0, 2026-10-12:
Remember that when a render fails we cache the erroneous results but not the error messages, so when we hit the same pattern again we pull from cache and get bad geometry with no messages.

jordanbrown0, 2026-10-12:
Also, under circumstances that I don't remember at the moment we suppress consecutive duplicate messages.  If you do a save during processing then that generates a message, which means that the next message isn't a consecutive duplicate and is not suppressed.