openscad/openscad #4122: Right-click menu turns everything black

dannypike, 2022-02-16:
I've just noticed that the right-click menu doesn't work for me, when running under Windows 10. The OpenSCAD window goes completely black - no model, no axes, no console window, no program menu. The only thing I can see is the Windows Title bar, with an "X" at the top-right that I can use to close OpenSCAD.

This is a Visual Studio build from the code that I pulled from the master branch on Sunday 13th February.

If the right-click happens to lie over the model, then I get a context menu with the list of elements, but the rest of the window is completely black and stays that way when I release the mouse button.

The Output Window from the code shows no exceptions and no error messages from the code. OpenSCAD shuts down cleanly without any problems; it seems like it is still quite happy, but just not displaying anything.

Any ideas where the problem might be?

t-paul, 2022-02-16:
Duplicate of #3725, I'll close this one, lets continue there.

If you have any idea how to debug that, that would be great. It does not happen on Linux so I've not been able to investigate.

The right click triggers the special renderer with separate shaders for matching mouse position to code. I suspect that somehow binds a wrong buffer in OpenGL or something like that resulting in broken rendering after that.

closed 2022-02-16 by t-paul

dannypike, 2022-02-16:
OK, thanks. I thought I searched for other reports, but my brain is obviously not working well.

I'll have a look to see if I can find out something and post to that ticket.