openscad/openscad #4130: Dark mode plz.

jmerc77, 2022-02-19:
option for dark UI. maybe use the OS color scheme???

t-paul, 2022-02-19:
It should use the OS theme automatically. If there's a specific issue, you'll have to give more details.

closed 2022-02-19 by t-paul

Nikorasu, 2023-06-27:
Does anyone know a way to fix this?
I'm on Linux Mint 20, using the system Mint-Y-Dark theme, and most windows obey that fine.. However, the AppImage for OpenSCAD still uses a white theme for the window colors outside the editor/3d preview. (If I install OpenSCAD from the terminal, that obeys the theme, but that installs a version from 2019..)

[image]

Is there any way I can get a more recent version to work with the system dark theme?

t-paul, 2023-06-27:
For AppImage see #4105.