openscad/openscad #4134: Default values for parameters is unreliable

dannypike, 2022-02-21:
This is possibly an obscure case for most people, but I keep hitting it, so I thought I'd add it to the discussion about quirks in the syntax...

I think that the default values system for parameters has a significant bug that makes it an unreliable syntax for me.

Consider this:
```
    module a(test, arg_a = 123)
    {
        echo(str(test, "=", arg_a));
    }

    module b(test, arg_b)
    {
        a(str(test, "#1"));
        a(str(test, "#2"), arg_b);
    }

    a("test a");
    a("test b", 456);
    b("test c");
    b("test d", 789);
```

what I see in the console is this:
```
ECHO: "test a=123"
ECHO: "test b=456"
ECHO: "test c#1=123"
ECHO: "test c#2=undef"
ECHO: "test d#1=123"
ECHO: "test d#2=789"
```

All of the cases are legal syntax and I have situations in which I might want to use them in my models. I do tend to use lots of small shared modules, but the `undef` in test case `c#2` means that I should never rely on the default value for `arg_a` when inside `a()`.

Instead, I must always use this code:

```
    module a(test, arg_a)
    {
        arg_a = is_undef(arg_a) ? 123 : arg_a;
        echo(str(test, "=", arg_a));
    }
```

which is rather ugly.

Is this a "known" problem? Can we fix it? I would much prefer to use the "default value" system, if we can plug this hole; it's much easier to read and maintain.

jordanbrown0, 2022-02-21:
This behavior can be demonstrated more simply like so:
```
module a(arg=123) {
    echo(arg=arg);
}
a(456);
a();
a(undef);
```
which yields
```
ECHO: arg = 456
ECHO: arg = 123
ECHO: arg = undef
```

The rule is simple:  explicitly supplying "undef" as an argument does not trigger default-argument processing.

Indeed that's sometimes annoying, when you want to pass through "use the default" behavior.  But it's not silly behavior.

nophead, 2022-02-21:
Seems correct to me. In module b, arg_b has no default so is undef. You then pass undef as the arg for a, so it doesn't use the default.

dannypike, 2022-02-21:
@nophead You're right, my example is not a good one because I was trying to simplify things for this ticket, and the act of doing that broke my example.

The problem is that I have no way to pass "use default value for arg_a" in a call to `b()`, because passing `undef` doesn't get interpretted as `missing`, as per @jordanbrown0 's comment. He explains much better than I did.

How to I call `b()` in such a way that `a()` uses its default argument without adding extra variables to handle that one case?

nophead, 2022-02-21:
You would need to call `a()` with no parameters when `arg_b` is not defined. Or in `a()` check for the parameter being undefined and set it to the default and then initialise it with `unde`f.

jordanbrown0, 2022-02-21:
I have a function
```
function default(val, def) = is_undef(val) ? def : val;
```
that I use for a lot of defaulting cases - in particular, where the default is derived from other arguments.  It would make explicitly setting the default in `a()` slightly less ugly.

dannypike, 2022-02-21:
@jordanbrown0 That is the sort of thing that I need to do, too. However, I think that it would be better for default parameter values to interpret `undef` as being the same as `missing` in OpenSCAD.

The current implementation sets the argument values based on how deep the call stack is, and I think that's not "correct behaviour" for something like this. I fear that it leads to weird subtle bugs when the model gets more complex, which is exactly the time when you don't want to have to analyse weird and subtle behaviour!

jordanbrown0, 2022-02-21:
I don't entirely disagree, but it does beg the question of how you would explicitly specify "undef" to override a default.  Maybe you shouldn't structure your arguments so that you would want to do such a thing, but it's legal.

dannypike, 2022-02-21:
I can't think of a case where I would want to interpret "use an undefined behaviour" as being different from "use a default behaviour". Have you got a particular example in mind?

It's a difficult thing to get right for all cases. If people like to pass `undef` explicitly, then maybe we could have `missing` as well as `undef`. I believe that more formal languages make that distinction to cover this sort of thing, but I'm not asking for us to do that. It's much better to spend our time on implementing other stuff.

I only want default parameter values not to behave differently based on how many modules are being called. Otherwise, I always have to use the `undef` gate that you showed above, and then they have no reason to exist in the language at all.

jordanbrown0, 2022-02-21:
The use case for explicitly specifying undef to override a default is where the default is to do something, and you want an option to not do it.
```
// Build a cube optionally with a sphere on top.  Cube defaults to 10,
// sphere defaults to radius 10, set sphere radius to undef to suppress
// sphere.
module cube_maybe_with_sphere(cube_size = 10, sphere_radius = 10) {
    cube(cube_size);
    if (!is_undef(sphere_radius)) {
        translate([0,0,cube_size+sphere_radius]) sphere(r=sphere_radius);
    }
}
```

You could do the same thing by saying that `sphere_radius=0` suppresses the sphere, but `undef` more clearly expresses the "not-ness" of the request.

JavaScript has `null` and `undefined`, with something of the distinction that you make, but since you can explicitly set something (notably an object member) to `undefined` you end up with **three** variations of "not-ness" - not defined, defined explicitly to undefined, and defined to null - not counting zero, empty strings, and zero-length arrays.  People's heads tend to explode with even one variation of "not-ness".

I don't know whether it's a distinction drawn in the OpenSCAD implementation, but another axis is whether the undefined-ness is at compile time, or at run time.  In a language like Java, where you can have a function that has a variation with one argument and a variation with two arguments, that distinction is drawn at compile time.  There's no way to start with two arguments and, at runtime, call the one-argument variation.  (Other than by explicitly testing and having two distinct calls, of course.)  I suspect that C++ is the same, but I'm not enough of a C++ person to be sure.

Anyhow, if we were starting from zero I'd probably recommend doing it the way you say, that a missing value or an `undef` value both trigger the defaulting mechanism, but... we're not starting from zero, and the current behavior is not totally silly, and people may well be depending on it.

dannypike, 2022-02-21:
Thanks for the example. I would say that it is "better" not to specify something when you don't want one, and to require the caller to say how "big" the something is, when it's optional. Saying I want an "undefined thing" feels even more weird to me.

But I get your point about not wanting to break existing stuff, which takes precedence over this. I'll avoid using default parameter values and "solve" the problem that way.

closed 2022-02-21 by dannypike

jordanbrown0, 2022-02-21:
I wouldn't think of it as "I want an undefined thing", but rather as "I don't want a thing".

dannypike, 2022-02-22:
Yes, it's a point-of-view thing. Although I have spent 30 years defining APIs, I don't have enough history in the OpenSCAD "domain" to say anything stronger than "this feels wrong to me".

So, I closed the ticket and will act as though default parameters aren't a part of the language at all. That seems the best way forward. Everyone has a system that works for them.

lgarron, 2025-03-03:
This issue appears to be resolved, but I thought I'd comment that this has also been confusing.

I find it common to make composable functions where the same default parameters are used by multiple functions in an API. Here is a simplified example:

```
module engrave(engraving_depth = 0.4, engraving_offset = 0)
{
    translate([ 0, 0, -engraving_depth ]) offset(engraving_offset) children();
}

module engrave_text(text_string, engraving_depth = 0.4, engraving_offset = 0)
{
    engrave(engraving_depth = engraving_depth, engraving_offset = engraving_offset) text("hi", halign = "center", valign = "center");
}

difference()
{
    translate([ 0, 0, -5 ]) cube([ 100, 100, 10 ], center = true);
    engrave_text("hello");
}
```

If I want to `engrave_text()` not to pick its own default parameter values, I could set them to `undef`. But now this also affects `engrave()` — even if `engrave()` didn't really sign up for this. Using the workarounds from this thread, this leads to:

```

function default(v, default_value) = is_undef(v) ? default_value : v;

module engrave(engraving_depth = undef, engraving_offset = undef)
{
    translate([ 0, 0, -default(engraving_depth, 0.4) ]) offset(default(engraving_offset, 0)) children();
}

module engrave_text(text_string, engraving_depth = undef, engraving_offset = undef)
{
    engrave(engraving_depth = engraving_depth, engraving_offset = engraving_offset) text("hi", halign = "center", valign = "center");
}

difference()
{
    translate([ 0, 0, -5 ]) cube([ 100, 100, 10 ], center = true);
    engrave_text("hello");
}
```

This leads to a situation that incentivizes setting parameters to `= undef` everywhere and defining defaults inside the functions themselves. This is not *difficult* with a bit of discipline, but:

- The lack of default values in the function signature makes makes it harder to communicate default values that are really "part of the API". It's possible to duplicate them in comments, but this is not DRY and thus can get out of sync more easily by accident.
- This is inconsistent with other languages, particularly my most familiar one (JavaScript). I would constantly be fighting "muscle memory".
- I either have to implement variations of `default()` (and be careful about shadowing/overwriting), or implement it in a common place and import it from everywhere in my library files. In the latter case, this means `default()` would leak into the global scope *often but not always*, which can cause unexpected surprises while refactoring.
- I either need to keep a whole bunch of `default(…)` code in sync throughout a function, or add boilerplate to each function for each parameter like this:

```
module engrave(engraving_depth = undef, engraving_offset = undef)
{
    engraving_depth = default(engraving_depth, 0.4);
    engraving_offset = default(engraving_offset, 0);

    translate([ 0, 0, -default(engraving_depth, 0.4) ]) offset(default(engraving_offset, 0)) children();
}
```

It would be a lot handier to have something more intuitive. Perhaps a syntax variation could ignore `undef` values in favor of the specified default:

```
module engrave(engraving_depth := 0.4, engraving_offset := 0)
{
  // …
}
```

As a mitigating measure, a built-in function like the `default(…)` function would at least avoid some of these issues for me.

jordanbrown0, 2025-03-03:
I wouldn't object to some mechanism that causes `undef` to yield the default value, but it's not important enough to me personally to work on it.

lgarron, 2025-03-05:
> I wouldn't object to some mechanism that causes `undef` to yield the default value, but it's not important enough to me personally to work on it.

Sounds good. If I were to propose something, I'd probably try to propose a `|=` operator that can be used either for as a statement or for a parameter:

```
module engrave(engraving_depth |= 0.4, engraving_offset |= 0)
{
  // …
}
```

I suppose the appropriate steps would be a new issue and a PR for that?

jordanbrown0, 2025-03-05:
That's certainly the most promising approach.  However, a caution:  the project is very cautious about accepting language changes.  They generally need to be *very* compelling.

jordanbrown0, 2025-03-05:
Clarification:  a PR is the most promising approach.  I have no real opinion on the syntax.