openscad/openscad #4136: Color-coding of axes

BenjaminPoilve, 2022-02-24:
Hello, 

This a little enhancement proposition.
I am a relatively new user and one of my issues with the editor if finding which of the three direction I am currently editing. 
One solution would be to color-code each axis (X,Y,Z) and to have each number in the interface be of the color of the corresponding axes. 
This might be a little harder to code when you have rotation to still map on the right axes, and of course once you step out of 90° angle this all goes out the window, but I feel this would quite significantly ease the onboarding. 

Thanks for this software, it became truly indispensable to me !

jordanbrown0, 2022-02-24:
Are you talking about:
* Color-coding the actual axis lines and associated numbers, in the preview window?
* Color-coding the translate and rotate numbers, in the preview window?
* Color-coding the x, y, and z in `translate([x, y, z]) ...`?

Have you noticed the small X-Y-Z key in the lower-left corner of the preview window?

BenjaminPoilve, 2022-02-24:
I was referring to the third option: using the colours of the XYZ key to "syntax highlight" the code in order to help writing it. 
This might be a specific need, but I often find myself lost (especially after a few 90° rotations) and I feel this could solve this issue.

jordanbrown0, 2022-02-24:
I suspect that even the trivial case of
```
translate([x,y,z]) ...
```
would be difficult because the editor doesn't really understand OpenSCAD; it's really an editor for other languages that has had OpenSCAD keywords pushed into it and OpenSCAD is similar enough that things like comments and strings are recognized.

But I can't even see how it could help with rotations in the picture.  Translate is relative to the local coordinate system.  The first coordinate is X, the second coordinate is Y, and the third coordinate is Z.  Trying to have it somehow unroll child or parent rotations to mark which coordinate transforms into the world XYZ axes would mean subverting that fundamental concept; it would make it *harder* to understand what's going on.  The fact that, as you say, any off-axis rotations would make it meaningless emphasizes this problem.

And the editor doesn't come close to recognizing the higher-level patterns that it'd have to recognize to do this highlighting.

---

You might build a little axis module that you could drop in to help you out when you get lost, something like

```
module axes() {
    size=100;
    color("red") cube([size,1,1], center=true);
    color("green") cube([1,size,1], center=true);
    color("blue") cube([1,1,size], center=true);
    for (r = [
        [0,0,0],
        [0,0,90],
        [0,-90,0],
    ]) {
        rotate(r) linear_extrude(height=0.1, center=true) {
            translate([10,1,0]) text("+", size=3, halign="center");
            translate([-10,1,0]) text("-", size=3, halign="center");
        }
    }
}
```

Any time you get lost, temporarily add in a call to axes() and it'll show you which direction the axes are in that particular context.

ChrisCoxArt, 2022-02-24:
OpenSCAD is not using another language for color coding syntax anymore - we have our own lexer. But it is just a lexer - it highlights words, not constructs/trees (which would be a bit difficult).  And yes, rotations before and after the fact would be even more difficult to detect and highlight, especially because a module defining a shape can be used with multiple different rotations.
For example, see https://www.thingiverse.com/thing:2960213  and the scad file-- test_core can be used aligned to X, Y, or Z depending on the options chosen.

thehans, 2022-02-24:
Closing this as not feasible for the reasons given above.

closed 2022-02-24 by thehans

BenjaminPoilve, 2022-02-25:
Obviously did not think this all the way through ! Thanks for the thorough answers and consideration !