openscad/openscad #4140: Closing with one unsaved tab says "Some tabs have unsaved changes"

jordanbrown0, 2022-02-24:
... which is a little confusing if you only have one tab (and so don't even have a tab bar visible).

ChrisCoxArt, 2022-02-24:
Technically it is correct English.  But it is odd to use the generic plural to refer to a single document.
Right now the code always uses the same message when it encounters the first unsaved document - and does not check the number of tabs open or unsaved.  Since the document/tab count is rarely very high, I think we could check all to get a count before posting the dialog.

And right now you're lucky to even get the message ( https://github.com/openscad/openscad/issues/3971 ).

ChrisCoxArt, 2022-02-24:
Or we could simply change the wording to "One or more documents has unsaved changes"

jordanbrown0, 2022-02-25:
Yes to all of that.

"Some of your documents ..." has the same problem of being awkward when there's only one document.

My preference would be to handle the one-tab case specially, with "You have unsaved changes" or something like that.  I can propose verbiage if desired.

I'm also happy to do the actual work.

ChrisCoxArt, 2022-02-26:
I think I've got the code ready, I just need some time to do more testing.