openscad/openscad #4143: PDF export defaults to A4 paper with no way to change it

amatulic, 2022-02-25:
It's nice to see PDF export in build 2021.01. However, it seems stuck on outputting A4 size pages.

European size A4 paper is not readily available in the United States. When exporting a 2D design to PDF, the purpose is to get a drawing with a 1:1 scale. When printed on US letter-size (8.5x11 inches) paper, the bottom and right edge of the drawing are cropped off, because A4 paper is larger than US letter size. Scaling to fit the page defeats the purpose of having a 1:1 scale drawing.

There should be a setting available for default PDF output size.

Now, I designed my drawings to fit letter-size paper, so you might think, what is the problem with cropping? The problem is that the axis scale gets cropped off the bottom, and this is critical for calibrating the laser printer's scaling when printing from Adobe Acrobat.

JonMcLin, 2022-10-06:
I have been thinking about implementing this for my own purposes.  
Is this the right place for discussing the implementation?

I'm considering two additional features, and I also have some questions about best-practices with respect to some implementation decisions.

Features:
1. Add paper selection.  Probably a pick-list, A4, A3, Letter, Tabloid.   Later could add additional sizes.
2. Add selection of landscape vs format.  (I could automate this based on the bounding box proportion, but it would seem better at least initially to make it selectable, along with paper size)
3. Generalize the Origin position behavior for finer granularity.  Rather than bottom-left or center, place the origin at the center of the bounding box, to center the geometry on the paper.

I am not at all familiar with Qt beyond what I've read.  I have been going through the source code, and I've generally mapped out what I need to do.  But... 
-   I understand I need to add the above widgets to ui/preferences.ui (which I am assuming I use Qtcreator for; I'll figure that out), and add equivalent setValue method/action to the preferences.cc (or is that code inserted by Qt in the build process?)
-  It's not clear where the best place to get these values from settings would be.  For other export actions, the get methods where settings are needed is in the mainWindow.cc code, and there are no gets in the export*.cc code.  But that's because they all can use a common export signature.    So should I put the settings get calls for this purpose in the exportPdf.cc, or change the export method signature and put the gets in mainWindow.cc.

Thoughts?

t-paul, 2022-10-06:
@JonMcLin Yes, discussion here is fine, if you like to have a chat, feel free to join the IRC channel too.

I don't think Preferences is the right place for this. Following the more general discussion in #649 this should go into some sort of export dialog. It's fine to start out with just PDF but it should be open to extension for most of the other formats.
It's important to keep the UI separation in mind, it's possible to compile OpenSCAD in headless mode (e.g. for Web usage (https://ochafik.com/openscad/) so the dialog would go somewhere into MainWindow (well, likely a new class created from there) collecting the information and passing it to the actual export code in a way that could later be added to command line too.
Sorry if that sounds complicated, but as mentioned it's fine to start with a relatively specific solution, we just need to structure it in a way to extend later.

JonMcLin, 2022-10-06:
> this should go into some sort of export dialog.

I agree, but Preferences already has an Export GroupBox (with "Default to ASCII STL" as a checkbox.

So I figured that expanding on this current would allow quick implementation - I just need to add a couple of items to an existing GUI, rather than create a new one.  And in [#649](https://github.com/openscad/openscad/issues/649), the discussion seems to be on creating a general Export GUI.   That doesn't feel like where I should start...  

(Or am I misunderstanding that discussion ?  My first thought was to pop-up a dialog specific to Export PDF.   There wouldn't be much on it).

t-paul, 2022-10-06:
Yes, the STL option should move to the (not yet existing) dialog too eventually. Right now that's an option which is unlikely to change a lot as most people probably do not care or even know about binary vs. ascii STL. Pretty much the only visible user impact is the resulting file size.

For PDF it seems much more likely that the options will change across different designs and exports and in addition to the options mentioned above, it would be nice to include at least 2 additional ones for
* enable / disable the info watermark
* enable / disable the rulers

Having all that in the already huge Preferences dialog is not ideal (and

A first go with a PDF specific pop-up dialog seems fine. It should be possible reuse that later if we add options for other exports. Just as mentioned it's not possible to call the dialog from the core export code, but needs to be opened in the GUI code collecting the data to pass to the core export.

JonMcLin, 2022-10-09:
Got a dev environment set up and tested, and have verified a (hard-coded) "Letter" size.  So I at least have the opportunity to be successful...

I'm first going to test/verify a new function signature, and add a mapping to the existing signature.  This will provide the means to allow the parameters which are now file constants to originate in the dialog box.

Is there an OpenSCAD coding standard or style guide?

t-paul, 2022-10-09:
There's no written style guide, but we have `scripts/beautify.sh` which runs `uncrustify` on top-level from the repo based on the [.uncrustify.cfg](https://github.com/openscad/openscad/blob/master/.uncrustify.cfg).

When run in a local branch, it will only touch files that are modified in that branch.

JonMcLin, 2022-10-13:
Control code implemented and tested, still have a little implementation logic to code (I have pseudocode).  And then the Qt dialog...

The export_pdf function has an added signature, with a single additional parameter:
struct ExportPdfOptions {
    bool showScale;
    bool showScaleMsg;  // Dialog should set this to FALSE if showScale is FALSE.  No harm if not though.
    bool showDsgnFN;  // scad filename
    paperOrientations Orientation;
    paperSizes paperSize;
};

For testing, the original signature is  now a new function which which simply adds the above structure, assigns values (for testing) and calls the modified original function with the new signature.  

Hopefully the names of the Booleans tell you what they do.  The other two are control enums:
// Move to header
enum paperSizes {
 A4,A3,LETTER,TABLOID
};  // Add "LEGAL"?

// Move to header
enum class paperOrientations {
PORTRAIT,LANDSCAPE,AUTO
};

paperSizes is a straight C enum rather than class/structure based to allow it to be used as an index into the array of paper sizes without adding a lot of code.  Since the usage of this is extremely scope-limited, I didn't feel it necessary (especially at this stage) to add the code overhead to use a scoped enum as an index, or to use a map.

// rows map to paperSizes enums
// columns are Width, Height
const int paperDimensions[4][2]={
{595,842},
{842,1190},
{612,792},
{792,1224}
};  // Add LEGAL?

Testing verified the flags work, and the paper size/orientation enums are processed by the code.  I've not yet updated the axis and geometry-draw code to map to the new orientations, but I do have the pseudocode ready to implement.

Comments solicited.

jordanbrown0, 2022-10-13:
Just looking at the design (and not the implementation)...

Are you saying that `showScaleMsg` is only meaningful if `showScale` is true?

Do we think there are other consumers for page dimensions?  SVG, maybe.

Suggest that the actual exporter accept dimensions, not an enum.  Have the list be in the UI.  That way in the future we could maybe have a preference for the user to define paper types, and the paper types could be more readily shared between exporters if that is ever an issue.  I'm not suggesting that you implement that custom-paper-type UI now, only suggesting that if you pass dimensions to the exporter now then it won't need to change if we ever *do* have one.

jordanbrown0, 2022-10-13:
I'm not asking for a more general GUI, just trying to plan for the future.  If the exporter accepts page dimensions as an input parameter, then in the future the GUI could be expanded to offer custom page sizes without having to change the exporter.  Take the existing enum, which already needs to be visible to the GUI, and put it solely in the GUI.  Take the existing table of sizes, and move it to be solely in the GUI.  Then instead of passing an enum to the exporter, pass the dimensions.

JonMcLin, 2022-10-15:
Functional logic completed, and I've attached a few test outputs (PDFs)
I did add two additional features:
-  A "grid" at 10mm, in addition to the scale.  This is separately selectable.
-  An additional paper size (legal), since this actually scratches my specific itch particularly well - I need to be able to test some patterns that are about 12" wide, so with this size I don't need a tabloid printer.

Summary of features:
- enable/disable for scale, scale message, design file name, and grid.
- Selectable paper size
- Selectable paper orientation, including "auto", where landscape is selected if the X span of the bounding box is greater than the Y span of the bounding box.
- auto-centering, where the geometry is centered on the page.

Files:
TestScale.pdf (Scale and Scale Message enabled)
TestScalenGrid.pdf (Scale, Grid, and Design Filename enabled)
TestGrid.pdf  (Grid and Design Filename enabled)

These are all in Letter, Portrait (although Orientation = "AUTO" was selected for all, the specific geometry results in Portrait).
[TestScale.pdf](https://github.com/openscad/openscad/files/9793298/TestScale.pdf)
[TestScalenGrid.pdf](https://github.com/openscad/openscad/files/9793299/TestScalenGrid.pdf)
[TestGrid.pdf](https://github.com/openscad/openscad/files/9793300/TestGrid.pdf)

Now to build a dialog box...

JonMcLin, 2022-10-16:
This UI implementation looks to be a bit challenging.   As I understand it, the new dialog must be called from the gui context, which essentially means the MainWindow.cc code.    The challenge I see is that the interface (function signature) is normalized early in the call chain, which would mean that adding the new parameters either affects lots of code, or requires duplicating that code with a special "Export PDF" variant.  The latter is not hard, but it seems wrong.  Maybe someone has some thoughts on this.

The general export flow is as follows:
1.  Each export action produces a specific signal.  In this case, it is "fileActionExportPDF".
2. The signal is connected to a corresponding slot.  In this case, it is "actionExportPDF()"
3.  The slot function is trivial, just populating a handful of attributes and then calling a common actionExport() method.  So in this case, it's   actionExport(FileFormat::PDF, "PDF", ".pdf", 2);  A file format enum, a couple of strings related to the file format, and the number of dimensions.   I could pop up a dialog here, but I've got no place to put the data...   The signature is:  
void MainWindow::actionExport(FileFormat format, const char *type_name, const char *suffix, unsigned int dim)
4.  actionExport is the departure point from MainWindow, into export.  It basically does three things: (1) gets the filename,  (2) populates the ExportInfo structure, and calls the exportFileByName() function in the export.cc code.
5. exportFileByName is a simple dispatcher, testing one member of exportInfo and calling one of two other exportFile functions (this is "std out" vs "stream"   (I might collapse these three into one, since the stream also does some dispatch processing in addition to opening the file.)
6. exportFileByNameStream  opens the file appropriately, then calls exportFile() with the streamptr added to the signature.
7. exportFile() here is a dispatcher, testing the format to decide which specific export function to call.   This is where I'd be making the call to my new exportPDF signature.

So how best to get my exportOptions to exportPDF?  Notionally, they should be added to the structure exportInfo.   The problem is that not all export types will need options, and even if they did, they wouldn't need the *same* options.   So if for example the new member of the structure was a reference to an export-specific structure, it would need to be subclassed from a common class.

Or at least I think...     Am I missing options?  As noted earlier, this is the first time I've written c++, so this seems I have plenty of opportunities to discover dead-ends.

As for the actual dialog.  There are really only two places it can go.  If it goes in the slot function, it's appropriately placed to be export-specific.  We'd add the "exportOptions" (top-level class) to the actionExport method signature, and it would add it to the exportInfo structure.   Alternatively, another set of dispatch code is added to actionExport to get the appropriate set of options.  (Or, the dialog itself has the dispatch feature, only showing the pane determined by the format type.).

Anyway, I'd welcome suggestions on a forward.

t-paul, 2022-10-16:
As discussed earlier, it's ok to keep it simple for now, so adding the dialog to the action handler `actionExportPDF()` and extending the `ExportInfo` with a PDF specific structure is fine. I'm not yet sure what the best way will be for the future (maybe attaching the information to the `FileFormat` information?), but we can figure that out when adding the next set(s) of options.

JonMcLin, 2022-10-26:
Still cogitating on implementing the dialog box and I have an incremental plan.  But I decided to make a couple of changes to the export_pdf code.  The interesting one is the grid size is now selectable.  The interface allows it to be any size (in mm) of 2 or more, but I intend to make it a discrete (enum) selection in the dialog, with 2, 5, and 10 as options.  It was previously fixed at 10mm.

Here's an image of the test geometry export with a 2mm grid:

[image]

JonMcLin, 2022-10-28:
I have a couple of uncertainties with respect to integrating a QT dialog which I have been unable to ascertain from available documentation and the project structure.   Perhaps someone can point me at a HOWTO resource, a document, an example, or provide a short bulleted list of what to do.

The easier one:
-  Is there anything I should do in my initial implementation to facilitate creating localization (translations)?  Eg, a naming convention, using constants vs. literals vs. variables for text strings, etc.. ?  

The more complex one:
-  What do I need to do to (if anything) to incorporate the dialog into the build system?  (I think I've figured out the "how" to implement it.)

I have perused the build directory and source and ascertained that the project is NOT using dynamic UI loads (I found the compiled ui headers and moc files in the autogen dir), although I've not been able to find in the make artifacts where uic and moc are called.   So I don't know if the build system will just find the .ui files and automagically process them, or if I have to add them to a build list somewhere.   So right now I'm stuck assuming it's automagic.

Here's what I *think* I need to do:
-  generate an "exportPdfDialog.h" header file.
-  #include in that header file "ui_exportPdfDialog.h".  I will not generate this; it will be autogenerated during the build.
-  #include that header file in the mainwindow.cc source.
- generate my "exportPdfDialog.cc" file with the implementation logic  of the dialog.
- generate (using QT Designer) the exportPdfDialog.ui file
- add to the mainwindow.cc file the exportPdf method modifications to instantatiate the dialog, populate the exportPdfOptions structure, and probably set (and get) with the existing Settings object.
- Any new signals and slots are probably contained within the dialog since I'm integrating the show and get of the dialog within a single method of the existing call chain.

Is this generally correct, or am I missing anything obvious?

Thanks in advance!

amatulic, 2023-01-24:
I never dreamed that my initial request to have a selectable paper size would end up being so complicated, and that implementing this feature requires building a whole new framework for dialog inputs. It's beyond my ability to help, and I appreciate the efforts so far.

JonMcLin, 2023-02-06:
**Design Notes for Review**

<!DOCTYPE html>
<html>
<head>
	
	<title></title>
	<meta name="generator" content="LibreOffice 7.3.7.2 (Linux)"/>
	<style type="text/css">
		@page { size: 8.5in 11in; margin: 0.79in }
		p { line-height: 115%; margin-bottom: 0.1in; background: transparent }
		a:link { color: #000080; so-language: zxx; text-decoration: underline }
	</style>
</head>
<body lang="en-US" link="#000080" vlink="#800000" dir="ltr"><p style="line-height: 100%; margin-bottom: 0in">
Status:</p>
<p style="line-height: 100%; margin-bottom: 0in">-	core functionality
implemented within exportPdf.cc</p>
<p style="line-height: 100%; margin-bottom: 0in">-	dialog box
designed using Qt Designer</p>
<p style="line-height: 100%; margin-bottom: 0in">-	dialog box
“integrated”  into project file structure (meaning: I now know
how to compile it).</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">Now I need to
implement the UI logic.  That means incorporate the dialog into the
mainwindow.cc gui code such that the user can actually select the
options at run-time.  Before I do so, I’d like to describe what I
currently expect to do such that others more knowledgeable can offer
constructive criticisms to minimize wasted efforts.  And thus this
discourse.</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">I think there are
three main things to describe:</p>
<p style="line-height: 100%; margin-bottom: 0in">-	Incorporation of
the “options” data structure</p>
<p style="line-height: 100%; margin-bottom: 0in">-	where and how I
will incorporate the dialog box</p>
<p style="line-height: 100%; margin-bottom: 0in">-	Options data
persistence</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">1.  Incorporation of
the Options data structure.</p>
<p style="line-height: 100%; margin-bottom: 0in">Previously, I
intended to add this to the existing ExportInfo structure.  At this
time, I think it best to make it a separate parameter
(ExportOptions).  This will allow incremental incorporation into the
export call processing, having different signatures with and without
the extra parameter.  The “without” will simply call the “with”
adding a nullptr.  We only have to touch methods/functions which are
in the “export” call stack.</p>
<p style="line-height: 100%; margin-bottom: 0in">ExportOptions will
be passed by reference.</p>
<p style="line-height: 100%; margin-bottom: 0in">As discussed
earlier, VARIANT (BOOST version) will be used, although initially the
only value will be a of the ExportPdf form.</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">For reference,
here’s that call-stack, discussed earlier in the thread.  Columns
starting at three are types of exports.  Rows starting at four are
functions, in sequence.   Items with no entry in Menu, Action, and
Dims mean I was not able to ascertain how they apply.  Eg, no “menu”
[... 57 lines cut ...]
would seem to be the most likely), and the scale usage statement.  So
for now, I will skip the integration with Preferences.</p>
<p style="line-height: 100%; margin-bottom: 0in">What I will do is
make the exportPdfOptions structure static, so it persists through a
run of the application.  The initial settings of the dialog on first
execution will be defaults, but subsequent exports will retain the
previous settings from this run.  This is probably what users would
expect.  In any event,  once it’s working I’ll decided which if
any preferences need to be persistent between runs via preferences.</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">If you have opinions
on any of this, please advise.  I want to ensure this can be
available in the headless/command-line mode, and I’m wondering how
preferences work in that case since Preferences are within the gui.</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
<p style="line-height: 100%; margin-bottom: 0in">Thanks in advance!</p>
<p style="line-height: 100%; margin-bottom: 0in"><br/>

</p>
</body>
</html> 

JonMcLin, 2023-02-19:
Well, I have a 95% solution.

Everything is implemented and works, *except* for the setters and getters associated with the radio buttons (which are used for the enumerations).  Booleans work fine, and are persistent within a single run of the application because the "setter" is used to restore the last setting from the persistent ExportOptions data structure.   The enums not working means one can't set the paper size, orientation, or grid size.

[image]

I attempted to use the QT buttongroup for the radiobuttons.  That seems to be useless except in the case where where the buttons live in a layout group with other buttons. 

I found exactly one Qt radio button example that provided for a setter when a buttonbox is used.  It does not compile, at least with Qt5.

If someone can provide good example, I'd appreciate it.  Else I will revert to brute force and use straight C-style code.

JonMcLin, 2023-02-20:
Working code has been pushed back to my openscad fork (https://github.com/jonmclin/openscad).
Note that I did the merge on my fork  *after* I committed it (oops), so you'll need to look back one change to see my changes highlighted. ("Implemented ExportPdfDialog...", or something like that.)

It still has a lot of unused or commented-out code to be removed, and I need to do some additional polishing.  However, it works fine with the following issues (when I compile it):
-  A Qt warning is sent to stderr when the dialog box "Ok" or "Cancel" button is pressed in the new dialog box, stating no signal is found.  But the connect() method is called in the ui code, and it works fine.  Needs investigation.
-  LOG messages are no longer being published to the Error-Log window (this seemed to occur about the time I started making changes to mainwindow.cc.  I don't know if I broke something in that file, or did something to the build environment.  I do know that the compile is different now, and the date in the status bar is now current as opposed to the last release.  Needs investigation. 

Contrary to what I stated in the "Design Notes for Review" post above, I decided to embed the Options into the ExportInfo structure.   This minimizes the changes to other code while still supporting addition of additional dialogs.   A pointer is used, although if the createExportInfo is modified to insert it at creation it might be possible to use a reference (need to think about that some more).   The ExportPdfOptions structure is declared in the exportActionPdf function, but as "static" so it has persistence within a run of the app.   

Also, I have not yet implemented the options as a variant, given that ExportPdf is the only current user.
In any event, the pointer can be used for a safety-check in a user function:  check for nullpointer (and once it's a variant, which() type), and if its empty or the wrong type, use a "default" options block.   

Anyway, comments solicited!

I have not make a pull-request to the main repository, assuming people might checkout the code from my repository.  If I should make that pull-request, please advise.   I'd guess it could be pulled into the main openscad as a separate branch, but I'm not familiar with the workflows being used.

JonMcLin, 2023-02-26:
#Implementing a "Magic Dust" API#

In addition to cleaning up the code, I've been looking at incorporating persistence via Qsettings.   This looks (mostly) trivial, the semi-complicated part is incorporating the enums.  Although it looks like since Qt 5.5 I just need an extra declaration vs a set of string to/from enum methods.

But what jumps out at me from this is the following:  Using Qsettings, we can eliminate the need for an ExportOptions structure:  simply have the dialog box populate settings in Qsettings, and have the exportPdf function read those settings from Qsettings.  *No parameter passing required*.   

This just feels wrong, but it greatly simplifies implementation (at the cost of future maintainers scratching their heads a lot until they understand Qsettings...)

Any thoughts on a path forward?

JonMcLin, 2023-02-26:
Offline, I've been reminded this must work "headless"; without Qt.  That means reliance on Qsettings to connect the dialog to the export function (in lieu of parameter-passing) would break functionality.

The structure shall remain...

JonMcLin, 2023-03-03:
I've successfully implemented Qsettings interfaces for the Boolean parameters, but the Enums are problematic.  (Nothing I've tried will successfully compile within Openscad, only in proof-of-concept applications.)

Qt, Boost, and C++ std are incompatible, and I've been unable to come up with a solution that that will compile.  All solutions work in isolation, but the enums exist in both the GUI code and the non-gui code, which heretofore do not both compile.   Conceivably I could revert to a pure C implementation, but even that's difficult since the compiler seems to complain about many pure C constructs.  Since this involves enums and strings, both of which C++ altered, I do not think I can be successful.  The avenue which seems to be suggested on stackexchange is to "shadow" the c++ enum with a QT Enum class and namespace - eg, duplicate it.   I'll end up with 200 lines of code for something which should take 2 lines.  I think I'd be more likely to be successful with assembly!

Are there any successful examples of enums used across openscad in the GUI, Qsettings, and the core code?

JonMcLin, 2023-03-03:
I've managed to get settings working properly with an assortment of additional type conversions.

The implementation will implement the same default settings as currently implemented, but once they are changed via the dialog box, the values are stored in Openscad settings and will be used (and updated) with all subsequent executions of PDF export.

Here's what showed up in my local settings files after the last test execution:
[exportPdfOpts]
gridSize=2.5
orientation=Auto
paperSize=Letter
showDsgnFN=false
showGrid=true
showScale=false
showScaleMsg=true

Reasonably-descriptive names, set off in an appropriately-named group.  This should be an easy pattern to replicate if/when options are added to other Export methods.  (I'm wondering if this would be directly appropriate for DXF?)

Anyway, a little more code cleanup/review and then I'll make a pull request.

t-paul, 2023-08-13:
Closed via #4537.

closed 2023-08-13 by t-paul

amatulic, 2023-08-13:
> Closed via #4537.

Excellent. I look forward to this appearing in the next release.

benjaminwand, 2024-05-08:
For the sake of using OpenSCAD for architecture drawing, I would love it if larger formats or custom size pdf would be possible as well.

t-paul, 2024-05-08:
@benjaminwand this will get lost as comment here on the closed issue. Please create a new feature request and maybe also list the known-formats you would like to see as there might be different ones in use in various countries.