openscad/openscad #4145: Feature Request: collapse/expand all for bracket sections

UBaer21, 2022-02-25:
- Feature Request
collapse All/expand All  (show All / hide All) for bracket sections 
[image]

as option in edit menu or context menue

osfesa, 2022-04-07:
I implemented this feature request.

I found, doing Ctrl+Shift and click on the area where the arrows are, it already does the job requested.
I added a context menu option which uses QsciScintilla::foldAll function to fold/unfold all sections.

Is there any explanation on how I can send the Pull Request so you can review it?

Thanks.

t-paul, 2022-04-07:
Neat. The official docs are there: [about-pull-requests](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests)

In short:
* Clone the openscad repo into your github account (or refresh to update the `master` branch)
* Create a branch and add your changes there, push that to your own github repository
* Use the "Create Pull Request" that should automatically show up in the github page

UBaer21, 2022-04-07:
good to know, Thanks!

UBaer21, 2022-04-09:
I just noticed that CTRL+SHIFT is unfolding all but only folds the top level (which hides the other below).
But if i have:
```
{// outer
 {// inner 
 a=[
 1,
 2,
 3
 ];
 
 }
}
```
I can collapse the inner  or the list manually which stays collapsed if i  toggle (fold / unfold)  the outer.  With unfold all - all gets unfolded which is expected but when folding "all" only the outer/top section of the tree is folded and when opening, the sections below are still unfolded.
Not sure if there is a way to change this or to add something to fold/unfold everything below a certain level (where i click). 
 As this is to get a better overview - it would be great  if i open one tree and still have everything below folded instead of seeing pages of codes unfolded.

Cheers

t-paul, 2022-04-10:
@osfesa If you'd like to chat regarding your change, find me in the `#openscad` channel on IRC, e.g. via the web interface: https://web.libera.chat/?channel=#openscad (no registration needed).

closed 2022-04-12 by t-paul