openscad/openscad #4150: Incomplete render with fast-csg-trust-corefinement and fast-csg being set

printerderby, 2022-02-27:
Hello

I just read about some cool performance improvements and had to give them a try. My current model is pretty simple, yet running with a painful $fn value.
- OpenSCAD-2021.04.13: 160sec
- OpenSCAD-2022.02.26: 80sec
- OpenSCAD-2022.02.26, "fast-csg": 40s

Amazing!

I followed your call to test "fast-csg-trust-corefinement", for even more performance. This option alone seems to have no effect. I guess it does nothing on it's own and has to be used with "fast-csg". Maybe this can be described in the settings or "fast-csg" activated implicitly. 

Combining "fast-csg-trust-corefinement" and "fast-csg", results in a worse performance (160s instead of 80s; see #4114). 
In addition, the model is incomplete. Even exporting it (and putting it on a slicer), does result in lost elements. Lines 152 to 158 from [lazy-susan-3.txt](https://github.com/openscad/openscad/files/8149317/lazy-susan-3.txt) are missing. 

Maybe I am doing something wrong, as I am pretty new to OpenSCAD. 

Some screenshots and the tested code are attached

Regards
Steffen

// no options set - everything normal: 
[image]

// "fast-csg-trust-corefinement" set - looks and performs exactly like not being set at all 
[image]

// fast-csg - faster and the appearance changes
[image]

// "fast-csg-trust-corefinement" and "fast-csg" - elements are missing, poor performance 
[image]

printerderby, 2022-02-27:
Only using "fast_csg", I played around with the customizer:
- Changing ttl_dia to 50mm results in the same loss. Even with fast_csg, only.  
- Setting "view" to "all" and the ttl_dia to values of 47, 50, 98 and 100 will result in different results. Sometimes the output is just complete blank. 

Turning off "fast_csg" and the result looks like the preview again. Maybe someone can retest this. I guess, I am doing something wrong.

MichaelAtOz, 2022-02-27:
Why doesn't fast-csg show edges with `View/Show-edges`, first image above v's 3rd?

MichaelAtOz, 2022-02-27:
Fast-csg (just that feature set), for this model, is using 2.8-3.7 times the memory.
This is from System Monitor/properties on Debian, OpenSCAD 2022.02.13.
Fast-csg / Memory / Virtual / Resident
```
Off /  874.4MiB / 1.4 GiB / 981.8 MiB
On / 3.3GiB / 3.9GiB / 3.4 GiB
```

ochafik, 2022-03-04:
> Why doesn't fast-csg show edges with View/Show-edges, first image above v's 3rd?

@MichaelAtOz I didn't reimplement the viewing code to deal with the new mesh data structure. Looking into options (I might just let it try and create a Nef that can be rendered at the end).

> Fast-csg (just that feature set), for this model, is using 2.8-3.7 times the memory.

I'm very curious of whether upcoming `fast-csg-remesh` (https://github.com/openscad/openscad/pull/4117) will improve things (it should drastically drop the # of faces). Would you be able to compare the # faces in the STL output with and without fast-csg?

printerderby, 2022-03-08:
Thanks for your feedback. $fn got reduced and any object eliminated, that is not involved in causing the error.  

```
/* [Accuracy] */
// 
$fn=12;

// this mirror-action breaks anything, as long as the subpart uses a difference
mirror([0, 0, 1])
  translate([0, 0, -15])
  {
    difference()
    {
      translate([0, 0, 0])
        cylinder(h=26.25, d=11);
      // This negative part seems important. 
      translate([0, 0, 0])
        cylinder(h=26.25, d=2.8);
    }
  }

// this is the part, getting lost
cylinder(h=13, d=75);
```

Used: OpenSCAD-2022.03.07.ai11323-x86_64.AppImage with fast-csg-trust-corefinement *and* fast-csg enabled. 

The preview and the render will differ. The negative part (the latter part in the "difference()") in combination with the "mirror()" seem to cause the issue.
The large cylinder (last line) will get lost when rendering. Interestingly, it will also remove the intersecting part of the smaller cylinder.

Regards
Steffen

[image]
[image]

printerderby, 2022-03-08:
Used: OpenSCAD-2022.03.07.ai11323-x86_64.AppImage with fast-csg enabled, only.

The original model showed a similar problem, with fast-csg only. You had to use the view "all". 
Using that error as a base, there is an even simpler example. It also debunks my theory, about the negative object. Maybe it is just the "mirror()"'s amount of children.

```
$fn=12;
cylinder(25, 5, center=true);
mirror([0, 0, 1])
{
  cylinder(h=2, r=8, center=true);
  cylinder(h=6, d=12, center=true);
}
```
Again, the disappearing object also removes any parts of other, intersecting objects. 

[image]
[image]

ochafik, 2022-03-08:
@printerderby amazing, super useful, thanks a lot! Fix on its way :-) (forgot to flip the faces of surface meshes after a transform with a negative determinant, oops!)

closed 2022-03-08 by t-paul

ochafik, 2022-03-09:
> Fast-csg (just that feature set), for this model, is using 2.8-3.7 times the memory.

@MichaelAtOz Thanks for reminding me of the memory issue! Good news: found the culprit, the workaround is to trust (when you can, which you will from tomorrow's dev snapshot) and I'll work on a fix straight away.

There are 2 causes of huge memory consumption at play here:
- Lazy numbers going beserk --> normally meant to be fixed by `fast-csg-exact` or `fast-csg-exact-callbacks`, but I never had it cover the case where we fall back to Nef operations when corefinement seems too risky (which `fast-cst-trust-corefinement` avoids, but it was triggering the disappearance). The fallback nefs are using the same lazy kernel so have the same issue with "surprising" numeric performance (and clearly, memory).
- To a much lesser degree: corefinement's known propensity to generate more triangles: fixed by `fast-csg-remesh`.

Some measurements on my local build w/ the mirroring fix (https://github.com/openscad/openscad/pull/4164):
- With only `fast-csg` it mostly falls back to Nefs, almost no numbers are made exact and RAM spikes to 3.6GB.
- With `fast-csg` + `fast-cst-trust-corefinement` it uses 408MB
- With `fast-csg` + `fast-cst-trust-corefinement`+ `fast-csg-remesh` + `fast-csg-exact-callbacks` +  it uses 270MB (each of remesh & exact-callbacks brings down the memory usage a bit)

I'll now try and add exact calls on Nef numbers to keep the non-corefinable cases in check :-)

PR #4164 (2022-03-08): [fast-csg] Fix mesh orientation after transforms in case it mirrored 
Fixes #4150, and also the [lizards](https://www.thingiverse.com/thing:1034488/files) mentioned in [#4109](https://github.com/openscad/openscad/issues/4109#issuecomment-1036968712)

PolySet::transform does that very clearly but I probably looked more at CGAL_Nef_polyhedron when I wrote CGALHybridPolyhedron::transform, and the Nef does that flipping under the hood without our help 😅
commit 41bcdcfe (2022-03-08): [fast-csg] Fix mesh orientation after transforms in case it mirrored (fixes #4150)