openscad/openscad #4152: Willtel

willtel, 2022-03-03:
Hello,
I have been using OpenSCAD for years with out any problems until now.
I recently downloaded the Latest Version of OpenSCAD 2021.01 and have had several Errors where OpenSCAD just seemed to Stall (See Event Log below) and a Major CRASH BsoD +Kaspersky NIS + Total System was Unable to Boot in NORMAL MODE.

CRASH on Win7 RAID System:
I was editing some Parameters in one of my Scripts when the Application Just FROZE?
Then the System Crashed to BsoD, when I Rebooted (NORMAL MODE) it Booted Ok until 
Kaspersky NIS started then CRASHED to a BsoD ( This could have been due to Malware??), <<<< NB ?
this Occurred Over and Over again.
SO I then Installed RESTORO to try to recover my System in SAFE MODE.
During the RESTORO Analysis it Reported the following. (See Image Below)

[image]

These are Copies of the THREE EVENT LOGS shown above:

Log Name:      Application
Source:        Application Hang
Date:          25/01/2022 3:48:49 PM
Event ID:      1002
Task Category: (101)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      WebServerRAID
Description:
The program openscad.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Action Center control panel.
 Process ID: 9d8
 Start Time: 01d811aebf02af0b
 Termination Time: 7
 Application Path: C:\Program Files\OpenSCAD\openscad.exe
 Report Id: 5736d2ce-7da2-11ec-9d90-001d7d09fdbe

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Hang" />
    <EventID Qualifiers="0">1002</EventID>
    <Level>2</Level>
    <Task>101</Task>
OpenSCAD_CRASHES

    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-01-25T05:48:49.000000000Z" />
    <EventRecordID>1256301</EventRecordID>
    <Channel>Application</Channel>
    <Computer>WebServerRAID</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>9d8</Data>
    <Data>01d811aebf02af0b</Data>
    <Data>7</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>5736d2ce-7da2-11ec-9d90-001d7d09fdbe</Data>
    <Binary>55006E006B006E006F0077006E0000000000</Binary>
  </EventData>
</Event>

Log Name:      Application
Source:        Application Hang
Date:          28/01/2022 11:40:14 PM
Event ID:      1002
Task Category: (101)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      WebServerRAID
Description:
The program openscad.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Action Center control panel.
 Process ID: 18b4
 Start Time: 01d8144b9f9858f8
 Termination Time: 11
 Application Path: C:\Program Files\OpenSCAD\openscad.exe
 Report Id: ba3a9e96-803f-11ec-bfab-001d7d09fdbe

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Hang" />
    <EventID Qualifiers="0">1002</EventID>
    <Level>2</Level>
    <Task>101</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-01-28T13:40:14.000000000Z" />
    <EventRecordID>1256930</EventRecordID>
    <Channel>Application</Channel>
    <Computer>WebServerRAID</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>18b4</Data>
    <Data>01d8144b9f9858f8</Data>
    <Data>11</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>ba3a9e96-803f-11ec-bfab-001d7d09fdbe</Data>
    <Binary>55006E006B006E006F0077006E0000000000</Binary>
  </EventData>
</Event>

Log Name:      Application
Source:        Application Hang
Date:          14/01/2022 7:52:28 PM
Event ID:      1002
Task Category: (101)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      WebServerRAID
Description:
The program openscad.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Action Center control panel.
 Process ID: f68
 Start Time: 01d80924b5457893
 Termination Time: 7
 Application Path: C:\Program Files\OpenSCAD\openscad.exe
 Report Id: 8712ec24-751f-11ec-974a-001d7d09fdbe

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Hang" />
    <EventID Qualifiers="0">1002</EventID>
    <Level>2</Level>
    <Task>101</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-01-14T09:52:28.000000000Z" />
    <EventRecordID>1254011</EventRecordID>
    <Channel>Application</Channel>
    <Computer>WebServerRAID</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>f68</Data>
    <Data>01d80924b5457893</Data>
    <Data>7</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>8712ec24-751f-11ec-974a-001d7d09fdbe</Data>
    <Binary>55006E006B006E006F0077006E0000000000</Binary>
  </EventData>
</Event>

I hope this is of some use.
Kind Regards,
William

ChrisCoxArt, 2022-03-03:
A system hang, reboot, or kernel panic cannot be caused by an application - only kernel bugs, driver bugs, or bad hardware can cause those.  The fact that you could not boot in normal mode indicates a low level driver or hardware issue, but could be caused by sufficiently low level malware as well (ie: it has modified the boot sector, CPU protections, or a low level driver).

Also, hangs are not the same thing as crashes, and all the logs you provided indicate application hangs (non-responsive). And all three of those occurred in January.

t-paul, 2022-03-03:
If it even crashes with just Kaspersky, it does not seem likely there's any involvement with OpenSCAD. Maybe update that thing (which likely has deep system integration) or check if there's any GPU driver updates.

closed 2022-03-03 by t-paul