openscad/openscad #4166: Fast CSG render without Preview missing face from quad

UBaer21, 2022-03-12:
// F5 F6  differs from  (flush cache)  F6 without F5 (no preview)
// v. 2022 02 18

fn=8;
r=[5,3];

p=[
for(i=[0:fn-1])[sin(360/fn*i)*r[i%2],cos(360/fn*i)*r[i%2],0],
for(i=[0:fn-1])[sin(360/fn*i)*r[i%2],cos(360/fn*i)*r[i%2],10],
];

f=[
[for(i=[fn-1:-1:0])i],
[for(i=[fn:1:fn*2-1])i],
for(i=[0:fn -1],i=i%fn)[i,i+1,i+fn,i+fn -1]
];

union(){
polyhedron(p,f);
cube();
}

MichaelAtOz, 2022-03-12:
The second face is wrong. `ECHO: "f[2]=[0, 1, 8, 7]"`
I have not checked the others.
```
fn=8;
r=[5,3];

p=[
for(i=[0:fn-1])[sin(360/fn*i)*r[i%2],cos(360/fn*i)*r[i%2],0],
for(i=[0:fn-1])[sin(360/fn*i)*r[i%2],cos(360/fn*i)*r[i%2],10],
];

f=[
[for(i=[fn-1:-1:0])i],
[for(i=[fn:1:fn*2-1])i],
for(i=[0:fn -1],i=i%fn) [i,i+1,i+fn,i+fn -1]
];

for (i=[0:len(f)-1]) echo(str("f[",i,"]=",f[i]));
echo();
for (i=[0:len(p)-1]) echo(str("p[",i,"]=",p[i]));

union(){
polyhedron(p,f);
cube();
}

polydebug(p, echoPoints=false);

module polydebug(p, echoPoints=true) {
  
  for( i=[0:len(p)-1]) {
    if (echoPoints) echo(i,p[i]);
    translate(p[i]) {
      color("black") sphere(0.2, $fn=12);
      translate([-0.5,0.5,( (p[i].z>0) ? 0.2 : -0.4) ]) 
        color("red") 
            linear_extrude(height=0.2) 
                rotate([0, ( (p[i].z>0) ? 0 : -180), 0])
                    text(str(i), size=1);
    }
  }
} 
```

```
Compiling design (CSG Tree generation)...
ECHO: "f[0]=[7, 6, 5, 4, 3, 2, 1, 0]"
ECHO: "f[1]=[8, 9, 10, 11, 12, 13, 14, 15]"
ECHO: "f[2]=[0, 1, 8, 7]"
ECHO: "f[3]=[1, 2, 9, 8]"
ECHO: "f[4]=[2, 3, 10, 9]"
ECHO: "f[5]=[3, 4, 11, 10]"
ECHO: "f[6]=[4, 5, 12, 11]"
ECHO: "f[7]=[5, 6, 13, 12]"
ECHO: "f[8]=[6, 7, 14, 13]"
ECHO: "f[9]=[7, 8, 15, 14]"
ECHO: 
ECHO: "p[0]=[0, 5, 0]"
ECHO: "p[1]=[2.12132, 2.12132, 0]"
ECHO: "p[2]=[5, 0, 0]"
ECHO: "p[3]=[2.12132, -2.12132, 0]"
ECHO: "p[4]=[0, -5, 0]"
ECHO: "p[5]=[-2.12132, -2.12132, 0]"
ECHO: "p[6]=[-5, 0, 0]"
ECHO: "p[7]=[-2.12132, 2.12132, 0]"
ECHO: "p[8]=[0, 5, 10]"
ECHO: "p[9]=[2.12132, 2.12132, 10]"
ECHO: "p[10]=[5, 0, 10]"
ECHO: "p[11]=[2.12132, -2.12132, 10]"
ECHO: "p[12]=[0, -5, 10]"
ECHO: "p[13]=[-2.12132, -2.12132, 10]"
ECHO: "p[14]=[-5, 0, 10]"
ECHO: "p[15]=[-2.12132, 2.12132, 10]"
Compiling design (CSG Products generation)...
Geometries in cache: 173
Geometry cache size in bytes: 1497600
CGAL Polyhedrons in cache: 69
CGAL cache size in bytes: 1055048
Compiling design (CSG Products normalization)...
Normalized tree has 34 elements!
Compile and preview finished.
Total rendering time: 0:00:00.017
```

MichaelAtOz, 2022-03-12:
Again I haven't checked them all but many quads are not co-planar.

MichaelAtOz, 2022-03-12:
I am correct that quads need to be co-planar??
With fast-csg OFF CGAL paints that [0,1,8,7] as a folded two triangle dual face, [0,1,8,0] & [0,8,7,0]. (Incorrectly?)
With fast-csg ON CGAL paints a triangle [1,8,7,1].

MichaelAtOz, 2022-03-12:
[image]

UBaer21, 2022-03-12:
@MichaelAtOz  All  vertical faces of  f are non planar - that is the point,  this is something what is working in preview and renders fine if a preview is done - assuming the preview generates the proper tri faces.

MichaelAtOz, 2022-03-12:
Preview does not concern itself with validity of geometry. It paints whatever garbage you give it.
There has been a F5-before-F6 v F6 problem for a long time. 
The problem here seems to be that preview is triangulating the quads, thus 'working-around' the invalid non co-planar quads.

MichaelAtOz, 2022-03-12:
Hmmm, OK Wiki says `If points that describe a single face are not on the same plane, the face is automatically split into triangles as needed.`.

UBaer21, 2022-03-12:
Maybe #4148 is related somehow .. however  an error "non planar faces - only planar supported"  would be fine too - but the inconsistency that the render is different to the preview is bad.  Also if something was working even if only due to some fixing and isn't with the new feature - it may break some designs. And this causes trouble if the render is e.g. done with different settings ($fa/$fs/$fn - or random without seed) as then there is never preview data.  I also have experienced this in other cases which very specific $fn number (was 72) and triangles causing troubles as somehow faces differed  - and leading to "not closed" meshes.  The Quad faces  also can behave different depending on the angles so i agree that it is not ideal to use them at all - but if the other tips of that star worked - so why is the first/last tip behaving differently.

nophead, 2022-03-12:
I think non-planar faces have to be supported because with floats, angled planes with more than three points are almost impossible to be perfectly planar.

closed 2022-03-14 by t-paul