openscad/openscad #4169: lazyunion-difference-for.scad test actually fails? (wrong expectation)

ochafik, 2022-03-12:
tests/data/scad/experimental/lazyunion-difference-for.scad

```
difference()
    for (i=[0:2])
        rotate(90+i*360/3) 
            translate([4,0,0])
                sphere(5);
```

[Without `--enable=lazy-union`: 3 spheres](https://tinyurl.com/22t9kfhx)

[With `--enable=lazy-union`: an eaten up sphere](https://tinyurl.com/2p8heecf)

UBaer21, 2022-03-12:
I would expect this as lazy U keeps the loop separate objects and doesn't group them.

so without there is a difference without second object ⇒  3 spheres
with lazy U  the difference is the first of the loop minus the following objects from the loop.

UBaer21, 2022-03-12:
```
difference()
    union()for (i=[0:2])
        rotate(90+i*360/3) 
            translate([4,0,0])
                sphere(5);
```

t-paul, 2022-03-12:
I think it should be 3 spheres even with lazy union. As indicated by the discussion in #3143, the slots for what's considered "the first object" likely need to be static. So regardless of how may objects are returned by the `for()`, all end up in the first positive slot of the `difference()`.