openscad/openscad #4174: Feature Request: Adjustable highlight color for editor.

Zenmaster43, 2022-03-14:
Currently, the highlight color in the built-in text editor is a very light gray color, which does not attract any attention and makes it very difficult to visually locate the text found by the search feature. To resolve this, I request that the following enhancements be made:
- Add a "Highlight Color" color picker control to the "Editor" tab of the "Preferences" window.
- Add a "Use system highlight color" checkbox control in the "Editor" tab of the "Preferences" window, next to the color picker control. When ticked, this would disable the color picker and show the OS highlight color and use it as the highlight color in the text editor. This allows the user to adjust their highlight color system-wide, rather than having to adjust it in the application and the system.

t-paul, 2022-03-14:
Color schemes can be customized by copying the one shipped to the local config folder and updating that. Maybe there's going to be a GUI for that someday, but that's not really high on the list.

Example on Linux
Color scheme: `/usr/share/openscad/color-schemes/editor/tomorrow.json`
User config folder: `$HOME/.config/OpenSCAD/color-schemes/editor`

The base of the user configuration can be found in `Help`->`Library Info` as `User Config Path`.

closed 2022-03-14 by t-paul

jordanbrown0, 2022-03-14:
If there's a system standard for the various highlight colors (selected text, search matches, current search match), I suggest that OpenSCAD use that standard without providing a UI to change it.  I don't need to have a way for OpenSCAD to behave differently from MS Word (or whatever), and in fact don't *want* it to be different.

I believe that on Windows there's a system standard for selected text.  I don't know about search matches.

Zenmaster43, 2022-03-14:
That sounds reasonable. OS X has a standard also, so it should be used, rather than the light gray color. However, OS X also has “Dark Mode”, so I’m looking at whether it would change the highlight color on the fly when switching to or from dark mode.

Zenmaster43, 2022-03-14:
I think search matches and selected text highlighting should be the same unless multiple search results are simultaneously highlighted. In that case, the selected one might be differently highlighted than the others.

jordanbrown0, 2022-03-14:
Under Windows there are at least three kinds of highlight in the OpenSCAD text editor:
- selected text
- all search matches
- the current search match
I don't know whether Windows has standards for all of those.