openscad/openscad #4179: Performance issues with 2022.03.14 on higher $fn values

printerderby, 2022-03-15:
Hello,

For my latest bugreport (#4178), I downloaded the latest build. 

Rendering time with 2022.03.08: 63 seconds
Rendering time with 2022.03.14: 52 seconds

Happy with another performance improvement, I noticed being unable to render my lazy-susan from #4150, at all. 

There is a very basic model, reproducing the problem. The rendering time seems to increase exponentially, with the $fn-value.

Thank you :)
Steffen 

[image]

```$fn=200;
shell=1;

rotate_extrude()
  translate([45, 0, 0])
    square([10, 5]);

if (shell==1)
  difference()
  {
    cylinder(h=10, d=120);
    cylinder(h=10, d=119);
  }
```

// OpenSCAD-2022.03.08: 0.5 sec
```
wurm@tark:~/Downloads$ ./OpenSCAD-2022.03.08.ai11331-x86_64.AppImage bench-donut.scad -D 'shell=1' --export-format stl -o /dev/null --enable fast-csg
Geometries in cache: 6
Geometry cache size in bytes: 424256
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 52680
Total rendering time: 0:00:00.472
Top level object is a 3D object:
   Facets:       3480
```

// OpenSCAD-2022.03.14: 68sec
```
wurm@tark:~/Downloads$ ./OpenSCAD-2022.03.14.ai11423-x86_64.AppImage bench-donut.scad -D 'shell=1' --export-format stl -o /dev/null --enable fast-csg
Geometries in cache: 6
Geometry cache size in bytes: 509504
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 85832
Total rendering time: 0:01:08.108
Top level object is a 3D object:
   Facets:       4664
```

// OpenSCAD-2022.03.14; Default render: 1.8sec 
```
wurm@tark:~/Downloads$ ./OpenSCAD-2022.03.14.ai11423-x86_64.AppImage bench-donut.scad -D 'shell=1' --export-format stl -o /dev/null 
Geometries in cache: 5
Geometry cache size in bytes: 173544
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 3227664
Total rendering time: 0:00:01.788
Top level object is a 3D object:
   Simple:        yes
   Vertices:     1600
   Halfedges:    4800
   Edges:        2400
   Halffacets:   1608
   Facets:        804
   Volumes:         3
```

// OpenSCAD-2022.03.14; Half the $fn, 10th the rendering time (6.5sec)
```
wurm@tark:~/Downloads$ ./OpenSCAD-2022.03.14.ai11423-x86_64.AppImage bench-donut.scad -D '$fn=100' --export-format stl -o /dev/null --enable fast-csg
Geometries in cache: 6
Geometry cache size in bytes: 254912
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 42824
Total rendering time: 0:00:06.417
Top level object is a 3D object:
   Facets:       2328
```

// OpenSCAD-2022.03.14; rendering the inner donut, only: 0.0sec
```
wurm@tark:~/Downloads$ ./OpenSCAD-2022.03.14.ai11423-x86_64.AppImage bench-donut.scad -D 'shell=0' --export-format stl -o /dev/null --enable fast-csg
Geometries in cache: 3
Geometry cache size in bytes: 115640
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.025
Top level object is a 3D object:
   Facets:       1600
wurm@tark:~/Downloads$ 
```

rcolyer, 2022-04-17:
While I don't mind a performance boost on it, I think this should probably be a "wontfix" unless it appears on a different testcase, because the initial design is invalid (and that looks like it might be the source of the problem).  This violates the overlap rule for unions and differences described under the "Note:" in the manual for those sections:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#union

Depending on what exactly is going on internally for a particular design or calculation pathway, there are many reasons this could lead to a performance problem in current or future code, such as differences in processing resulting in any arbitrary number of infinitesimal volumes to calculate along the coincident surfaces, or attempting to do a bunch of calculations on nans.

printerderby, 2022-04-18:
Thanks for pointing this out. I was unaware of this general problem. 
Adding a translate() within the difference() implicit union(), solved this issue. 

```
$fn=200;
shell=1;

rotate_extrude()
  translate([45, 0, 0])
    square([10, 5]);

if (shell==1)
  difference()
  {
    cylinder(h=10, d=120);
    translate([0, 0, -1])
      cylinder(h=12, d=119);
  }
```

closed 2022-04-18 by printerderby

nophead, 2022-04-18:
Not it is much faster to difference two circles and linear extrude than it is to difference two cylinders. It could also be a rotate_extrude of a square like the other ring and that would be faster than a 3D difference as well.

Presumably one could rotate extrude two squares to get the same result instantaneously.

printerderby, 2022-04-18:
@nophead : Thank you for the input. It did not come to my mind, to use an extruded 2D ring, instead of the difference on 2 cylinders. 

Regarding the rotate_extrude(): This example was crafted by simplifying a complex model (seen in https://github.com/openscad/openscad/issues/4150 ). This model got hit by a MASSIVE performance hit, with the newest version under fast-csg. So massive in fact, it did not even seem to ever complete.

MichaelAtOz, 2022-04-21:
(Note, sorry dragging this up, but I'm way behind in reading these emails)

> ...because the initial design is invalid (and that looks like it might be the source of the problem). This violates the overlap rule for unions and differences described under the "Note:" in the [...]

@rcolyer this has never been the case that `It is mandatory for all unions, explicit or implicit, that external faces to be merged not be coincident.` (and the diference equivalent). You put that comment in the wiki in 2021.

It is good practice to not do that, but there are lots of historic .scad's which have acceptable coincident faces (aligned with the axis for example). Preview is often .broken' with z-fighting, but CGAL handles many cases.
I don't think we can just introduce mandates, but explain why is is not good practice.