openscad/openscad #4188: Support for 3D Mouse Standard interface

brainlessboy, 2022-03-23:
I am using the 3DConnexion Space Mouse which uses the standard 3D Mouse Interface, it would be great if this would work out of the box since i also use the mouse on other software (Blender, Prusa Slicer, Zbrush) on these software the mouse works out of the box and does require to uninstall the standard browser

L-K-M, 2022-03-23:
I wonder what Blender is doing. They appear to support the Space Mouse's native drivers (at least it feels the same as other 3D applications), and it works while the Space Mouse driver is running - there's no setup required, it just works.

t-paul, 2022-03-25:
As said before, native driver is impossible from legal perspective. To change that the topic has to be brought up with 3DConnexion. So unless someone can bring some clarity on how the legal challenges could be solved, I have to close the issue as impossible to solve.

I repeat that I'm happy to get the setup improved, but that will need someone to **directly** and **actively** help with getting that working. That does **not** need developer knowledge, the main work would be continued testing and discussion on how to improve from user perspective.

I can't just buy all the devices myself and spend countless hours just to find some setup that may or may not work for others. Blender has a huge community, commercial sponsors and paid developers, we can't match that by any means.

closed 2022-03-25 by t-paul

jordanbrown0, 2022-03-25:
I registered for 3DConnexion's developer program so that I could see the SDK license agreement.

The relevant portions seem to be:
> 1. License Grant
> Subject to the terms and conditions contained in this Agreement, 3Dconnexion grants to you a personal limited non-exclusive, non-transferable, non-sublicensable, world-wide, royalty-free, revocable license to install, use, copy, modify and distribute in object code form the 3DCONNEXION SDK. In addition no license is granted under any patents that may be or later become infringed by your modifications or derivative works in which the 3DCONNEXION SDK may be incorporated. No other rights are granted.
> 
> 2. Attribution Requirements and Trademark License
> In the format set forth below, you must provide attribution to 3Dconnexion in one of the following locations: (1) the "About" or "Info" box menu items (or equivalent) of your software application, (2) the "Help" file (or equivalent) of your software application:
> 
> "3D input device development tools and related technology are provided under license from 3Dconnexion. © 3Dconnexion 1992 - 2020. All rights reserved."
> 
> 3. Restrictions
> The 3DCONNEXION SDK contains copyrighted material, trade secrets, intellectual property and other proprietary materials of 3Dconnexion. You agree that in order to protect those proprietary materials this license is granted to you only to integrate a hardware or software product to operate and communicate with a hardware product designed and distributed by 3Dconnexion.  You may not use any element of the 3DCONNEXION SDK to create or enhance a product that competes with a product developed and/or distributed by 3Dconnexion.
> 

I will ask them for their advice for supporting open-source projects.

jordanbrown0, 2022-03-25:
> 5. Confidentiality
The source code and instructions contained in the 3DCONNEXION SDK are trade secrets of 3Dconnexion and you agree to maintain the confidentiality of such source code and instructions, to use them only as authorized in this License and not to disclose them to any third party.

jordanbrown0, 2022-03-25:
That last might be the hard part.

t-paul, 2022-03-25:
Asking can't hurt, I suppose, but I know the strategy: creating a GPL shim around the libraries. See [Kicad MR#845](https://gitlab.com/kicad/code/kicad/-/merge_requests/845/diffs#dd497a6bfe1583992860758f2f5aa9a3379770ea_0_89). I don't consider this a valid solution and that topic has been debated to death in context of linux device drivers.

jordanbrown0, 2022-03-25:
I agree that that doesn't seem like a valid solution - or, more precisely, I don't see how it helps.  Either it's OK to use the SDK with your open-source program (after jumping through whatever hoops), or it's not OK to use the SDK with your open-source shim.  I don't understand how it could be OK with an open-source shim, and not OK with the primary open-source program.

jordanbrown0, 2022-03-26:
https://spacemice.org/index.php?title=Blender says that Blender talks directly to the device, bypassing 3DConnexion's driver.

jordanbrown0, 2022-03-26:
At least as of Blender 2.78.  Current is 3.1.

MichaelPFrey, 2022-03-26:
I always wonder how much devices we want/need to support natively.

>  creating a GPL shim around the libraries.

We have the dbus interface.
"The shim" thus could be a standalone (closed source) program, not part of this project.

> I don't consider this a valid solution and that topic has been debated to death in context of linux device drivers.

I would not mind if a closed source program is using the dbus interface.
It does not expose core internal openscad functionality.

The linux kernel developers **do mind** linking against kernel files.
It does expose core internal Kernal functionality and workings.

My examples for dbus uses qdbus and I copy pasted a file from openscad into the other project.
That could be borderline with the GPL conformity, when done in a closed source project.
(qt and closed source is also a can of worms)

If I remember correctly, you can scan a dbus interface. (get method names and signatures)
Using information openscad exposed it self in an interface to use the interface sounds like a issue not affected by GPL.

>  I don't understand how it could be OK with an open-source shim, and not OK with the primary open-source program.

I suspect you see some inner working details.
By controlling openscad via dbus, all the opensource code sees is camera manipulation (not correlated to any raw values of the device or user interaction).

L-K-M, 2022-03-26:
> That does not need developer knowledge, the main work would be continued testing and discussion on how to improve from user perspective.

One thing that would help would be to enable the HIDAPI driver in the development snapshot builds. That would make testing and providing feedback easier. For example, the current release build doesn't restore axes mappings correctly, so it's very difficult to try to set things up correctly. This seems to be fixed in the dev snapshot, but HIDAPI doesn't work there.

t-paul, 2022-03-26:
@L-K-M Please be specific, "HIDADPI does not work in development snapshots" is wrong, I'm running the Debian/Testing package `2022.03.22.nightly (git e34e756)` and that has it enabled. So *which specific* build does not have it enabled?

L-K-M, 2022-03-26:
Apologies, I only checked on Mac and Windows, and made the assumption that it would be the same for all builds.

If I open a development snapshot build on Windows (e.g. 2022.03.22.ci11484 git e34e756d2) and go to Preferences > Axes, "HIDAPI" is disabled. If I hover over the checkbox, it says "The driver was not enabled during build time and is thus not available."

t-paul, 2022-03-27:
Windows builds updated via #4192.