openscad/openscad #4189: handling of vectors in the Customizer

krhackbarth, 2022-03-23:
I can define a variable has being initialized with the value "[]"

I then change that value to a non-null value in the customizer - e.g., "[1,3,5]"

I then try to delete the non-null values and return the variable to a null vector "[]".

However, when I press the enter key (or move focus to another input), the non-null value is repopulated "[1,3,5]".

Please support the ability to undo a change in the Customizer to the value of a vector.

t-paul, 2022-03-23:
This sounds like #4177. Editing arbitrary length vectors as text is not supported currently.

closed 2022-03-23 by t-paul

krhackbarth, 2022-03-24:
I'll take your word for it.  Is there a list of feature requests that this issue has gone into?

t-paul, 2022-03-24:
I'm not sure there is one, but the current customizer is targetting features of the Thingiverse customizer and there's no plan to extend this.
At some point there's hopefully going to be a native customizer definition (in the scripts, as opposed to the Thingiverse comment style) too. This might be the base for extending the customizer editor behavior too.